tsduck/tsduck #398: 'illegal instruction' error on ARM926EJ-S platform

torfl, 2019-08-27:
I've tried to execute a tsduck application on an ARM926EJ-S device running linux as OS.
The application crashes immediately with the message 'illegal instruction'.

This error is caused by the memory barrier code that uses the _MCR_ assembler instruction, because on ARM926EJ-S CPUs this instruction can only be executed in priviliged mode.

For now I replaced it with the following code that should prevent that the compiler reorders memory access.
`__asm__ __volatile__ ("" :  : :  "memory");`

lelegard, 2019-08-27:
Hi @torfl,

Thank you for reporting this weird problem.

Normally, a _memory barrier_ is some physical instruction which "does something" on the hardware. This is not only a compilation trick to prevent instruction rescheduling. This is more than that. A memory barrier forces the CPU to flush all memory instructions currently executing in the pipeline and stalls the CPU until those flushes are completed. This is of course very CPU-dependent.

So, in theory, what you are doing is not a true _memory barrier_.

That being said, I admit that using memory barriers in TSDuck is extremely pedantic and probably - or even certainly - useless. It certainly comes from my former background as an O/S developer and some habits I took when I worked on CPU architectures such as the DEC Alpha which was extremely sensitive to that class of problems.

So, in practice, you are safe.

I am not familiar with the various flavours of the ARM architecture. On my Raspberry Pi 3 with a Cortex-A53 CPU, the code executes fine. So, is the instruction MCR reserved to privilege mode in some ARM CPU's only? Is there a way to detect that class of CPU using conditional compilation? At run time? 

Otherwise, considering its lack of real use in TSDuck, I may simply remove the memory barrier or replace it with the compilation trick you use on ARM.

torfl, 2019-08-28:
I need to correct myself. It is not the MCR instruction itself that requires the privileged mode, but the access to the CP15 registers. Unfortunately it depends on the ARM architecture, which CP15 register can also be accessed in user mode.

Regardless of this, the specific memory barrier instruction ("`MCR p15, 0, <Rd>, c7, c10, 5`") is not supported by the ARM926EJ-S CPU.

The CPU's architecture can be detected during the compilation. GCC provides specific defines for that, e.g. `__ARM_ARCH_5TEJ__` for the ARM926EJ-S.

The platform specific memory barrier could be implemented like this:
```
#if defined(__ARM_ARCH_5TEJ__)
        __asm__ __volatile__ ("" : : :  "memory");
#else
        // For later reference, not sure this is valid.
        unsigned dest = 0;
        __asm__ __volatile__ ("@MemoryBarrier\n mcr p15,0,%0,c7,c10,5\n" : "=&r"(dest) : :  "memory");
#endif
```

lelegard, 2019-08-28:
OK, I implemented your suggestion.

closed 2019-08-29 by torfl