tsduck/tsduck #399: INJECT plugin: repetition interval based on section size

lars18th, 2019-08-27:
Hi @lelegard,

I have a problem with the `inject` plugin and the repetition interval. For the smoothing of the PSI/SI tables (commented at https://github.com/tsduck/tsduck/issues/370#issuecomment-516394193), I’m using the technique of extracting the binary table (using the `tables` plugin) and reinserting it with the `inject` plugin. However, I have a problem with the NIT table...

When the NIT table is large (the size is two or more packets) then the interval between packets is based on individual packets, and not on the full size. Note this example when using `-P inject -p 16 --binary table.nit=5000`:

```
- NIT size 1 packet:   ---N---/  5 sec. /---N---/  5 sec. /---N---  ...
- NIT size 2 packets:  ---Na---/ 5 sec. /---Nb---/ 5 sec. /---Na--- ...
```

However, the behavior I expected would be:

```
- NIT size 2 packets:  ---NaNb---/ 5 sec. /---NaNb---/ 5 sec. /---NaNb--- ...
```

So, I request for one parameter like `-P inject -p 16 --full-section --binary table.nit=5000`.
Could it be possible, please?

Regards.

lars18th, 2019-08-29:
Hi @lelegard ,

To solve this problem I'm trying something like:
https://github.com/tsduck/tsduck/blob/master/src/tsplugins/tsplugin_inject.cpp#L517
```
     // In non-replace mode (new PID insertion), replace stuffing packets when needed.
     if (!_replace && !_completed && pid == PID_NULL && _packet_count >= _pid_next_pkt) {
         replacePacket(pkt);
-        _pid_next_pkt += _pid_inter_pkt;
+        if (pid _pzer.atSectionBoundary())
+           _pid_next_pkt += _pid_inter_pkt;
     }
```

However, it doesn't work.

My problem is with tables like SDT/BAT. With large sizes I want to write the entire table (2 or 3 TS packets) in packets together, and not in TS packets with a big separation between them.

Any idea?
Regards.

lars18th, 2019-08-29:
Hi @lelegard ,

I solved my problem with this simple change:
```
diff --git a/src/tsplugins/tsplugin_inject.cpp b/src/tsplugins/tsplugin_inject.cpp
index 1b62fc2c..22f3c673 100644
--- a/src/tsplugins/tsplugin_inject.cpp
+++ b/src/tsplugins/tsplugin_inject.cpp
@@ -514,7 +514,8 @@ ts::ProcessorPlugin::Status ts::InjectPlugin::processPacket(TSPacket& pkt, TSPac
     // In non-replace mode (new PID insertion), replace stuffing packets when needed.
     if (!_replace && !_completed && pid == PID_NULL && _packet_count >= _pid_next_pkt) {
         replacePacket(pkt);
-        _pid_next_pkt += _pid_inter_pkt;
+        if (_pzer.atSectionBoundary())
+           _pid_next_pkt += _pid_inter_pkt;
     }

     return TSP_OK;
```

I hope you'll agree to include it.
Regards.

lars18th, 2019-08-29:
Hi @lelegard ,

I need to advise that this proposed change implies that the repetition rate is reduced by the number of chunks. So, I suggest to implement some option to enable this mode (disabled by default), and advise to the user to multiply the repetition rate by the number of expected chunks.

Regards.

lelegard, 2019-08-29:
Sorry, but this is not what I am seeing. It works correctly for multi-packets tables.

Using the DTTV [stream here](https://tsduck.io/streams/france-dttv/tnt-uhf22-482MHz-2019-01-22.ts), I extract the NIT. It uses one section, over 4 packets.

Now, I reinject it on the same stream after cleaning up PID 16 and a few services to make sure I can insert packets wherever they should be:

~~~
tsp -I file tnt.ts \
    -P svremove 0x0301 --stuffing \
    -P svremove 0x03F5 --stuffing \
    -P svremove 0x03F6 --stuffing \
    -P filter --negate --pid 16 --stuffing \
    -P inject --pid 16 nit.bin=5000 \
    -P analyze -o tnt-out.txt \
    -O file tnt-out.ts
~~~

The stream is 60 seconds long. The NIT uses 4 packets. At 5 seconds repetition rate _a section_, we should have 48 packets. And we get 49.

Look at those excerpts from the output analysis:
~~~
|  Broadcast time: ..................................... 60 sec (1 mn 0 sec)  |

|=============================================================================|
|  PID: 0x0010 (16)                                                  DVB-NIT  |
|-----------------------------------------------------------------------------|
|  Global PID                Transport:                Discontinuities:       |
|  Bitrate: ..... 1,211 b/s  Packets: ............ 49  Expected: ......... 0  |
|  Access: .......... Clear  Adapt.F.: ............ 0  Unexpect: ......... 0  |
|                            Duplicated: .......... 0  Sections:              |
|                            PCR: ................. 0  Unit start: ...... 15  |

|=============================================================================|
|  PID: 0x0010 (16)                                                  DVB-NIT  |
|-----------------------------------------------------------------------------|
|  0x40 (64, NIT Actual), TID ext: 0x20FA (8442)                              |
|      Repetition: .... 4327  ms  Section cnt: ....... 14                     |
|      Min repet.: .... 3748  ms  Table cnt: ......... 14                     |
|      Max repet.: .... 5002  ms  Version: ........... 30                     |
~~~

The repetition rate is slightly faster than the requested 5 seconds. There is probably some room for improvement in accuracy. But the repetition rate is definitely the section rate, not the packet rate.

You wrote you got this:
~~~
- NIT size 2 packets:  ---Na---/ 5 sec. /---Nb---/ 5 sec. /---Na--- ...
~~~
while you expected this:
~~~
- NIT size 2 packets:  ---NaNb---/ 5 sec. /---NaNb---/ 5 sec. /---NaNb--- ...
~~~
But in fact you should get this:
~~~
- NIT size 2 packets:  ---Na---/ 2.5 sec. /---Nb---/ 2.5 sec. /---Na--- ...
~~~
And this is what I get (except that there are 4 packets per section and an inter-packet interval of 1.25 second).

The packets are smoothened on the stream. Packets from the same section should not be adjacent. At least not in a broadcast system.

lars18th, 2019-08-30:
Hi @lelegard ,

Thank you for your response!

> Sorry, but this is not what I am seeing. It works correctly for multi-packets tables.

> [...]

> The packets are smoothened on the stream. Packets from the same section should not be adjacent. At least not in a broadcast system.

OK. That's the point. From my point of view, the packets from the same section should be _approximately_ adjacent. If not, then my request doesn't have sense.

Once again, thank you so much for your help. :wink:
I'll close this issue.
Regards.

closed 2019-08-30 by lars18th