tsduck/tsduck #403: Option not in DVB

joolzg, 2019-08-27:
I would like to be able to do the same as --satellite-frequency but on a dvb input.

Reason for this is an exotic LNB which seems to not behave and the only way i have it working with dvblast is using the IF frequency.

If its possible without this command can you give me a hint

lelegard, 2019-08-28:
Hi @joolzg,

With `-I dvb`, the `--frequency` option is already a satellite frequency. The intermediate frequency, at the input of the tuner,  is computed from the LNB characteristics. If you have a non-standard LNB, then specify its frequency characteristics using option `--lnb`.

With `-O dektec`, the `--frequency` option is the output frequency of the modulator. The option `--satellite-frequency` is a way of computing  `--frequency` as if it was a downlink and there were a reception LNB. This is easier to reproduce the frequency of an actual satellite transponder.

I agree that the intermediate frequency and the satellite frequency are not consistent but, for the sake of simplicity, the option `--frequency` is mapped to the most frequent use case for the corresponding plugin.

joolzg, 2019-08-28:
OK

closed 2019-08-28 by joolzg