tsduck/tsduck #404: Memory alignment problems on some ARM architectures

torfl, 2019-08-28:
The header tsPlatform.h contains several methods that are used to extract integers of different sizes and sign (uin16_t, uint32_t, ...) from serialized data, which are usually provided as byte array.

These methods do a simple cast of the byte array pointer to the pointer of the specific integer type that should be extracted. This doesn't work properly on systems that require a strict alignment, e.g. older ARM architectures like armv5te (ARM926EJ-S CPU).

For example, the method ts::TSPacket::getPID() always returns a wrong value for the PID.
Assuming a TS packet contains the following first 4 bytes: `0x47, 0x40, 0x00, 0x14`.
The expected PID is 0, but the returned value is 0x740.

The following rework of the method GetUInt16() in tsPlatform.h fixes this problem:
```
TSDUCKDLL inline uint16_t GetUInt16(const void* p)
{
        uint16_t value = *(static_cast<const uint8_t*>(p)) << 8;
        value |= *(static_cast<const uint8_t*>(p) + 1);

        return value;
}
```
The same issue applies to the methods that serializes integers.

I've added a patch ([fix-memory-alignment-issue.patch.txt](https://github.com/tsduck/tsduck/files/3549905/fix-memory-alignment-issue.patch.txt)) that fixes the issue in all the relevant methods. For now this patch applies to all CPU architectures. But its also possible to make this architecture dependent.

lelegard, 2019-08-28:
Hi @torfl,

I adapted your suggestion. Using the new version for all CPU architectures is a bad idea because the performances would be significantly degraded on widely used architectures such as Intel.

So, I did the following modifications:

- Define symbol `TS_STRICT_MEMORY_ALIGN` for CPU's requiring strictly aligned memory access.
- This symbol is currently defined for `__ARM_ARCH_5TEJ__` only (more will be added if necessary).
- Your version of the memory functions are used when `TS_STRICT_MEMORY_ALIGN` is defined. The previous versions remain active for other platforms.
- When compiled, the new versions are no longer inlined since the code would be too big. The more compact previous versions remain inline.

Please test on your platform and let us know if it works.

Just out of curiosity, what kind of platform use such armv5te CPU?

torfl, 2019-08-29:
The changes work on my platform. Thank you.

The armv5te CPU is part of a hardware we developed several years ago for DVB applications. Its an old platform but still in use.
Currently I'm trying to use TSDuck for the analysis of PSI/SI tables directly on this hardware.

closed 2019-08-29 by torfl