tsduck/tsduck #405: HLS output

joolzg, 2019-08-28:
How does hls generation work.

I set hls to have 12 second segments and 5 segments per m3u8, takes 60 seconds before any HLS data is written

12 second segments and 20 segments per m3u8, takes 50 seconds before any HLS data is written

What is the criteria to start outputting data

CLI
tsp -v -I http "http://your/favourite/live/stream" -P pcrbitrate -P fork 'tsp -v -O hls  /tmp/test-0-.ts -p /tmp/test-0.m3u8 -d 12 -l 5' -P regulate -O ip -e 224.224.224.1:1234

VERSION
tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.19-1392

I would ask that the HLS starts outputting on the 1st segment it creates and populates the m3u8 with filenames that are not available, or make this an option.

lelegard, 2019-08-29:
Hi @joolzg,

The hls output plugin does not introduce any initial delay. It starts outputting packets as soon as they arrive.

This is maybe due to buffering artifacts (especially because you serialize two instances of tsp). Try to reduce buffering and burst size in both tsp commands. See options `--max-input-packets` and `--max-flushed-packets`. When dealing with low bitrates, the initial buffering means a lot of time.

joolzg, 2019-09-02:
Added in a --max-input-packets and  works

closed 2019-09-02 by joolzg