tsduck/tsduck #406: pes plugin not showing audio attributes for unaligned audio

bobben1no, 2019-08-29:
Hi!  

I'm using pes plugin to extract video and audio attributes (tsduck_3.18-1361)
tsp -I file testfile.ts -P pes -a -v

It works just great most of the time but pes plugin will not show audio attributes for some services.

I have found that this probably relates to the audio ES not being aligned inside the PES packet. The PES data_alignment_indicator is also set to "0".  

The audio header can still be found inside the payload start TS packet but not at the first byte of the PES payload.

Here is example of Ts packet with payload start & PES header:
(I have marked the audio header with arrows)
```
47 4C 3E 16 00 00 01 C0 18 6A 80 80 05 23 33 BF 
FC 95 E3 18 C6 31 87 89 77 7C DC F9 3D 08 17 A3 
55 17 14 6E 22 B2 96 35 43 AE B3 19 4D 28 F2 92 
4F 70 AB 11 D6 B6 54 34 48 29 D3 12 84 6C 4D 0B 
11 44 A0 78 C1 DB E8 76 A9 44 C2 8A 8B 9C 97 39 
5A E7 B0 4F B4 21 28 2D 6D E4 0A 84 95 2A D7 AE 
D6 E2 67 D0 19 D6 6F 69 5C BC 8B 27 64 BD C4 55 
FD 75 4E 9A 53 A8 1B 70 4C 58 0B 10 94 E7 19 B6 
B4 AC 54 DC C6 B9 AD 13 52 26 A2 0C DC C9 F1 1B 
DE 28 5E 20 AD 56 41 64 B1 A4 C4 20 ->FF FD C4 00<- 
87 56 55 77 77 77 77 66 66 66 55 B6 46 D2 24 00 
00 00 00 00 00 00 EA AA A5 AA AA BA
```

I have also seen services with data_alignment_indicator set to "0" when data is actually aligned and also seen data_alignment_indicator set to "1" when data is not aligned. It seems not like like a good idea to blindly trust the data_alignment_indicator.

lelegard, 2019-08-29:
Hi @bobben1no,

What is the problem exactly? What kind of audio is this? The plugin `pes` displays audio and video attributes on some formats but not all. The code was written quite a while ago and new formats have emerged since then, starting with HEVC. Recent audio formats are not analyzed.

bobben1no, 2019-08-30:
Hi !  

I am aware that recent formats like AAC,HE-AAC,HEVC is not supported.

The audio attributes not found  problem is for MPEG-1  Layer II audio (ISO/IEC   11172-3) where syncword in the audio header is not aligned with the first byte of the PES payload.  

It is absolutely not a huge problem that needs urgent fix. But if its easy to solve I hope it can solved be sometime in the future  :)

I've been using pes plugin and other tsduck resources to monitor and report conditions on the streams.

bobben1no, 2019-08-30:
I found some additional information. I can see by logging TS packets in hex format that sometimes there are no syncword /audio header in the TS packet where there is a PES start code. I have done a lot more attempts on getting audio attributes now and I can actually see that pes plugin display audio attributes some times while not at other times.  It may be that the audio attributes is not shown if there is no syncword/audio header in the ts packet with PES start code.

Example TS packet with PES start and following ts packets:
(syncword/audio header marked with <- -> arrows)

```
47 4C 3E 1A 00 00 01 C0 18 6A 80 80 05 23 33 C3 
1B 75 CF 3C 73 C7 2D 32 CF 49 CB EA BD 6F 6F D1 
3B 8C CA EB F3 96 F4 84 4E AE 33 2F 69 A9 79 AD 
8A 6B 5E E5 C7 7A 6D B8 1E 82 27 9C E8 DD C1 B0 
94 81 25 07 63 76 56 96 7A 6A C6 5F 20 59 7C B2 
51 89 16 2E 60 2B 16 4B 83 01 BD 33 85 AE 5E F8 
71 3C 33 0A 85 07 12 A8 3B 55 17 D5 45 E9 A6 7E 
89 A1 51 9B 13 6A D8 42 52 8C 95 18 B2 51 55 85 
68 74 B6 5B 87 DA A9 61 3F B6 1B 07 D5 E1 09 B7 
3E 24 F5 AB 7E DD 75 C5 14 FE 0B 8F 32 B7 85 9B 
62 1C BB 26 CB 81 82 E0 7E B6 DD AD C7 E1 56 08 
D5 83 AF 07 DD 70 B2 63 5B 99 A9 86 

47 0C 3E 1B 12 71 2A EE A7 1B 45 69 B8 D3 A5 46 
52 60 40 EC 4D 4B 11 D5 63 3E 34 C6 B9 1A B1 31 
86 A5 75 D5 AA 2D B5 46 A9 4D A9 A3 4E 29 13 48 
07 54 02 05 64 6A C9 24 9D 45 5D B5 49 43 78 57 
75 9C EA 6A C0 B2 6C 76 3A D6 D6 BB D4 E1 EB AB 
CB 00 9D 33 31 3D 6B 2F 62 86 95 E4 51 C3 2E 86 
09 F1 AA 7C 2C 62 F9 96 DE A9 6F 72 5B DB 76 06 
DB 89 F7 4C B0 B4 D4 5C 48 4A 85 A4 C6 E9 AA FA 
BE BB 54 7D 0E F7 2E 9D D0 07 2C E8 37 45 94 B3 
DC A1 E9 27 28 72 14 78 F0 E0 C0 54 BB 17 3E C9 
DF 5E F7 11 52 92 D3 AC C6 62 4C CC BA EB 75 B5 
59 C6 8B D7 EC 8E 6A D9 61 7E 5D 4C 

47 0C 3E 1C A6 D8 70 9A D7 83 C5 46 B2 A6 76 F3 
9C 26 F7 31 BF 35 A7 75 2E CB 49 82 4A 6B 91 3E 
B4 0E 6C F4 DA 18 F9 66 6E 69 DA 4B 5A 96 39 51 
AA A1 28 7B 86 BA A9 9B 94 7A A1 56 C5 AE 3B 1D 
AD 6D AD 6C A9 F3 6E 2D F5 49 E6 52 19 21 93 B8 
ED 7D A7 1C 18 99 5F C6 3C 39 51 0A 38 A7 8D EB 
63 7A E6 5F B9 79 41 D4 83 EC 98 D9 50 75 07 50 
49 93 98 74 22 60 DB 36 CD 6A B7 0E 04 2E 76 76 
33 81 77 AF 13 EA CA A2 F9 E7 57 5D 95 1F BB 51 
01 CA A2 D9 74 B6 9D C9 F8 BE 7A 44 9B 45 16 12 
AA 5D 34 B9 68 A3 11 53 98 39 96 A8 6A 84 BE A9 
61 E7 36 AD 56 E1 C1 A3 3A 14 19 55 

47 0C 3E 1D CC 29 11 94 CC 0A 98 4A AC 8D 32 BE 
5D 8C D2 2C 65 4C 79 14 1A 35 A3 85 29 9C BE 6A 
70 9A AB F1 97 EB 5D 48 BE F0 89 D9 9C 48 B5 8A 
76 EA 9D C0 6B 8C 6A 4D B5 1A B7 D3 8A 30 44 36 
F3 7B 55 E2 B5 C7 28 B1 64 4B 6A B2 8D 75 C4 28 
5D 61 F7 97 9E AC 6C C7 9C 16 C7 15 EF A2 23 44 
C6 12 8C 27 28 59 86 97 B1 8C 27 02 DE 23 07 78 
C6 63 7C 94 84 A8 52 A4 B8 5B B9 4F 4A 53 19 11 
15 B1 45 C4 FD 4C 99 49 E7 4D 55 52 D5 48 A6 54 
89 34 64 1A 7D 84 9F 43 D1 A7 93 5A 24 B4 4A 85 
97 59 75 96 69 48 06 11 C6 71 9A 6D 37 CF 9A A0 
->FF FD C4 00<- 87 55 55 87 77 77 77 66
 
47 0C 3E 1E 66 55 55 92 D8 D3 48 00 00 00 00 00 
00 00 AA AA AA AA AA AA FA A6 8E 3B 2B AE BA EB 
AE BB AE BA EC 30 BE FC 71 C7 1C F3 D3 0D 6F CB 
2C F2 D3 2C A3 B9 39 D5 8E 74 5F 8F 51 B3 51 D0 
D0 57 3C 15 CA 69 A8 A9 1A 68 32 99 CD 9F 57 26 
D9 DC 11 87 14 61 1F 26 3E CD AE 33 1C 67 98 79 
9C 48 C3 74 55 66 B2 A9 4D 56 DA 6E 0B 68 52 BE 
9D EC D2 82 C3 B8 DC F3 83 73 8D 38 65 49 EE 31 
D6 71 2B 0E D6 C4 16 2C BC 14 AD 0D 50 18 D3 E7 
36 87 B1 4E 6D 64 8B 4D 1D 51 D5 11 69 16 75 7C 
BC D1 67 5C B5 A6 EE 3B 49 25 40 41 65 4E B6 D5 
A6 61 09 5D B6 42 F4 0C E4 E7 F0 38
```