tsduck/tsduck #407: HLS output stops after 24h

premultiply, 2019-08-29:
I try to generate a long-run HLS output with this command but I found tsp silently exitting after 24 hours of flawless operation.
`/usr/bin/tsp --timed-log -I http 'http://127.0.0.1/$test' -P filter -n -p 0x1FFF -O hls --live 6 --duration 10 --playlist /run/nginx/temp/hls/test.m3u8 /run/nginx/temp/hls/test-.ts`

Is this a bug in the HLS output module?

lelegard, 2019-08-29:
Hi @premultiply,

I do not see anything that would cause the output to stop after 24 hours.

Is this reproduceable? Does the session last _exactly_ 24 hours? Or which duration?

You mention that `tsp` exits without error. This is usually what happens when the input reaches an end-of-file. You use the `http` input plugin. So, you had a permanently opened HTTP connection during 24 hours. Is this possible that the HTTP server places some timeout on connections and closes a connection after 24 hours? You should try something like `time tsp -I http ... -O drop` and let it run in parallel with your other command. With this one, `tsp` can only terminate on end of input. Then, you could see if the connection breaks also and after how long.

premultiply, 2019-08-30:
Hi @lelegard,
thank you! I will try to verify if it may be a problem of the HTTP source or input plugin first.

If it is the source it would be helpfull to get a log from `tsp` message for the break reason.

closed 2019-11-14 by lelegard