tsduck/tsduck #408: tsp does not recover on http synchronization loss

zebr, 2019-08-29:
Hello Everyone.
I'm using freshly built tsduck 3.19-1392 on Ubuntu 18.04 trying to grab a http live ts stream on input and produce a simple udp output
```
tsp \
-I http --ignore-errors --infinite --reconnect-delay 100 http://host:port \
-O ip 225.0.0.126:15126 -l 172.16.16.5 
```
This typically work for a minute or two followed by next message:
```
* Error: http: synchronization lost after 566,590 packets, got 0x1F instead of 0x47
* Warning: http: MIME type is application/octet-stream, maybe not a valid transport stream
```

According to doc --ignore-errors together with -- infinite should actually do the trick and recover on error but it does not.
At the same time an instance of TSReader on windows machine analyzes the same http input and does not report any errors.

What am I doing wrong?

lelegard, 2019-08-29:
Hi @zebr,

Options `--ignore-errors --infinite` should disconnect and reconnect, not recovering the current stream. And HTTP uses TCP/IP, so there should be no loss of bytes.

Can you reproduce the issue? Perform a Wireshark capture and analyze the dialog. Just to see what is actually sent by the server?

lars18th, 2019-08-30:
Hi,

To indentify and solve this problem, I have these comments:

- It would be interesting to have some debug code that prints the buffer content when the synchronization fails. You can then identify the problem if the buffer contains some HTTP headers.

- Perhaps some `--resync` option for the HTTP input plugin might also be interesting (based on the code from the `tsresync` tool).

I hope it helps.
Regards.

zebr, 2019-08-30:
Guys, thanks a lot for answering this.
It appeared that I had an issue with one of the network interfaces. Just that simple.

closed 2019-09-23 by lelegard