tsduck/tsduck #409: IDEA: External processor for TABLES plugin

lars18th, 2019-08-30:
Hi @lelegard ,

In order to be able to perform some tasks (such as those described in #355 and #323) I suggest this improvement in the `tables` plugin:

- Add a new parameter to call an external tool (a script, a binary program or an XSLT parser) that will process the output XML file.

This call can be blocking (preferable) or non-blocking (if the user request it). The idea is to use it in this way: after a new table arrives, then it is written and the transform tool is called. After that, one `inject` plugin can replace it in the stream when the file is written.

What you think?
Regards.

lars18th, 2019-09-02:
Hi @lelegard ,

As a simplification, the idea can be reformulated as: "A method to filter a table using the SED command over the XML representation".

I hope you can see the utility.
Regards.

lars18th, 2019-09-06:
Hi @lelegard ,

This idea has sense (use the SED command to change the XML representation of the table on-the-fly), or not?

Regards.

lelegard, 2019-09-06:
SED is probably not a good option for XML because this is a line-oriented editor. Calling a general command is probably better without adding any complexity and allow the usage of an XSLT processor.

Making it asynchronous is a bit more complicated, but just a matter of development time.