tsduck/tsduck #410: BUG? boostpid and psimerge

lars18th, 2019-08-30:
Hi @lelegard ,

I'm not sure if this is a bug or not. However I document it:

```
tsp -I file --interleave=1 --label-base 2 --first-terminate file1.ts file2.ts \
 -P boostpid 18 1 1 \
 -P psimerge --main-label 3 --merge-label 2 \
 -O file test-eit.ts
```

Then if you execute 
```
tsp --debug=3 -I file test-eit.ts -P tables --all-once --include-next -p 18 -m -b tables-eit -O drop
```
You'll see that the tables are _corrupted_ (in fact, some parts are missing). But if you remove the `boostpid` plugin in the first process then all goes right.

Why this behaviour?

lelegard, 2019-08-30:
I think the reason is the following. After reading interleaved files, the resulting TS is "invalid" because PID 18 (and all other PSI PID's) is made of a mixture of unordered packets from two streams. Specifically, continuity counters (CC) are not ordered when adjacent packets on the PID's come from distinct files. The plugin `boostpid` adds empty packets in PID 18 with CC identical to the previous packet (empty packets do no increment the CC). So, PID 18 now contains packets with label 2 (originally from file1), label 3 (file2) or no label (added by `boostpid`).

Now `psimerge` only consider packets with labels 2 and 3. It recreates a correct PID 18 with correct CC's over all packets with those labels. But packets without labels remain unmodified. These packets are empty on PID 18 and are arbitrarily interleaved with packets from the reconstructed PID 18. Since the empty packets have CC which are no longer synchronised with the CC in the new PID 18, every empty packet from `boostpid` now creates a discontinuity in PID 18. And when a discontinuity occurs in the middle of a section, the section is supposed corrupted and dropped.

So, in theory, any modification in the intermediate invalid TS after reading interleaved files but before `psimerge` creates a mess in the TS. This should be avoided.

Now, in practice, with this specific case, after clearly understanding what happens, adding `-P continuity --fix --pid 18` after `psimerge` should fix the problem.

lars18th, 2019-08-30:
Hi @lelegard ,

Thank you for the response.

> Now, in practice, with this specific case, after clearly understanding what happens, adding `-P continuity --fix --pid 18` after `psimerge` should fix the problem.

After reading your explanation.. Sorry, but I don't think so! The problem is _before_ the `psimerge` action, so this `continuity` processing after I feel it will don't solve the problem.

Based on you description, perhaps this can solve the problem:
```
tsp -I file --interleave=1 --label-base 2 --first-terminate file1.ts file2.ts \
 -P --only-label 2 boostpid 18 1 1 \
 -P --only-label 3 boostpid 18 1 1 \
 -P psimerge --main-label 3 --merge-label 2 \
 -O file test-eit.ts
```
Because it works at _source_ level. Then the `psimerge` will read correct input packets.

What you think?

lelegard, 2019-08-30:
> After reading your explanation... Sorry, but I don't think so! The problem is before the psimerge action, so this continuity processing after, I feel it will don't solve the problem.

It is true that boostpid can steal null packets from either label 2 or 3. So, yes you are right, in the mixed stream you may have situations such as packet in PID 18 with label 2, followed by a null packet with label 3 which is stolen and marked as PID 18 with a CC based on the last packet with label 2. But the stolen package will retain label 3. So, yes, the corruption can occur before psimerge and psimerge will not extract correct PSI.

Your solution should work. Try it.

lars18th, 2019-09-13:
Hi @lelegard ,

Before closing this issue, I suggest to document this potential error (caused by the user, not the library). Perhaps you want to include this message for the `psimerge` plugin:

"If you mix one instance of the `boostpid` plugin with the `psimerge` pluging, then apply it with the corresponding label. Example `-P --only-label XX boostpid`. And use two instances for the two labels (main-label and merge-label)."

I hope you agree.
Regards.

closed 2019-11-14 by lelegard