tsduck/tsduck #412: HiDes UT-100C: Windows problem

lars18th, 2019-09-03:
Hi @lelegard ,

Using the HiDes UT-100C modulator in Windows I have this problem:
```
tsp --debug=2 -I ip 239.0.0.1:10000 -O hides -f 474000000 -g "1/32" -h "3/4" 2>&1 | grep
 hides
```
In the debug I see time to time, not 384 aligned packet writes:
```
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 5828, packets: 31
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 5828
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
```
Please, not the _three_ messages "after", without any other similar "calling" before.
Futhermore, when these messages appear I get in the analyzer PCR Accurate Errors. So I suspect something is wrong with the driver when filling the internal device buffer.

Thinking for a workaround, I ask if some parameter like `--force-aligned-writes` can solve the problem. The idea is to write _only_ 348 chunks, so if the data is smaller, then you wait until the buffer is ful.

What you think?

lars18th, 2019-09-03:
Just another example:
```
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 37788, packets: 201
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 37788
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
* Debug[2]: hides: HiDesDevice: calling IOCTL_IT95X_SEND_TS_DATA, size = 65424, packets: 348
* Debug[2]: hides: HiDesDevice: after IOCTL_IT95X_SEND_TS_DATA, size = 65424
```

lelegard, 2019-09-06:
Hi @lars18th 

The lack of log messages is not significant. As I already explained in some past issue, tsp does not enforce the log messages to avoid the common pitfall of some applications which want to log absolutely _every_ message and for which the abundance of messages creates delays which in turn create errors which in turn add error messages, etc. until the application crashes.

So, tsp logs messages from a low priority thread and the input queue of this thread has a limited size. When a message cannot be enqueued (typically because the queue is full), then the message is simply dropped. In other words, the actual work to be done takes precedence over messages describing the work. In corporate parlance, we would say that operations take precedence over marketing :)

When you activate a lot of debug messages, typically for each packet write operation, some random messages will be dropped. If you want all messages to be displayed, then use `tsp --synchronous-log`. But the price to pay is that application threads may wait when the log queue is full, which may create problems.

So, if you want full and accurate debug control, you may get functional problems. And if you want operational reliability, you lose visibility on what is going on. This is Heisenberg principle, applied to computing.

In practice, just be aware of this. If you need accurate debugging, use `--synchronous-log` but take care to segment the problem you  are investigating with problems which are incurred by `--synchronous-log`.

lars18th, 2019-09-06:
Hi @lelegard ,

Thank you for the explanation! Now, I understand the reason for the lack of "calling" messages.

However, regarding the _glitch_ in the broadcasted TS, I repeat the question: It will be possible to write **only** 348 chunks to the hides driver? I think this "simple" solution can solve the problem.

What you think?
Regards.

lelegard, 2019-09-06:
The output plugin never writes more than 348 packets, but sometimes less. Writing only 348 packets at a time is a problem because the output plugin has no reason to be called with a multiple of 348 packets. It loops on chunks of 348 packets and then write the shorter end. Writing only 348 packets would need extra buffering of all packets since the end of the first write operation would cause a permanent buffer shift.

HiDes software is very low quality software, drivers included, as I already had to opportunity to comment here. I do not have the source code for the Windows driver. But we have the one for the Linux one. The driver has an internal buffer of 348 packets and, when full, it reports an error and forces to application to do _active wait_ or polling. Insane... So, we write only chunks of 348 packets. But, at least on Linux, the driver does not wait for more packets if the buffer is not full. It just writes what is inside it.

So, I do not think that the glitches are due to the shorter-than-348 writes.

lars18th, 2019-09-06:
Hi @lelegard ,

> Writing only 348 packets at a time is a problem because the output plugin has no reason to be called with a multiple of 348 packets.

Really? Think on this:

- Sending to a modulator never ends, as the TS is infinite. So shorter writes can be handled with an intermediate buffer.
- And you don't need to do a buffer shift, you only need to add the intermediate 384 packet buffer. And when this buffer is full, then send it to the driver.

> HiDes software is very low quality software, drivers included

I completely agree!

> So, we write only chunks of 348 packets. But, at least on Linux, the driver does not wait for more packets if the buffer is not full. It just writes what is inside it.

I'm not sure, but I feel (almost in Windows) that the problem can be this:

- When you write less than 348 packets, the _driver_ waits until the buffer is full.
- But when the _device_ buffer is empty, it adds NULL stuff.
- So, when you write the reamining packets to complete the 384 number, if the _device_ has inserted the NULL stuff, then you're sending data over the bitrate. And as a consequence the driver will force you to wait as the _device_ refuses to receive more data.

Or making a simplification: It's possible that the implemented transfer protocol (over the USB wire) is using a pseudo-synchronous transfer mode with a fixed 384 packet size.

So a workaround could be then to write to the driver using the fixed size of the device buffer.
And I know, this is a simple hypothesis, but it makes sense. Don't you think?

Regards.

lelegard, 2019-09-06:
You could verify the hypothesis of the null packet insertion using a DVB-T tuner. Just compare what you send with what you capture. Two instances of `tsp` on the same machine.