tsduck/tsduck #413: Suggestion: File write with read support in Windows

lars18th, 2019-09-09:
Hi @lelegard ,

On Windows, when running the `tsp` process and writing to a file (`-O file out.ts`), the `out.ts` file is locked for reading. However, other programs that are appending to files do not suffer this behavior.

Any solution not to block the file?
Thank you for this project.

lelegard, 2019-09-10:
Fixed. I have added "SHARE_READ" option on Windows to output TS files (apply to plugins file (output and packet), t2mi, hls and command tsstuff). I just hope that the corresponding file system locking activity does not degrade the performances.

There is no sharing issue on Unix (Linux and macOS).

lars18th, 2019-09-10:
> Fixed. I have added "SHARE_READ" option on Windows.

Thank you! Waiting to test it (I hope you add some pre-compiled binary, as I can't compile in Windows).

Regards.

lelegard, 2019-09-10:
> I hope you add some pre-compiled binary, as I can't compile in Windows

See here: https://tsduck.io/download/prerelease/

But, since you are a developer as well as a user, you should install a build environment on Windows. [It's easy, it's free](https://tsduck.io/doxy/building.html). Then, just double-click on a PowerShell script to rebuild the binaries and installers at the same time. This is just what I did to build those binaries (except that it was on a Windows VM running on my Mac).

lars18th, 2019-09-11:
Thank you @lelegard ,

I'll try it (both, the prerelease and the Windows build environment). :wink: 
Regards.

lars18th, 2019-09-11:
Hi @lelegard ,

The "SHARED_READ" mode in Windows works like a charm!
Thanks a lot! :+1:

closed 2019-09-11 by lars18th