tsduck/tsduck #415: DTA 115A

michaeladams123, 2019-09-10:
Hi all,

kindly note that when I applied to run one channel HD from udp on DTA 115 rev3 i got this error

* dektec: Got DMA underflow.
* dektec: Got FIFO underflow.

tsp -v -I ip 232.1.10.210:1234  -O dektec --device 0 --modulation DVB-T --frequency 277500000 --bandwidth 7 -g 1/32 -r 7/8
 and no signal found please advise where is the issue please ?

note:  the channel bit rate about 8 meaga bit maximum

lelegard, 2019-09-10:
When the source is real time (ie. not a file for instance), a modulator requires a stable real time stream with exactly the right expected bitrate. 
~~~
$ tsterinfo --bandwidth 7 -g 1/32 --high 7/8
Transport stream bitrate: 27,709,893 b/s
~~~
Unless you have a specific IP source, it is unlikely that your IP source has _exactly_ the right bitrate.

You may add option `--stuffing` to compensate underflow.

michaeladams123, 2019-09-30:
Hello,
I added the option --stuffing but still same issue