tsduck/tsduck #418: BUG? Continuity with PSI tables

lars18th, 2019-09-10:
Hi @lelegard ,

I do this test (only for fun, it doesn't have sense... And I know it, we can use two files with one `inject` for the same pid):
```
 -P filter  -p   20-21 -n -s \
 -P inject  -p   20       --binary tot-tdt_p0014_t70.bin=30000 \
 -P inject  -p   21       --binary tot-tdt_p0014_t73.bin=30000 \
 -P remap   -u  -n 21=20 \
 -P continuity  -f -p 20 \
```
And after the `continuity` the CC counters aren't updated! The sequence is like 1,1,2,2,3,3,4,4... where it's expected to be 1,2,3,4,5,6,7,8...

Can you verify it, please?
Regards.

lelegard, 2019-09-10:
~~~
  -u
  --unchecked
      Do not perform any consistency checking while remapping PID's; remapping
      two PID's to the same PID or to a PID which is already present in the
      input is accepted. Note that this option should be used with care since
      the resulting stream can be illegal or inconsistent.
~~~
You had been warned !

Remapping a PID changes the PID of the packets (and updates the corresponding PSI/SI). Mixing two existing independent PID's produces an invalid ES. Having inconsistent CC is a natural consequence.

lars18th, 2019-09-10:
> You had been warned !
> 
> Remapping a PID changes the PID of the packets (and updates the corresponding PSI/SI). Mixing two existing independent PID's produces an invalid ES. Having inconsistent CC is a natural consequence.

Sorry? I pointed that the bug is inside the `continuity` plugin, not in `remap` !!! :fearful: 

Please, check the example: I'm injecting a TOT binary table and a TDT binary table. So, imposible to produce any invalid ES. And I know (and I commented it before), the correct way to do it is: `-P inject -p 20 -s --binary tot-tdt_p0014_t70.bin=30000 --binary tot-tdt_p0014_t73.bin=30000`. However, the example is just an _example_. It doesn't have sense, but it illustrates the BUG prsent inside the `continuity` plugin.

Let me to know if this is false _in the example_:
1. The first `filter` removes all packets with pid 20 & 21.
2. The first `inject` pushes the TDT in the pid 20.
3. The second `inject` pushes the TOT in the pid 21.
4. The `remap` plugin, as it's using `-u -n`, then it _only_ changes the pid number (from 21 to 20) without touching anything else.
5. And the problem is then with the `continuity` plugin, as instead of recalculating the CC values, it leaves them. HERE IS THE BUG!

Am I wrong?

lelegard, 2019-09-10:
Ooops, you're right, sorry, I did not pay attention enough. In a hurry, I just saw the mixture of PID's and did not notice the continuity plugin. I will have a look.

BTW, you probably already noticed that you can inject the two files, TDT and TOT, in the same PID, in the same inject plugin. But you mentioned the test did not have sense anyway.

lelegard, 2019-09-10:
It's the same as #267 because the two PID's have exactly identical CC, packet by packet. So each second packet is incorrectly interpreted as a duplicate. This is still to be fixed.

lars18th, 2019-09-10:
OK, so I continue in #267 

Regards.

closed 2019-09-10 by lars18th