tsduck/tsduck #419: Plugin to remove TS header (with AP)

dukey914, 2019-09-10:
Hello. I would like to ask you, if is possible to add some little plugin for removing TS header (47 xx xx xx) and AP, if it is present. Just keep only payload (184 bytes without AP and 184-AP bytes with AP).
Something like that:

tsp -I file "input.ts" -P filter -p 100 -P header --remove-ts-header --remove-ap -O file "output.ts"

And if it can be done with "ip" output, it would be awesome too!

Thank you so so much for your awesome software!!! It's really powerfull tool!!!
Have a nice day!

lelegard, 2019-09-10:
The principle of a plugin is to process TS packets. It does not process data. What would be the output of such a plugin? It must be used as input of the next plugin, which only accepts packets as input.

If what you want is a demux / depacketizer, then a specific command should be used. The inverse of `tspacketize`, sort of, because `tspacketize` only processes sections. Such a depacketizer should process / extract only one PID. Otherwise the result would be meaningless. Could be trivial to implement.

BTW, what do you mean by "AP" ? Adaptation Field ?

dukey914, 2019-09-10:
Sorry for that. I missed that. Yes, yes, yes! I meant AF. Adaptation Field. Not AP... Sorry.

dukey914, 2019-09-10:
I believe it is not forbidden, but I found some data broadcasting on satellite and in this data PID is hidden transport stream. It's packetsized (probably) to one PID and only thing what I have to do to get that hidden transport stream is just remove TS header. Sometimes, they're broadcast AF, so if I remove only TS header, after some period the hidder transport stream will drop due to extra data on his payload.
Same thing can be done to catch data from satellite (download firmware, etc.)

lelegard, 2019-09-10:
T2MI maybe? See [an example here](https://tsduck.io/streams/?name=russia-t2mi-scte35).

dukey914, 2019-09-10:
It looks more simple than T2-MI. Just transport stream in transport stream. Nothing about PLP.

lelegard, 2019-09-10:
Someone playing with TSDuck plugin `encap` then ?
Try plugin `decap` just to see if it can get some inner TS out of the PID.

dukey914, 2019-09-10:
OK. I will try.

dukey914, 2019-09-10:
So decap is not working for this case.
I made a little screenshot of that header, which have to be removed. Hope this will help!
https://i.postimg.cc/YMwLDsQQ/01.png (see it in full resolution)

lars18th, 2019-09-11:
Hi,

My two cents: TS (not mpeg-TS, but a bitstream transport stream) inside a PID is a common use case in other scenarios. For example, when encapsulating DAB ensembles (radio feeds) inside a MPEG-TS for a satellite link. Typically, such encapsulations can be: trivial or complex. The trivial case is quite simple: a 4 bytes header plus 184 bytes of payload. The header only adds the 0x47 byte and the PID number plus the CC counter. Nothing more. And the payload is a simple chunk copy of the original bitstream.

This trivial encapsulation could be implemented in TSDuck. My suggestion is a new OUTPUT plugin that implements this simple "cut&copy". Then with a simple pipe you can get the encapsulated bitstream. Example: `tsp -I dvb ... -P filter -p 1000 -O decap --mode 1 | my-processor`. The `--mode 1` represents this simple 4bytes header. And in the future some other encapsulations can be implemented.

The key point here is that such plugin is an OUTPUT version, so the output can be anything different to a mpeg-TS.

What you think @lelegard ?

lars18th, 2019-09-11:
Hi,

I just forget to mention that another simple encapsulation mode exists: The simple intercalling of the 0x47 TS-Sync byte every 187 bytes of the original bitstream. And another futher simple mode is to use a complete TS header of 16 bytes (including PCR timestamps, but the PES is the payload of the encapsulated bitstream).

So perhaps the `--mode 1` encapsulation can use the option `--offset` to mark the start of the payload: 1 for the 187 byte size, 4 for the 184 byte size, and 16 for the 172 byte size.

I hope it helps.
Regards.

dukey914, 2019-09-11:
@lelegard It's from 30W, where are distribution streams for Spanish DTT after decoding process. 11222 H 30000 for example.

lelegard, 2019-09-11:
I do not have a dish to this one. DTT feeds and any DTO feed in general may use any proprietary encapsulation. So, it is not easy to guess what it is.

If you can, you could capture 60 seconds of the complete TS and post it somewhere. If this is an interesting or unique structure, I will publish it in the TSDuck sample stream repository.

lelegard, 2019-09-13:
Interesting. The ECM's are from EBU but the EMM's are from Scientific Atlanta (then Cisco, then Synamedia). How is this organized?

dukey914, 2019-09-13:
I don't know for what reason they're broadcasting it on this TP. Maybe they
were using EMM in the past? It's hard to say... Originaly, it's really
encrypted with one constant CW key and (I believe) it looks like it's not
the case of temporally frozen CW.

(I don't know, if you will receive this reply, because I'm sending it
throught my e-mail. For the first time).

lelegard, 2019-09-13:
So, where did you get the constant CW from ?

dukey914, 2019-09-13:
From satellite forum. You can simply google it as "Abert*s 30W BI*S Key".
"Ab*rtis" is probably the name of DTT provider.
I really believe that it's not forbidden to do that. I believe that we can
do it for education. The stream is originally encrypted and if it cannot be
add to TSDuck because of encrypted signal, I will fully understand that
decision.

lelegard, 2019-09-15:
Let me rephrase this: You really believe that it is not forbidden to decrypt an encrypted feed using an encryption key you retrieved from a public forum online...

Let's transpose that to the real world: On a public street market, people sell or distribute copies of keys for various cars which are parked on the street. You retrieve one of those keys, unlock a car, start it and go away with it. Would that be forbidden?

I agree that a DTTV feed probably carries channels which are otherwise broadcast on the clear on DVB-T and using a _fixed_ DVB CSA control word is probably the most stupid thing to do in terms of security. But, when there is no - or close to zero - security, does it mean that you are allowed to trespass? When there is a sticker "private property, do not enter" on a house but no locked gate, does this mean that you are allowed to enter and settle down?

dukey914, 2019-09-15:
You are right. All of this are bad things. Include decrypting encrypted
things (signals). If they're using weak encryption, it's still encrypted
and not allowed to public.
I'm really sorry for that. I just wanted to try to ask, if it is possible
to add this function, because if I won't ask, I will never find the answer.
Now I know it's not possible, because someone can utilize this function for
bussines.
TSDuck is still very usefull tool for MPEG-TS and I want to leave it as it
is - awesome and free!!
I think now you can close this issue. :)
Have a nice days @lelegard !!!
...and sorry for my bad English.

closed 2019-09-15 by lelegard