tsduck/tsduck #420: TSDuck Streams Repository: Modulation parameters?

lars18th, 2019-09-11:
Hi @lelegard ,

I discovered the TSDuck Streams Repository at https://tsduck.io/streams and I want to check some captures broadcasting them with my UT-100C modulator (for searching errors). However, in the capture files I don't found the ORIGINAL MODULATION mode.

For example, for the France TNT (https://tsduck.io/streams/?name=france-dttv) in the files "*.analyze.txt" is not printed any information about the modulation. Futhermore the Transport stream bitrate differs between different files, and the values aren't consitent with the regular DVB-T modulation parameters.

Any option to include such information?
Or to enhance the `analyze` plugin to print the info inside the NIT table?

Thank you for this amazing project!
Regards.

lelegard, 2019-09-11:
The `analyze` plugin works on the TS that is passing through it. It does not know anything about which input plugin was used, even less which modulation parameters were used 

The bitrates from the analysis file come from the PCR analysis. The value is maybe not _exactly_ the right one, but very close to it. Use the command `tsterinfo` to list the possible combinations of DVB-T modulation options for a given bitrate.

You may find the modulation parameters in the PSI file, in the NIT. But note that the French DTTV network,, which is under control of the national regulator, does not manage PSI/SI transformation in the local transmitters. So, in this specific network, the NIT is the same over all the national territory and it cannot carry any valid local frequency information. I am not even sure that the other modulation parameters are identical on all transmitters for a given TS. I was involved a lot in the deployment of this network around 10 years ago but it has changed  lot since then.

lars18th, 2019-09-11:
Hi @lelegard ,

Thank you for the explanation. However, please consider one (or more) of these suggestions:

1. Put some _manual_ info inside the file `scan.sh` to indicate (with some comments) the modulation parameters used.
2. Add the option to print the modulation parameters of the NIT table when using the `analyze` plugin.
3. Include some `tsterinfo` use inside the `capture.sh`.
4. Modify the plugin `dvb` to print the modulation parameters and write them to a text file. Then use this functionality inside the `capture.sh` script.

In any case: Do you know the exact values? The problem is that similar different modulation values are in the range of the 22-24Mbps, and I can't derive the exact value from the TS examples.

Regards.

lelegard, 2019-09-11:
> Put some manual info inside the file scan.sh to indicate (with some comments) the modulation parameters used.

In the general case, modulation parameters are in the capture script. Otherwise, you cannot tune.

The specific magic with DVB-T is that, unlike DVB-S, you can tune on the frequency without specifying the modulation parameters. Just keep the default parameters (or any other value), and the tuner will tune anyway.

Now, how to find the appropriate the right parameters afterward? Typically because you want to remodulate the output of the chain and you definitely need the parameters in order to generate a carrier.

In the best case, the tuner hardware device driver can report the actual parameters and the O/S tuning framework has the API to report it. Fine, just run `tslsdvb -v`  from another terminal while the capture is running and you get all _actual_ tuning parameters. This is (was) what happened on Linux with the tuners I used around 2007. 

Now, with the current "modern" DVB-T tuners on Linux, it seems that the devices I now use no longer report the actual parameters. As usual, "modern" is a synonym for "less reliable".

On Windows, DirectShow never reported correctly the actual modulation parameters. The same tuner which correctly reports its parameters on Linux no longer does on Windows. So, on Windows, it has never been possible to report the actual parameters. As usual, "Windows" is a synonym for "less reliable".

So, to solve that issue, I developed `tsterinfo` which can convert back and forth between modulation parameters, bitrates, frequencies, channels, etc.

As an example, use the analysis file [here](https://tsduck.io/streams/france-dttv/tnt-uhf22-482MHz-2019-01-22.analyze.txt). You can see that the estimated bitrate based on PCR's is 24,146,440 b/s. 

`tsterinfo` will give you all possible combinations of DVB-T modulation parameters which produce that bitrate or an approaching one:
~~~
$ tsterinfo --bitrate 24,146,440
  Nominal bitrate .......... 24,128,342 b/s
  Bitrate difference ........... 18,098 b/s
  Bandwidth ..................... 8-MHz
  FEC (high priority) ............. 2/3
  Constellation ................ 64-QAM
  Guard interval ................. 1/32
$
~~~

Here, there is only one possible combination. There are a few cases which give two or more combinations. But if your goal is to remodulate the output with the same bitrate, you can use any of them. You do not need the exact same parameters as the input, as long as the bitrate is the same.

If you want a full list of bitrates and DVB-T modulation parameters, you may have a look at [this file in the TSDuck test suite](https://github.com/tsduck/tsduck-test/blob/master/reference/test-029.log).

lars18th, 2019-09-11:
Hi @lelegard ,

After downloading some files from the French TNT in your repository, I see that _all_ files have reception errors. Please, can you update these files? It seems to me that the capture process has been executed with a poor signal.

Regards.

lelegard, 2019-09-11:
> Regarding the tsterinfo tool, I know it. But the problem is that it only approximates.

On the contrary, tsterinfo is accurate. The estimated bitrate from the stream is not. And tsterinfo finds the closest value amongst all possible valid DVB-T bitrates.

lars18th, 2019-12-17:
> > Regarding the tsterinfo tool, I know it. But the problem is that it only approximates.
> 
> On the contrary, tsterinfo is accurate. The estimated bitrate from the stream is not. And tsterinfo finds the closest value amongst all possible valid DVB-T bitrates.

Hi @lelegard ,

Regarding this I have a question: If you use the _closest_ value, then how you can use the `pcrbitrate` to add _precise_ marks?

The problem that I see is this: The official DVB-T bitrates aren't the values reported by the `tsterinfo` tool, because the official values aren't whole numbers. Then using a DVB-T modulator the problem can be solved using the nearest lower value for the bitrate, as the modulator will insert one NULL packet when it will be necessary to achieve the target bitrate of the DVB-T modulation. But then the PCR marks require to be adjusted to the _exact_ bitrate and not to the _approximated_ bitrate. If not, when using the closest value a drift will appear over time.

Any idea to overcome this problem?

lelegard, 2019-12-17:
> The official DVB-T bitrates aren't the values reported by the tsterinfo tool, because the official values aren't whole numbers.

That is right. TSDuck precision is limited to one bit per second. This is a known limitation which has been reported a couple of times under different classes of issues. I have had the intention to fix this for a long time but there are implications in the code everywhere. Unlike other improvements I made to TSDuck as iterative cycles of small developments, this is a long work that must be done in one single step because it virtually breaks everything, everywhere. And I do not have the time for that. This is the limitation of a benevolent open source model.

lars18th, 2019-12-17:
OK, @lelegard, understand the limitations.

So, please consider then my original suggestion about to update the repository of the **captured** examples with simple changes. For example with a TXT file indicating the original modulation values of the signal captured. And optionally, if you can, to check too for Continuity Errors in the captured stream and remove files with errors (except for examples of streams _with_ errors).

I hope you can update the French TNT captures in the future.
Regards.

SS-FF, 2020-01-28:
> On Windows, DirectShow never reported correctly the actual modulation parameters. The same tuner which correctly reports its parameters on Linux no longer does on Windows. So, on Windows, it has [...]

This is not quite correct. The Windows TV component model DOES have that feature. It's just a matter of driver implementation whether it is reported or not. No difference to Linux ;-)

lars18th, 2020-01-28:
OK, but my objective is to have only some "clean" Fench TNT (because the author lives in France) captured examples _and_ the bitrate used to broadcast these MPTS. In the current repository the captures have errors and the _exact_ broadcast parameters are unkown.