tsduck/tsduck #421: Detection of SCTE 18 Messages

ltrayanov, 2019-09-11:
Hi, Is it possible to add detection of SCTE 18 messages?
SCTE18 carries emergency alert signaling within atsc streams.  
https://www.scte.org/documents/pdf/Standards/ANSI_SCTE%2018%202007%20J-STD-042A%202007.pdf

I can currently detect the presence of SCTE18 PID's in the stream with tsp -P filter -p 8188 -p 8187
The pids are always either 0x1ffb or 0x1ffc. 
I would like to add the ability to also read/interpret the SCTE18 table, Where can we add the EAS table ID 0xD8 and table format?

ltrayanov, 2019-09-18:
In windows the scte18 packet is received and processed:
* Debug[2]: ip: received UDP packet, source: 69.240.168.186:33174, destination: 232.44.101.47:5050
* Debug: ip: initial buffer load: 1 packets, 188 bytes
* ip: unknown initial input bitrate
* Debug: ip: input thread started
* Debug: tables: packet processing thread started 

However in ubuntu the tsduck doesn't seem to receive the packet. I can see it with tcpdump: 
15:01:57.300767 IP 69.240.168.186.33174 > 232.44.101.47.mmcc: UDP, length 188
15:01:57.312763 IP 69.240.168.186.33174 > 232.44.101.47.mmcc: UDP, length 188
15:01:57.322798 IP 69.240.168.186.44674 > 232.44.101.47.mmcc: UDP, length 188
15:01:57.334608 IP 69.240.168.186.44674 > 232.44.101.47.mmcc: UDP, length 188
15:01:59.091921 IP 69.240.168.186.50376 > 232.40.2.99.mmcc: UDP, length 188
15:01:59.101914 IP 69.240.168.186.58375 > 232.40.2.99.mmcc: UDP, length 188

lelegard, 2019-09-19:
Support for SCTE 18 EAS messages has been added.

It is very strange that you have a different behaviour on Windows and Linux. It is possible to imagine different multicast reception behaviours, due to system configurations. But I would be extremely surprised that the processing of the same stream of received packets would be different. This is pure application code which is identical on all platforms.

Could you first capture a stream on both systems:
~~~
tsp -I ip ... -P until .... -O file out.ts
~~~
Please do that on both systems at the same time. There may be some initial and final desynchronizations but the rest of the file should be identical.

Then, try the command `tstables` (same as `tsp -I file ... -P tables ... -O drop`) on both systems. For a given file, the result shall be _exactly_ identical on Windows and Linux.

ltrayanov, 2019-09-20:
@lelegard Thanks!
I found my issue, I assumed that tsp with ip input runs in promiscuous mode. Once i fixed my network subnets to match. tsp worked as expected.
Is there a way to run tsp in  promiscuous mode?

ltrayanov, 2019-09-20:
Also a small correction is need it the locations, the county codes are wrong. in the raw packets we have **02 2a 0c 5b 2a 0c 65**, where **02** is locations count and **2a 0c 5b** as first location, and **2a 0c 65** as second. 

TSP output is:
_State code: 42, county: 3072, subdivision: All or unspecified (0x0)
 State code: 42, county: 3072, subdivision: All or unspecified (0x0)_

the county codes should be 91 and 101. 

2a is a state code. The next 16 bits are for: 
- subdivisions code (4 bytes) 
- reserved "11" (2bytes) 
- county code (10bytes)

lelegard, 2019-09-20:
> Is there a way to run tsp in promiscuous mode?

No. TSDuck is a true UDP client, it works as any regular client. Promiscuous mode is merely designed to steal traffic from other clients, e.g. for probe tools such as WireShark.

> the county codes are wrong

Sorry, should be fixed now.