tsduck/tsduck #423: DTA 110

michaeladams123, 2019-09-24:
Hi,
 any support please ?
I used DAT 110 DTA

 tsp -v  -I file /root/astra192E-ts1080-2018-05-11.ts -i -O dektec --device 0 --modulation DVB-T --frequency 177500000 --bandwidth 7 -g 1/32 -r 7/8
* file: initial input bitrate is 57,489,093 b/s
* dektec: using Dektec device 0, output channel 0 (DTA-110 port 1)
* dektec: using DVB-T FEC 7/8, bandwidth 7-MHz, constellation 64-QAM, guard 1/32, transmission 8K
* dektec: setting output TS bitrate to 27,709,893 b/s
* Error: dektec: error while setting modulation mode: DTAPI_E_NOT_SUPPORTED (DTAPI status 4119)

michaeladams123, 2019-09-24:
> Hi,
>
> Is it a DTA-110 (cable) or a DTA-110T (terrestrial) ?
> Try tsdektec --all --verbose
>
> DVB-T is supported on the 110T, not on the 110. See the DTA-110 leaflet<https://www.dektec.com/products/PCI/DTA-110/downloads/DTA-110%20Leaflet.pdf>.
>
> I used a DTA-110T quite a lot about 10 years ago and it worked perfectly.

Hi,

I used dta 110 t rev 3

Regards

michaeladams123, 2019-09-25:
* Device 0: DTA-110 (DTA-110T OFDM Modulator/UHF Upconverter)
  Physical ports: 1
  Channels: input: 0, output: 1
  Output 0: DTA-110 port 1, Modulator (UHF-band 400-862MHz, Uni-directional
      output, MPEG-2 transport stream, Modulator output, QAM-A modulation, QAM-C
      modulation, ATSC 8-VSB reception, Lock TS rate to input port)
  Subsystem id: 0xD10A (DTA-110)
  Subsystem vendor id: 0x14B4
  Device id: 0x9056
  Vendor id: 0x10B5
  Serial number: 00000000F5026029
  Firmware version: 3 (0x00000003)
  Firmware variant: 0 (0x00000000)
  PCI bus: 7, slot: 4
  Customer id: 301948
  Engineering change level: Rev 3
  Manufacture id: 03
  Production date: 2006.05
  Part number: DTA-110T
  Serial number: 4110573609
  Crystal stability: RF:1ppm;Sym:25ppm

lelegard, 2019-09-25:
Several tracks:

- Make sure that the application and drivers are consistent (latest versions of TSDuck and latest version of the Dektec drivers). Use `tsversion --version=all`.
- You set the frequency to 177.5 MHz while the supported range of the DTA-110T is displayed as 400-862MHz.
- Does it work with other modulation parameters, specifically 8 MHz bandwidth instead of 7?

michaeladams123, 2019-09-25:
tsp -v  -I file /root/astra192E-ts1080-2018-05-11.ts -i -O dektec --device 0 --modulation DVB-T --frequency 477500000 --bandwidth 7 -g 1/32 -r 7/8

michaeladams123, 2019-09-25:
same issue

lelegard, 2019-09-25:
> same issue

Please be more specific. Did you check the driver version? Did you try 8 MHz bandwidth?

I can just tell you that I used DTA-110T with TSDuck in the past and it worked perfectly. I no longer have any (and no longer have any machine with PCI slots, PCIe is the standard now). I cannot debug your configuration. You need to debug it yourself. Try different configurations. Check which ones work and which ones fail, until you converge to the root cause of your problem. Do not expect any one to tell you all steps, one by one, day after day.

closed 2026-04-24 by github-actions[bot]