tsduck/tsduck #424: DTA 115

michaeladams123, 2019-09-30:
Hi,

Kindly I am using the command as beolow and the cjhannel working for 5 seconds and I got the error  

tsp -v -m -I ip 232.1.10.215:1234 -O dektec --device 0 --modulation DVB-T  --frequency 177500000  --bandwidth 7 -g 1/32 -r 7/8

* dektec: Got FIFO underflow.
* dektec: Got DMA underflow.
* dektec: Got FIFO underflow.
* dektec: Got DMA underflow.
* dektec: Got FIFO underflow.
* dektec: Got DMA underflow.
* dektec: Got FIFO underflow.
Please I need support to fix this issue and I used --stuffing but same issue

pabloko, 2019-10-25:
Please dont duplicate #415 
You should match the bitrate for your dektec output, as author pointed you in the other issue
```
>tsterinfo --frequency 177500000 --bandwidth 7 -g 1/32 --high 7/8
VHF channel: 5, offset: 0
Transport stream bitrate: 27,709,893 b/s
```

using tsterinfo to know the bitrate then craft your ip stream to have stuffing until  27709893 b/s
you should also specify the bitrate so tsp know it in advance `tsp --bitrate 27709893`, if you cant produce a stream with fixed bitrate, use `-I null` then you can use `-P mux`. Also in real time always consider if you need to use `-P regulate` depending of your source.

Heres a working ffmpeg command that you can adapt and use to your needs

``` 
ffmpeg -re -i "http://sourcexxxxxxxx.com/directo/live.m3u8" -c:a mp2 -b:a 128k -c:v mpeg2video -r 25 -g 50 -b 4000k -minrate 4000k -maxrate 5000k -bufsize 1835k -flags cgop+ilme -sc_threshold 1000000000 -vf scale=720
20 -pat_period 0.10 -sdt_period 0.25 -metadata:s:a:0 language=esp -mpegts_flags +pat_pmt_at_frames -metadata service_name="InteralmeriaTV" -metadata service_provider="TL02AL" -mpegts_service_id 0x139D -mpegts_start_pid 0
start_pid 0x190 -mpegts_original_network_id 0x22d4 -mpegts_transport_stream_id 0x01f6 -f mpegts -muxrate 24882353.0 "udp://192.168.1.86:9977?pkt_size=1316&bitrate=24882353"
```
And corresponding tsduck output
```
tsp -v -m --bitrate 24882353 -I ip 0.0.0.0:9977 -P inject NIT.xml=30 --pid 16 --bitrate 300 --stuffing -P inject PAT.xml=20 --pid 0 --replace --stuffing -P inject BAT.xml=50 --pid 17 --replace --stuffing -P inject EIT.xml=10000 EIT_pf.xml=1800 --pid 18 --bitrate 26001 --stuffing -P inject PMT.xml=40 --pid 400 --replace --stuffing -P inject AIT.xml=500 --pid 412 --bitrate 5000 --stuffing -P inject TDT.xml=20 --pid 20 --bitrate 105 --stuffing -P timeref --start system -O dektec --device 0 --modulation DVB-T --frequency 578000000 --bandwidth 8 -r 3/4 -g 1/8
```
You should notice that ffmpeg generated streams are not dvb compilant, but this resulted on playable live muxing that i could use for lab propourses.

Its easy to see why ffmpeg muxing inconsistency if you analise the PCR (on the left, muxed tv on our station, on the right, ffmpeg muxing)
[image]
Anyway this stream is playable on TV, but probably will glitch hard on VLC, ive tried a lot of plugins like pcradjust but ended with even worse results. It seems ffmpeg just spit a lot of ts packets when it encounter a frame, instead of timing it on its bitrate. Ive got some good results using ffmpeg and opencaster in the middle (ffmpeg pipe video->esvideompeg2pes->pesvideo2ts->tsp) but im just starting to know tsduck and its enormus capabilities.

Im going a bit offtopic but you will encounter this issues when you start playing on your dectek, so you know in advance.

Our local tv station is now emitting EIT tables and HbbTV and soon subtitles, all powered by tsduck so id like to thanks @lelegard for such great broadcasting tool

phhusson, 2019-11-06:
Thanks @pabloko for this comment:

> if you cant produce a stream with fixed bitrate, use -I null then you can use -P mux.

I also have DTA-115, and hit the same issue as @michaeladams123 

To complete on #415, tsduck documentation mentions "This option
is ineffective on modulators which are partially software-based (DVB-T on DTA-110T or DTA115)."

So `--stuffing` on dektec plugin indeed can't work. And every examples in tsduck documentation assumes the output stuffs as-needed.

Thanks to @pabloko after a bit of work I managed to understand how to plug this in:

`tsp -v -b 24128342 -I null -P merge 'tsp -v -I file TS_422_466Mhz.ts -i -P filter -c -P filter -n -p 0x1fff -P pcrbitrate -P regulate -O file' -t -P pcrbitrate -P regulate -O dektec -i --modulation DVB-T -f 530000000 -b 8`

The `-b 24128342` parameter needs to be changed based on `tsterinfo`. You can also instead trust dektec log: `* dektec: initial output bitrate: 24,128,342 b/s`.

You can put whatever you want in the merge command obviously, the only point is that its output needs to be regulated.
the second pcrbitrate is probably useless since bitrate is already assumed (but at least logs shows that the pcrbitrate is the expected bitrate)

zjaffesyna, 2022-01-25:
hi , i have same issue, with dektec 315

tsp -v -I ip 239.254.253.31:1234 -O dektec -d 0 --frequency 1528000000 --symbol-rate=13000000 -r 8/9 --modulation=DVB-S2-QPSK --stuffing

any idea to fix it?

pabloko, 2022-01-26:
> hi , i have same issue, with dektec 315
> 
> tsp -v -I ip 239.254.253.31:1234 -O dektec -d 0 --frequency 1528000000 --symbol-rate=13000000 -r 8/9 --modulation=DVB-S2-QPSK --stuffing
> 
> any idea how to fix it?

You're expected to fill the entire frequency bandwidth, to do so, you need stuffing, this means, transmitting zeros to achive the frequency bandwidth capacity. Depending on your frequency and settings this value is calculable, see my last comment, the part about using `tsterinfo`. 
@phhusson posted a nice command that can get you started.

zjaffesyna, 2022-01-26:
this is the case, the mux fill all the BW with null packets if it has nothing else.

i'm not sure i understand what i should do, because the symbole-rate and frq , are given.

lars18th, 2022-01-26:
Hi @pabloko ,

> You're expected to fill the entire frequency bandwidth [...]

Please, excuse me if I'm wrong. But I'm doing a lot of testing for several years with TSDuck and cheal DVB modulators (hides), and my experience it's different. Perhaps @lelegard could help to us to resolve this topic:

- First of all, we need to assume that with non-professional tools (open source specifically) it is not possible to achieve a perfect DVB-T bitrate. The bitrate with this modulation are fractions of bits/seconds. For this reason the Dektec driver will have options to set it using two numbers. Futhermore, in recent releases TSDuck has enhanced the bitrate to use sub-bit fractions. However, this can't match at 100% the DVB-T bitrates.
- However, to correctly feed a DVB modulator you only need to provide a conformant DVB Transport Stream (MPTS) with _smaller_ bitrate than the total bitrate of the modulation used. Why _less_ and not _more_? Because when the modulator doesn't have sufficient data it adds _automatically_ an stuffing packet. That's the common way to resolve unmaching bitrates. And it's preferable to _remove_ stuffing packets because without sufficient stuff the modulator could be in troubles. Futhermore, this only works if the modulator will recalculate the PCR timestamps. Without this restamp the broadcasted signal will have incorrect marks as the target bitrate will be different from the incoming bitrate.

I want to comment this to suggest to use every time (for broadcasting) a bitrate slightly smaller than the target bitrate. For example 0.1-2.0 bit/sec less. In this case you only need to assume that the modulator has enabled the PCR restamp.

I don't know how Dektec driver works, but I feel the driver could do this for you.
I hope it helps.
Regards.

lars18th, 2022-01-26:
Hi @zjaffesyna ,

From my experience mixing in TSDuck a _realtime_ source and the `-P regulate` plugin it's a very bad idea. Why? Because this plugin doesn't work as you expected. It only _adds_ delays time-to-time, and never _advance_. So it only has sense with _files_ or other non-timing related sources. So, if you're reading from the network, then the source is realtime. And you need to feed as the source is producing. So remove at all the "regulate" plugin.

Then what you need to check it's that the source is producing at the correct bitrate. Select a slightly smaller bitrate in your muxer. Then in TSDuck you can read from it in realtime and feed to the modulator. I'm sure then the results will be what you expect.

Regards.

pabloko, 2022-01-26:
Hi @lars18th what im posting here is specific to dektec drivers and most specifically for the DTA-110 which i own. Those are the perks of those boards. On the bright side, on my tests, even non conformant dvb streams played very well with live restream, so its just matter of getting the correct settings to play with the used device. Ill try your advice on bitrates and check results. Thanks!

zjaffesyna, 2022-02-08:
sorry for the delay it took me time to make sure with the modulation team about the parameters the steam gets out.
so those are the stream full param:.

```
The stream is 12128M
QPSK – DVB-S2
Symbol rate 13M
Bit rate 22.417M
FEC 8/9.
```

and still after running this command:

`tsp -v -I ip 239.254.253.31:1234 -O dektec -d 0 --frequency 1528000000 --symbol-rate=13,000,000 -r 8/9 --modulation=DVB-S2-QPSK --stuffing`

after 2-3 min, i get:

```
* dektec: Got DMA underflow.
* dektec: Got DMA underflow.
* dektec: Got DMA underflow.
* dektec: Got DMA underflow.
* dektec: Got DMA underflow.
* dektec: Got DMA underflow.........
```

zjaffesyna, 2022-02-09:
is it possible to make a fix in the driver level? it feels to me, if we could have Xmb of buffer in the driver level, it won't happen.