tsduck/tsduck #425: Anyone interested in ISDB tuners ?

lelegard, 2019-10-01:
Dear TSDuck users,

Is anyone interested in support for ISDB tuners ?

Being based in Europe, I do not have ISDB tuners and I do not have access to real operational ISDB networks. If anyone is interested in TSDuck supporting ISDB tuners, I will need cooperation with someone based in an ISDB country, Japan, Brazil or similar.

Linux support for ISDB-T and ISDB-S looks correct. So, Linux is a priority. Windows support for ISDB is not documented but the header files contain references to ISDB-S (not much to ISDB-T). So, I do not know if using ISDB on Windows is possible at all. Any experience in DirectShow with ISDB would be welcome.

sionjujuy, 2019-10-01:
Hello how are you? I am from Argentina, I have reception equipment in isdb tb with the ability to receive interactive applications, middleware ginga.ar (ginga argentina).

If I can collaborate with tsduck on anything, it would be a pleasure.

regards

Marco Camacho

lelegard, 2019-10-02:
Hi @sionjujuy,

Thank you for offering your collaboration. Do you have any ISDB tuner that can be plugged into a computer to receive a transport stream, either PCIe or USB? Do you have any Linux driver for it? Which Linux distribution are you using?

The idea is to open the `dvb` plugin to ISDB reception so we need devices to test. I agree that `dvb` is now a bad name for the tuner reception plugin. It started with DVB tuners only but ATSC tuners are also supported now.

sionjujuy, 2019-10-02:
Hi @lelegard , I have a dektec dtu 215 Usb  with an isdbt license, I can get an usb tuner isdbt and linux ubutu, to do the necessary tests.

At the moment I am traveling back in two weeks and I do the tests.

Regards

lelegard, 2019-10-02:
Thanks. Currently, it won't work since the `dvb` plugin does not support ISDB yet. So, I need to add support for the required modulation parameters and then you could test.

Could anyone give a complete list of the required modulation parameters for ISDB-T (and ISDB-S if known)?

sionjujuy, 2019-10-02:
@lelegard, could you send me an email so I send you pdf files that can help you, viatsat@gmail.com

sionjujuy, 2019-10-02:
https://www.kernel.org/doc/html/v4.14/media/uapi/dvb/fe_property_parameters.html

lelegard, 2019-10-03:
I had read the list of LinuxTV parameters but the ISDB specificities were not clearly summarized.

Let's try to build an exhaustive list of possible modulation parameters for ISDB on Linux. Please report any error or additional useful information.

| Parameter | ISBD-T | ISDB-S |
| --------- | ------ | ------ |
| DTV_FREQUENCY | value in Hz | value in kHz |
| DTV_MODULATION | QPSK, DQPSK, 16-QAM, 64-QAM | 8-PSK, QPSK, BPSK |
| DTV_BANDWIDTH_HZ | 5MHz, 6MHz, 7MHz, 8MHz |  |
| DTV_INVERSION | ? | ? |
| DTV_SYMBOL_RATE | unused | value in sym/s ? |
| DTV_INNER_FEC | unused | which values are valid ? |
| DTV_ISDBT_PARTIAL_RECEPTION | need explanations | unused |
| DTV_ISDBT_SOUND_BROADCASTING | 0, 1, -1 (AUTO) | unused |
| DTV_ISDBT_SB_SUBCHANNEL_ID | 0..41, -1 (AUTO) | unused |
| DTV_ISDBT_SB_SEGMENT_IDX | 0..12 | unused |
| DTV_ISDBT_SB_SEGMENT_COUNT | 1..13 | x |
| DTV_ISDBT_LAYER_ENABLED | 3 bits, bitmask | x |
| DTV-ISDBT_LAYER[A-C]_FEC | FEC_AUTO, FEC_1_2, FEC_2_3, FEC_3_4, FEC_5_6, FEC_7_8 | unused |
| DTV-ISDBT_LAYER[A-C]_MODULATION | QAM_AUTO, QPSK, QAM_16, QAM_64, DQPSK | unused |
| DTV-ISDBT_LAYER[A-C]_SEGMENT_COUNT | 0..13, -1 (AUTO) | unused |
| DTV-ISDBT_LAYER[A-C]_TIME_INTERLEAVING | 0, 1, 2, 4, -1 (AUTO) | unused |
| DTV_CODE_RATE_HP | which values are valid ? | unused |
| DTV_CODE_RATE_LP | which values are valid ? | unused |
| DTV_GUARD_INTERVAL | which values are valid ? | unused |
| DTV_TRANSMISSION_MODE | which values are valid ? | unused |
| DTV_STREAM_ID | unused | 1..65535, NO_STREAM_ID_FILTER |

Additional question: 

- What is ISDB-Tsb ? "Sound broadcasting" ? Or something related to ISDB-Tb (Brazil) ?
  - If it is sound broadcasting, what is this exactly? Is this useful to support in TSDuck?
- What are the main differences between ISDB-T and ISDB-Tb ?
  - Modulation ?
  - Signalization (PSI/SI) ?
- Is there any ISDB-Sb ? Or are the Brazilian / South America specificities limited to terrestrial ?

Thanks for your help.

sionjujuy, 2019-10-07:
Hello, more information Isdbt - isdb-tbs

lelegard, 2019-10-08:
> Hello, more information Isdbt - isdb-tbs

What do you mean?

sionjujuy, 2019-10-08:
I send you an attachment where it explains about isdb and isbdtsb so you
can see the differences.

I still don't have time to answer your questions from the previous email,
excuse me.

lelegard, 2019-10-27:
Hi @sionjujuy and all ISDB users,

I have implemented all ISDB-T and ISDB-S parameters (see the updated user's guide, plugin `dvb`). The Linux implementation is done (not the Windows one).

If this is possible for you, could you please test with your tuners. You have to pull the repo and rebuild in any Linux distro ([instructions here](https://tsduck.io/doxy/building.html)).

This code was done without any test (no ISDB tuner or network here). So, there is a risk that it does not work at once. I suggest to start with `tslsdvb -v`. Please post the result. Please also post the name and characteristics of the tuners you use (the idea is to add it in the user's guide table at section 6.1.4, so please include all relevant details or required firmware). Make sure that there is no error message (missing driver or missing firmware) using the `dmesg' command.

Thank you all for your feedback.

lelegard, 2020-04-05:
First tested ISDB-T tuners: A user from Japan reported that the following two tuners work on Linux and Windows: MyGica S270 and PLEX PX-S1UD. They are internally identical.

nuclidjp, 2020-04-28:
Mygica S2780
I have confirmed the operation on raspbian buster and Windows 10.
raspbian requires linuxtv.org/downloads/firmware/dvb-usb-dib0700-1.20.fw.

lelegard, 2020-04-29:
> Mygica S2780

Do you mean Mygica S270?

nuclidjp, 2020-04-29:
It was a typo.
Mygica S2870.

lelegard, 2020-04-29:
Thanks, doc updated.

ljccitterio, 2023-09-16:
Greetings in advance.

I have a TDA STB for DTT in Venezuela with ISDB-Tb protocol (Brazilian and Japanese version). The equipment distributed in Venezuela is OEM assembled in Argentina. 
What is the possibility of modifying the STB firmware? 

Any help or advice?. 

I have attached the boot of the device.

Placa: 742
Modelo de Memoria:  [32-bit mode]

U-Boot 1.3.1 (Oct 18 2012 - 16:37:27) - 

DRAM:  256 MiB
SPI:  info: found MXIC MX25L6455E (8MiB) device (page=256,erase=4096)
In:    serial
Out:   serial
Err:   serial
Hit ENTER key to stop autoboot:  0 
## Booting image at 00120000 ...
   Image Name:   Linux 2.6
   Image Type:   SuperH Linux Kernel Image (gzip compressed)
   Data Size:    2628083 Bytes =   2.5 MiB
   Load Address: 80800000
   Entry Point:  80801000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel console=ttyAS0,115200 root=/dev/mtdblock0 rw rootfstype=jffs2 rootdelay=2 ramdisk_size=5120 nwhwconf=device:eth0,hwaddr:00:14:49:E4:B6:49 ip=.

Linux version 2.6.23.17_stm23_A27--SDK7105_7105-STSDK (microtrol@UTE742A) (gcc version 4.2.4 (snapshot) (STMicroelectronics/Linux Base 4.2.4-56)) #1 PREEMPT 2
Booting machvec: pdk7105
Node 0: start_pfn = 0x40000, low = 0x54000
Zone PFN ranges:
  Normal     262144 ->   344064
Movable zone start PFN for each node
early_node_map[1] active PFN ranges
    0:   262144 ->   344064
STMicroelectronics PDK7105-SDK board initialisation
STx7105 version 4.x
Built 1 zonelists in Zone order.  Total pages: 81280
Kernel command line: console=ttyAS0,115200 root=/dev/mtdblock0 rw rootfstype=jffs2 rootdelay=2 ramdisk_size=5120 nwhwconf=device:eth0,hwaddr:00:14:49:E4:B6:40
bpa2: partition 'bigphysarea' created at 0x40f53000, size 8000 kB (0x007d0000 B)
PID hash table entries: 2048 (order: 11, 8192 bytes)
Using tmu for system timer
Using 25.000 MHz high precision timer.
Console: colour dummy device 80x25
console [ttyAS0] enabled
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Memory: 303488k/327680k available (3731k kernel code, 846k data, 148k init)
PVR=04909200 CVR=60880000 PRR=00009e40
I-cache : n_ways=2 n_sets=512 way_incr=16384
I-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
D-cache : n_ways=2 n_sets=512 way_incr=16384
D-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
Mount-cache hash table entries: 512
CPU: STx7105
NET: Registered protocol family 16
Generic PHY: Registered new driver
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
Bluetooth: Core ver 2.11
NET: Registered protocol family 31
Bluetooth: HCI device and connection manager initialized
Bluetooth: HCI socket layer initialized
Time: SuperH clocksource has been installed.
[... 107 lines cut ...]
    not configured!
---------------------------------------------------------------
stm_rng hardware driver 1.0 configured
heartbeat: version 0.1.1 loaded
TCP cubic registered
NET: Registered protocol family 1
NET: Registered protocol family 17
Bluetooth: L2CAP ver 2.8
Bluetooth: L2CAP socket layer initialized
Bluetooth: SCO (Voice Link) ver 0.5
Bluetooth: SCO socket layer initialized
Bluetooth: RFCOMM socket layer initialized
Bluetooth: RFCOMM TTY layer initialized
Bluetooth: RFCOMM ver 1.8
Bluetooth: BNEP (Ethernet Emulation) ver 1.2
Bluetooth: BNEP filters: protocol multicast
Bluetooth: HIDP (Human Interface Emulation) ver 1.2
ieee80211: 802.11 data/management/control stack, git-1.1.13
ieee80211: Copyright (C) 2004-2005 Intel Corporation <jketreno@linux.intel.com>
sh4 suspend support registered
stmmac_timer: TMU2 Timer ON (freq 256Hz)
IP-Config: Complete:
      device=eth0, addr=192.168.99.162, mask=255.255.255.0, gw=192.168.99.254,
     host=UTE742, domain=, nis-domain=(none),
     bootserver=255.255.255.255, rootserver=255.255.255.255, rootpath= ....."

lelegard, 2023-09-18:
Hi @ljccitterio 

> I have a TDA STB for TDT (Television Digital Terrestre) in Venezuela with ISDB-Tb protocol (Japanese-Brazilian norm or standard). The equipment distributed in Venezuela is OEM assembled in Argentina.
>
> What is the possibility of modifying the STB firmware?

First, this is not a topic for TSDuck. Your device is an independent set-top box, not a tuner which is attached to a computer running TSDuck.

Second, jailbreaking a consumer device is usually not permitted and, in the case of Pay-TV, illegal. This is not a topic which is allowed to be discussed here.

ljccitterio, 2023-09-18:
> Hi @ljccitterio
>
>
> First, this is not a topic for TSDuck. Your device is an independent set-top box, not a tuner which is attached to a computer running TSDuck.
>
> Second, jailbreaking a consumer [...]

Greetings. The device I mention is not for paid service. In Venezuela the digital television service is free. I don't know if I violate any laws or regulations by modifying these STBs for TDA. Since these STBs for the ISDB-Tb standard in Venezuela do not have support for updating their firmware, I wanted to know if it is possible to adapt other embedded software to the device. Thank you for your time.

ljccitterio, 2025-02-25:
Hi bro.

What kind of ISDB tuner?

Could you send some specifications and photos of the tuner?

Siso79, 2026-04-07:
> Me consigues aquí en IG
> 
> https://www.instagram.com/electronica.reparaciones?igsh=MXdiMmgxaXQ0MjliZQ==
> 
> El lun., 6 de abril de 2026 11:06 p. m., Siso79 ***@***.***>
> escribió:
> […](#)

Ok gracias, cuando puedas lee lo que te escribi por aqui, aunque tambien te escribi en IG pero  con menos detalles. Y bueno para saber algunas respuestas sobre todas esas dudas que escribi aqui. Ok despues si te encontre en IG. Muchas gracias.

ljccitterio, 2026-04-07:
Dale, estaré revisando. Habría que revisar cada placa de esos STB para ver
con qué MCU cuentan.

Esos fallas la mayoría de l las veces se debe a EEPROM o SPI Flash
corrompido por una variación inestable de tensiones, suelen ser muy
susceptibles a esas inestabilidades.

Estaré al pendiente.