tsduck/tsduck #426: Unable to analyze ASI input on Dektec DTA-2174 - DTA-2174

themoog, 2019-10-10:
STEPS:

1) Where -i = input 1 (asi)

C:\Program Files\TSDuck\bin>tsdektec -i 0

C:\Program Files\TSDuck\bin>

2)
C:\Program Files\TSDuck\bin>tsp -I dektec -P analyze
* Error: dektec: capture error on Dektec device 0: DTAPI_E_INVALID_SIZE (DTAPI status 4110)

  Subsystem id: 0x0000 (DTA-2174)
  Subsystem vendor id: 0x0000
  Device id: 0x087E
  Vendor id: 0x1A0E
  Serial number: 000000008194A736
  Firmware version: 6 (0x00000006)
  Firmware variant: 0 (0x00000000)
  PCI bus: 218, slot: 33
  Customer id: 305365
  Engineering change level: Rev 2
  Manufacture id: 04
  Production date: 2019.01
  Part number: DTA-2174
  Serial number: 2174002998

Thanks in advance

Neil

lelegard, 2019-10-10:
Hi,

I do not understand the beginning of your post, before the tsp command. There must be some copy/paste issue.

I do not have any DTA-2174 card, so I cannot make any test. Looking at the code, the message is issued on the receive operation. So, all configuration phases properly pass. 

The Dektec documentation says, for the receive operation:

`DTAPI_E_INVALID_SIZE`: The specified transfer size is negative or not a multiple of 4

I do not see how this is possible since the size of the reception buffer is always a multiple of the TS packet size. So, it must be something else. But I have no idea what it could be.

Maybe there is some undocumented constraint on the reception size of the DTA-2174. Try to play with the option `--max-input-packets` in the `tsp` command to see if different reception sizes make a difference.

Does the card work with Dektec software tools?

themoog, 2019-10-10:
Hi sorry my bad,

Thanks for responding
So:
Yes card works in the Dektec tools

Have you a full example of the --max-input-packets line?

Some questions about the command line?
Can I confirm how you map the device and input?

c:\Program Files\TSDuck\bin>tsdektec -i 2
* Error setting port 2 to input mode: DTAPI_E_INVALID_ARG (DTAPI status 4140)

c:\Program Files\TSDuck\bin>tsdektec -i 0

c:\Program Files\TSDuck\bin>tsdektec -i 1
* Error setting port 1 to input mode: DTAPI_E_INVALID_ARG (DTAPI status 4140)

c:\Program Files\TSDuck\bin>tsdektec -i 2
* Error setting port 2 to input mode: DTAPI_E_INVALID_ARG (DTAPI status 4140)

c:\Program Files\TSDuck\bin>tsdektec -i 3
* Error setting port 3 to input mode: DTAPI_E_INVALID_ARG (DTAPI status 4140)

Here's the full output of dektec -av command

c:\Program Files\TSDuck\bin>tsdektec -av

DTAPI: 5.35.0.123, Service: 3.5.0.75, Dta: 4.27.0.256, DtaNw: -1.-1.-1.-1

* Device 0: DTA-2174 (Quad 3G-SDI/ASI in+out)
  Physical ports: 8
  Channels: input: 4, output: 0
  Input 0: DTA-2174 port 1 (top), ASI, SDI (Locked to a genlock reference,
      Genlock reference input, Uni-directional input, Uni-directional output,
      Double buffered output, ASI output signal can be inverted, Matrix API
      support, High-level Matrix API support, Forms a Quad-link group with next
      3 ports, 10-bit network byte order, 64-bit timestamping, Transparent mode,
      MPEG-2 transport stream, 3G-SDI, DVB-ASI transport stream, HD-SDI, SD-SDI,
      1080p/50 lvl A, 1080p/50 lvl B, 1080p/59.94 lvl A, 1080p/59.94 lvl B,
      1080p/60 lvl A, 1080p/60 lvl B, 1080i/50, 1080i/59.94, 1080i/60, 1080p/23.
      98, 1080p/24, 1080p/25, 1080p/29.97, 1080p/30, 1080psf/23.98, 1080psf/24,
      1080psf/25, 1080psf/29.97, 1080psf/30, 720p/23.98, 720p/24, 720p/25,
      720p/29.97, 720p/30, 720p/50, 720p/59.94, 720p/60, 525i/59.94, 625i/50,
      Lock TS rate to input port)
  Input 1: DTA-2174 port 2, ASI, SDI (Locked to a genlock reference,
      Uni-directional input, Uni-directional output, Double buffered output, ASI
      output signal can be inverted, Matrix API support, High-level Matrix API
      support, 10-bit network byte order, 64-bit timestamping, Transparent mode,
      MPEG-2 transport stream, 3G-SDI, DVB-ASI transport stream, HD-SDI, SD-SDI,
      1080p/50 lvl A, 1080p/50 lvl B, 1080p/59.94 lvl A, 1080p/59.94 lvl B,
      1080p/60 lvl A, 1080p/60 lvl B, 1080i/50, 1080i/59.94, 1080i/60, 1080p/23.
      98, 1080p/24, 1080p/25, 1080p/29.97, 1080p/30, 1080psf/23.98, 1080psf/24,
      1080psf/25, 1080psf/29.97, 1080psf/30, 720p/23.98, 720p/24, 720p/25,
      720p/29.97, 720p/30, 720p/50, 720p/59.94, 720p/60, 525i/59.94, 625i/50,
      Lock TS rate to input port)
  Input 2: DTA-2174 port 3, ASI, SDI (Locked to a genlock reference,
      Uni-directional input, Uni-directional output, Double buffered output, ASI
      output signal can be inverted, Matrix API support, High-level Matrix API
      support, 10-bit network byte order, 64-bit timestamping, Transparent mode,
      MPEG-2 transport stream, 3G-SDI, DVB-ASI transport stream, HD-SDI, SD-SDI,
      1080p/50 lvl A, 1080p/50 lvl B, 1080p/59.94 lvl A, 1080p/59.94 lvl B,
      1080p/60 lvl A, 1080p/60 lvl B, 1080i/50, 1080i/59.94, 1080i/60, 1080p/23.
      98, 1080p/24, 1080p/25, 1080p/29.97, 1080p/30, 1080psf/23.98, 1080psf/24,
      1080psf/25, 1080psf/29.97, 1080psf/30, 720p/23.98, 720p/24, 720p/25,
      720p/29.97, 720p/30, 720p/50, 720p/59.94, 720p/60, 525i/59.94, 625i/50,
      Lock TS rate to input port)
  Input 3: DTA-2174 port 4, ASI, SDI (Locked to a genlock reference,
      Uni-directional input, Uni-directional output, Double buffered output, ASI
      output signal can be inverted, Matrix API support, High-level Matrix API
      support, 10-bit network byte order, 64-bit timestamping, Transparent mode,
      MPEG-2 transport stream, 3G-SDI, DVB-ASI transport stream, HD-SDI, SD-SDI,
      1080p/50 lvl A, 1080p/50 lvl B, 1080p/59.94 lvl A, 1080p/59.94 lvl B,
      1080p/60 lvl A, 1080p/60 lvl B, 1080i/50, 1080i/59.94, 1080i/60, 1080p/23.
      98, 1080p/24, 1080p/25, 1080p/29.97, 1080p/30, 1080psf/23.98, 1080psf/24,
      1080psf/25, 1080psf/29.97, 1080psf/30, 720p/23.98, 720p/24, 720p/25,
      720p/29.97, 720p/30, 720p/50, 720p/59.94, 720p/60, 525i/59.94, 625i/50,
      Lock TS rate to input port)
  Subsystem id: 0x0000 (DTA-2174)
  Subsystem vendor id: 0x0000
  Device id: 0x087E
  Vendor id: 0x1A0E
  Serial number: 000000008194A736
  Firmware version: 6 (0x00000006)
  Firmware variant: 0 (0x00000000)
  PCI bus: 218, slot: 33
  Customer id: 305365
  Engineering change level: Rev 2
  Manufacture id: 04
  Production date: 2019.01
  Part number: DTA-2174
  Serial number: 2174002998

themoog, 2019-10-11:
Could this be important? For each of the above it starts with Input x: DTA-2174 port x
Do I have to declare the input/outputs in a different way?
Whats an input and whats a port?

Thanks so much

Neil