tsduck/tsduck #428: BDA Namespace Iteration

SS-FF, 2019-10-14:
Your approach in ts::Tuner::buildGraph to iterate over all existing namespaces and "try" each one is not quite the best possible solution:

https://github.com/tsduck/tsduck/blob/e10daf28eaa46e0728a27e9c716ef93f1d910ad2/src/libtsduck/windows/tsTuner.cpp#L825-L837

This is bad for two reasons:
- In some cases, the network provider will accept the wrong tuning spaces
(e.g. 'Digital Cable' for DVB-C tuners)
- You're actually making yourself dependent on what's currently configured in the registry
  - What you get when enumerating is just what is stored here: Computer\HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Multimedia\TV\Tuning Spaces in the registry
  - This involves the danger of tuning spaces being picked that were installed by other applications or tools, or problems when the required one does not exist
  - There have even been issues here in the repo caused by this. I remember two:
    - Some tuning space where the name started with "DD2..." was picked instead of the right one
    - A Hauppauge DVB-C tuner only worked after installing the Hauppauge software which created the tuning space for DVB-C

SS-FF, 2019-10-14:
A better approach is this:

```cs
    ComPtr<::ITunerCap> tuner_cap;
    tuner_cap.queryInterface(_tuner.pointer(), ::IID_ITunerCap, debug_report);
    if (tuner_cap.isNull()) {
        report.error(u"failed to get ITunerCap interface");
        return false;
    }

    static const ::ULONG MAX_TYPES = 5;
    ::ULONG networkTypeCount = MAX_TYPES;
    ::GUID networkTypes[MAX_TYPES];

    hr = tuner_cap->get_SupportedNetworkTypes(MAX_TYPES, &networkTypeCount, networkTypes);
    if (!ComSuccess(hr, u"ITunerCap::get_SupportedNetworkTypes", report)) {
        return false;
    }

    if (networkTypeCount == 0) {
        report.error(u"Tuner did not return any supported network types");
        return false;
    }

    report.debug(u"Supported Network Types:");

    for (UINT n = 0; n < networkTypeCount; n++)
    {
        const UString netTypeString(FormatGUID(networkTypes[n], true));
        report.debug(u"  network type %d = %s", {n, netTypeString});
    }

    for (UINT n = 0; n < networkTypeCount && !tspace_found; n++)
    {
        const UString netTypeString(FormatGUID(networkTypes[n], true));

        ////ComPtr<::ITuningSpace> tuningSpace;
        if (GetTuningSpaceFromNetworkType(networkTypes[n], tspace, report)) {

            assert(!tspace.isNull());

            // Display tuning space in debug mode
            const UString fname(GetTuningSpaceFriendlyName(tspace.pointer(), report));
            const UString uname(GetTuningSpaceUniqueName(tspace.pointer(), report));
            report.debug(u"trying to use tuning space \"%s\" (%s)", { fname, uname });

```

You query the supported network types from the device (usually it's just a single one).

The double for loop is just for informational purposes, to see if there would have been a second one..

Everything else should be familiar, up to the `GetTuningSpaceFromNetworkType` call.

That function is creating locator and tuning space manually for each network type in a fixed and predefined way that is independent from local registry entries.

SS-FF, 2019-10-14:
This is how it looks like:

```cs
bool ts::GetTuningSpaceFromNetworkType(::GUID networkType, ComPtr<::ITuningSpace>& tuningSpace, Report& report)
{
    if (networkType == CLSID_DVBCNetworkProvider) {

        ComPtr<::IDVBCLocator> loc(CLSID_DVBCLocator, ::IID_IDVBCLocator, report);

        if (loc.isNull() ||
            !PUT(loc, CarrierFrequency, long(-1)) || 
            !PUT(loc, Modulation, ::BDA_MOD_NOT_SET) ||
            !PUT(loc, InnerFEC, ::BDA_FEC_METHOD_NOT_SET) ||
            !PUT(loc, InnerFECRate, ::BDA_BCC_RATE_NOT_SET) ||
            !PUT(loc, OuterFEC, ::BDA_FEC_METHOD_NOT_SET) ||
            !PUT(loc, OuterFECRate, ::BDA_BCC_RATE_NOT_SET) ||
            !PUT(loc, SymbolRate, long(-1)))
        {
            return false;
        }

        ComPtr<::IDVBTuningSpace> newTs(CLSID_DVBTuningSpace, ::IID_IDVBTuningSpace, report);

        if (newTs.isNull() ||
            !PUT(newTs, UniqueName, L"TsDuck DVB-C Tuning Space") ||
            !PUT(newTs, FriendlyName, L"TsDuck DVB-C Tuning Space") ||
            !PUT(newTs, _NetworkType, CLSID_DVBCNetworkProvider) ||
            !PUT(newTs, SystemType, DVBSystemType::DVB_Cable) ||
            !PUT(newTs, DefaultLocator, loc.pointer()))
        {
            return false;
        }

        tuningSpace.assign(newTs);
        return true;
    }

    if (networkType == CLSID_DVBTNetworkProvider) {

        ComPtr<::IDVBTLocator> loc(CLSID_DVBTLocator, ::IID_IDVBTLocator, report);

        if (loc.isNull() ||
            !PUT(loc, CarrierFrequency, long(-1)) || 
            !PUT(loc, Modulation, ::BDA_MOD_NOT_SET) ||
            !PUT(loc, InnerFEC, ::BDA_FEC_METHOD_NOT_SET) ||
            !PUT(loc, InnerFECRate, ::BDA_BCC_RATE_NOT_SET) ||
            !PUT(loc, OuterFEC, ::BDA_FEC_METHOD_NOT_SET) ||
            !PUT(loc, OuterFECRate, ::BDA_BCC_RATE_NOT_SET) ||
            !PUT(loc, SymbolRate, long(-1)))
        {
            return false;
        }

        ComPtr<::IDVBTuningSpace> newTs(CLSID_DVBTuningSpace, ::IID_IDVBTuningSpace, report);

        if (newTs.isNull() ||
            !PUT(newTs, UniqueName, L"TsDuck DVB-T Tuning Space") ||
            !PUT(newTs, FriendlyName, L"TsDuck DVB-T Tuning Space") ||
            !PUT(newTs, _NetworkType, CLSID_DVBTNetworkProvider) ||
            !PUT(newTs, SystemType, DVBSystemType::DVB_Terrestrial) ||
            !PUT(newTs, DefaultLocator, loc.pointer()))
        {
            return false;
        }

        tuningSpace.assign(newTs);
        return true;
    }

    if (networkType == CLSID_DVBSNetworkProvider) {
[... 102 lines cut ...]
            !PUT(newTs, CountryCode, 0) ||
            !PUT(newTs, InputType, TunerInputCable) ||
            !PUT(newTs, MaxMajorChannel, 99) ||
            !PUT(newTs, MaxMinorChannel, 999) ||
            !PUT(newTs, MaxPhysicalChannel, 158) ||
            !PUT(newTs, MaxChannel, 9999) ||
            !PUT(newTs, MinMinorChannel, -1) ||
            !PUT(newTs, MinMajorChannel, -1) ||
            !PUT(newTs, MinPhysicalChannel, 2) ||
            !PUT(newTs, MinChannel, -1) ||
            !PUT(newTs, MinSourceID, 0) ||
            !PUT(newTs, MaxSourceID, 0x7fffffff) ||
            !PUT(newTs, DefaultLocator, loc.pointer()))
        {
            return false;
        }

        tuningSpace.assign(newTs);
        return true;
    }

    return false;
}

```

SS-FF, 2019-10-14:
BTW: Those values above are just the default values from a fresh Windows install. 

Apologies that I'm unable to create a PR at this time, but this should be easy to integrate.

lelegard, 2019-10-14:
Hi @SS-FF,

Thanks a lot for your help. I am currently refactoring a lot of code around tuners in a separate branch. I am currently testing the Linux implementation. I haven't touched the Windows implementation for now. But I will definitely use your suggestion when it comes to Windows.

The reason for this refactoring is the support of multi-standard tuners (DVB-T + DVB-C for instance). Currently, a tuner is identified with a given type. This was OK ten years ago when I wrote this code. This is no longer the case. So, the approach of using distinct subclasses of TunerParameters is no longer appropriate. The impact on the code is quite important and requires a substantial amount of refactoring.

With multi-standard tuners, the problem is to be able to detect which standards are supported by a given device and how to switch from one to the other. Linux now has a standard way to do that. This is why I started with Linux. 

Do you have an idea on how to do that on Windows? I mean detect which standards are supported by a given device and how to switch from one to the other. If there is no such standard method in Windows, multi-standard tuners will probably continue to use only their current standard with TSDuck.

SS-FF, 2019-10-14:
> With multi-standard tuners, the problem is to be able to detect which standards are supported by a given device and how to switch from one to the other. Linux now has a standard way to do that. This [...]

The answer about how to detect the supported standards is right here in this post: `get_SupportedNetworkTypes`

I've been doing tests with quite an amount of devices recently, but all multi-standard tuners in the test are modeling their standards as separate tuners.
But I'm pretty sure that you just need to set the appropriate tuning space in the network provider to "switch between standards", that's all.

The right order would be:
- Connect tuner to network provider
- Set tuning space
- Connect receiver filters

lelegard, 2019-10-17:
Hi @SS-FF 

In fact, I have to rephrase my previous report. The method `get_SupportedNetworkTypes` does not work before putting a tuning space (method `put_TuningSpace`) on the network provider filter. It does work _after_ putting the tuning space. But this does not bring any benefit to us. We still have to try tuning spaces one by one. And when we succeed to put the DVB-S tuning space (for instance), then you can get the supported network types, which happen to be ... DVB-S.

Moreover, the method `get_SupportedNetworkTypes` does not work on the tuner filter (the proprietary filter for the device). The tuner filter does not implement the `ITunerCap`  interface (I tried, it says this interface is not supported).

The `ITunerCap`  interface is implemented by the standard Microsoft Network Provider filter, through its `ITuner` interface. So, the `ITunerCap`  interface does not give you the capabilities of a device, it gives you the current capabilities of a network provider after it is completely configured for a network. In short, the `ITunerCap`  interface informs you of something you already know (because you explicitly configured it before).

What we would need is some interface which queries a given physical device for its potential capabilities, regardless of the current configuration. This is trivial on Linux, just one `ioctl`. This seems impossible on Windows.

SS-FF, 2019-10-18:
> In fact, I have to rephrase my previous report. The method `get_SupportedNetworkTypes` does not work before putting a tuning space (method `put_TuningSpace`) on the network provider filter. It does [...]

No, no and no.

> Moreover, the method `get_SupportedNetworkTypes` does not work on the tuner filter (the proprietary filter for the device). The tuner filter does not implement the `ITunerCap` interface (I tried, it [...]

Exactly! The ITuner interface is the '_tuner' variable in your original code as well as in the code I posted here. I didn't expect this to be confusing.

> So, the `ITunerCap` interface does not give you the capabilities of a device, it gives you the current capabilities of a network provider after it is completely configured for a network. In short, [...]

No.

**After** you connect the network provider to the (BDA Source) tuner filter, you will get correct results from `get_SupportedNetworkTypes`.

SS-FF, 2019-10-18:
BTW, you can easily watch this behavior by using GraphEdit or GraphStudioNext:
- Add MS Network Provider
- Right-Click the provider and select 'Properties'
  => Tuning Spaces List is empty
- Add tuner ('BDA Source Filter') 
- Connect if it didn't auto-connect to the network provider filter
- Again: Right-Click the provider and select 'Properties'
  => Tuning Spaces List shows those tuning spaces that the tuner supports

Note: The tuning spaces you see there are taken from the registry. If your device is DVB-C and there is no tuning space configured for this in the registry, you won't see any DVB-C entry in that list either.
But get_SupportedNetworkTypes would still return DVB-C as a supported network type, regardless of the tuning space registry entries.

lelegard, 2019-10-18:
Hi @SS-FF 

> After you connect the network provider to the (BDA Source) tuner filter, you will get correct results from `get_SupportedNetworkTypes`.

You are right. If I use `get_SupportedNetworkTypes` from the network provider, after connecting it to the tuner but before putting a tuning space (thus making no assumption on the network type), then it works. 

This is good to know. Now I have a starting point to thing about a different strategy.

Thanks a lot.

SS-FF, 2019-10-19:
You're welcome. As explained above, I'd suggest creating tuning spaces manually instead of enumerating the existing ones.
This will resolve the case you mentioned, where the included software needs to be run first which creates a DVB-C tuning space as there is none by default..

lelegard, 2019-10-29:
Hi @SS-FF,

I have implemented your suggestions.

I get the list of network types from each tuner filter. BTW, my DVB-T/T2 tuner (PCTV NanoStick) now appears as two tuners. The second one supports two networks: `DIGITAL_CABLE_NETWORK_TYPE` and `CLSID_DVBCNetworkProvider`. Previously, because there is no predefined cable tuning space, this tuner was not seen by TSDuck. So, this is an improvement.

I have a question. How do we know if a DVB-T tuner supports T2? Same for DVB-S tuners with S2? I still have a few old DVB-T tuners which don't support DVB-T2. On Linux, the list of supported "delivery systems" is quite precise. DVB-T and DVB-T2 are individually reported. My old tuners report only DVB-T while the more recent ones report DVB-T and DVB-T2. But on Windows, you just get `CLSID_DVBTNetworkProvider` from the list of supported networks.

This question is not new. Previously, on Windows, I assumed that tuners accepting the DVB-T tuning space also supported T2 (same for S/S2). It was an identified limitation of Windows. I wonder if we can do better now.

Additionally, I have a weird - although harmless - behaviour with setting the UniqueName and FriendlyName of some tuning space. The actual code is derived from your example. When I set the UniqueName and FriendlyName of some tuning space (not all) and the code is compiled in release mode (no problem in debug compilation), then `put_UniqueName` and `put_FriendlyName` return the error "The specified object is too large". I do not see the error when compiled in debug mode. Since the names are simple string literal, I do not understand why it would make a difference. But the debug external libraries are sometimes different. Moreover, when it happens, I notice that the error occurs on the first tuning space only. With a DVB-T/T2/C tuner, if I check the DVB-C tuning space first, then the error occurs on the DVB-C tuning space, not on the DVB-T one. But if I check DVB-T first, the error occurs on DVB-T and not on DVB-C. Finally, when I use a DVB-S/S2 tuner, the error does not occur at all.

Since this is harmless (just a name in the tuning space), I simply ignore the error (the error message is displayed only with option `--debug`). But I do not like this kind of behaviour. It is often an indication that something is wrong deeper and worse consequences may be expected any time.

SS-FF, 2019-10-30:
> I get the list of network types from each tuner filter. BTW, my DVB-T/T2 tuner (PCTV NanoStick) now appears as two tuners. The second one supports two networks: `DIGITAL_CABLE_NETWORK_TYPE` and [...]

Use CLSID_DVBCLocator and CLSID_DVBTuningSpace (watch closely: no T, no C. This is meant to be used for both, not satellite, though).

Ignore DIGITAL_CABLE_NETWORK_TYPE (=> CLSID_DigitalCableLocator, CLSID_DigitalCableTuningSpace). This is for US ClearQam tuners only (I've never seen one actually).

SS-FF, 2021-05-03:
Already implemented..

closed 2021-05-03 by SS-FF