tsduck/tsduck #430: Documentation errata

samuraibrian, 2019-10-17:
Couldn't think of anywhere else to feedback documentation issues... So, as a starter:
User's Guide, **5.2.12 Monitoring the stuffing rate of all transponders in a network**, the example script shows the following usage of the dvb plugin:

    `tsp -I dvb -u $uhf \`

which borks with the error message `tsp: dvb: unknown option -u`

Works with `--uhf-channel`, in place of `-u`

BR, Jon

lelegard, 2019-10-17:
Hi @samuraibrian,

This place is suitable for all kinds of feedback, including documentation. Any report which improves the documentation is welcome. By the way, this includes language issues such as incorrect or awkward wordings (I am not a native English speaker).

Thank you for reporting this. As you can see in the output of this example, it was run in 2008 ! At that time, the option `-u` was valid for the `dvb` plugin. It was later removed, when the option `--vhf-channel` was added. At that time, for the sake of consistency between `--uhf-channel`  and `--vhf-channel`, either both short options `-u` and `-v` should be defined or none. Since `-v` is typically used everywhere for `--verbose`, I killed `-u`.

I will fix the doc.

samuraibrian, 2019-10-17:
> By the way, this includes language issues such as incorrect or awkward wordings (I am not a native English speaker).

Your English grammar is better than most native English grammar!

closed 2019-10-17 by samuraibrian