tsduck/tsduck #431: scramble stream with two ECM streams

adambeveridge1, 2019-10-17:
Hi Thierry,

Thanks again for this really amazingly helpful software! Truly appreciate all of the help you have been able to provide.

I was wondering...is it possible to scramble a service and be able to connect to two ECMG provided the same control words with different Access Criteria?

Best Regards,

Adam

lelegard, 2019-10-17:
Hi @adambeveridge1 

> is it possible to scramble a service and be able to connect to two ECMG provided the same control words with different Access Criteria?

No, not with the `scrambler` plugin as it is today. The plugin generates the random CW on the fly and inserts only one ECM stream. This is the typical requirement for a test, debug or integration scenario, which is the TSDuck target. Implementing an actual simulcrypt with several CAS and several ECM streams is more a production requirement (not a target for TSDuck).

That being said, starting from the source code of the `scrambler` plugin, you could develop your own symulcrypt plugin. Take care that this plugin is a complex one with some synchronization issues.

adambeveridge1, 2019-10-17:
Thank you!

Best Regards,

closed 2019-10-17 by adambeveridge1