tsduck/tsduck #432: Bitrate results are different since 3.18 with "analyze" plugin

apallier, 2019-10-18:
Hi,

I experiment a new behavior since TSDuck 3.18 with  "analyze" plugin.

For example,  the "ts-analysis" result with 3.16 was:

```
{'bitrate204': 32553189,
 'bitrate': 29999998,
 'pcrbitrate204': 32553189,
 'pcrbitrate': 29999998,
 'userbitrate': 0,
 'userbitrate204': 0
...
}
```
and now with 3.18:

```
{ 'bitrate204': 32878357,
 'pcrbitrate204': 32553189,
 'bitrate': 30299663,
 'userbitrate': 30299663,
 'userbitrate204': 32878357,
 'pcrbitrate': 29999998
...
}
```

I think it's strange that `userbitrate` has a value even if I don't provide a user-specified value (exactly the same command than 3.16).

Do you have an idea of such behavior change ?

Thanks in advance for your explanation.

lelegard, 2019-10-19:
Hi @apallier,

There are two topics here: 1) slightly different bitrate results, 2) what is user bitrate.

Concerning the first one, there was a fix in 3.17 which is described in the `CHANGELOG.txt` file:
~~~
  * Fixed bitrate computation based on PCR's. Null packets uselessly reset the
    bitrate computation and introduced a small bias. New computed values can be
    slightly different but more accurate.
~~~

Concerning the second one, the TS analyser gets two bitrate information. The first one is computed by the TS analyser itself, based on PCR analysis and is called "PCR bitrate". The second one comes from outside the TS analyser and is named "user-specified bitrate". With the command `tsanalyzer`, the user-specified bitrate comes from the option `--bitrate`. With the `analyze` plugin, it comes from the `tsp` framework. Typically, when the original input is a hardware device with a fixed bitrate (tuner or ASI), this is the nominal bitrate of the device. See the TSDuck user's guide for an explanation about the propagation of the bitrate information in `tsp`. 

I cannot explain why it was zero in 3.16. Maybe there was some situation where is was zero that was fixed with a the other bitrate fixes in 3.17. I do not have an explanation on this specific point, sorry.

I see that you use JSON format. I assume that you convert it from the `--normalized` output, isn't it?

apallier, 2019-10-19:
Hi @lelegard,
Thank you for your answer. I have now a better understanding of how it works.

Actually, my results are not exactly JSON format but Python dict (it's mostly the same). And yes, they come from parsing of the normalized output.

closed 2019-11-04 by apallier