tsduck/tsduck #433: Linux 32-bit compilation error

samuraibrian, 2019-10-18:
Ubuntu 18.04 32-bit. Trying to compile from git clone. Followed instructions at https://tsduck.io/doxy/building.html. Did have to install older gcc / g++ - (had to go for 4.8, as g++-4.9 not available via apt).
Ran:
`sudo make CC=gcc-4.8 GCC=gcc-4.8 CXX=g++-4.8 AR=gcc-ar-4.8`
which ran without error until:

  [DEP] tstabdump.cpp
  [CXX] tsanalyze.cpp
  [LD] release-i386/tsanalyze
/home/me/Downloads/tsduck/src/libtsduck/release-i386/tsduck.so: undefined reference to `ts::DescriptorList::fromXML(ts::DuckContext&, std::vector<ts::xml::Element const*, std::allocator<ts::xml::Element const*> >&, ts::xml::Element const*, std::__cxx11::list<ts::UString, std::allocator<ts::UString> > const&)'
collect2: error: ld returned 1 exit status
../../Makefile.common:910: recipe for target 'release-i386/tsanalyze' failed
make[2]: *** [release-i386/tsanalyze] Error 1
Makefile:43: recipe for target 'default' failed
make[1]: *** [default] Error 2
Makefile:58: recipe for target 'default' failed
make: *** [default] Error 2

Anything obvious, I got wrong?

Jon

lelegard, 2019-10-19:
Hi @samuraibrian,

If you compiled with a given version of the compiler and then recompiled on top of a previous compilation with another version of the compiler, you can run into ABI compatibility issues. Try to cleanup first (`make distclean`) and then recompile. I cannot guarantee that this is your problem, just a guess.

If you pulled a recent repo, I added workarounds to avoid GCC 6/7 bugs. So, TSDuck should compile with Ubuntu 18.04 default compiler. I should update the building instructions. So, you can have a try.

But, in all cases, do not forget to `make distclean` before switching to another compiler.

An BTW, compiling using the root account is never a good idea...

samuraibrian, 2019-10-19:
Hi Thierry. Thanks for the reply. I had a number of issues, not least of which was that the clone folder was owned by root (hence needing to `sudo make`) - so I fixed that.
Installed GCC 9, and used that for a (non-root!) compile (before your reply). That went fine, as did the `sudo make install`. The only remaining problem is that, if I access my DVB-T/T2 USB stick (PCTV 290e) with tsp, I have to `sudo` it. (Definitely not necessary on my 64-bit installed-from-deb-binary machine.)
Thanks for your help. Will close this issue, unless you want to respond to the root requirement thing...

BR, Jon

lelegard, 2019-10-19:
If you made a proper `make install`, the udev rules and perms files are installed. I would expect the `/dev/dvb/adapter` files to be accessible from users in the `video` group after reboot

samuraibrian, 2019-10-19:
Thanks! For some reason, I wasn't a member of the `video` group...

BR, Jon

closed 2019-10-19 by samuraibrian