tsduck/tsduck #434: Triggered 'ring buffer' recording

samuraibrian, 2019-10-20:
Sorry for posting as an 'issue' - there's no mailist for exchanging usage examples...
Requirement: to capture SPTS for a few seconds, either side of a PMT change.
Proposal:
[image]

1) DVB-T tune: zaps a specific service
2) buffer: delays service by configurable amount (e.g. 1,000 packets)
3) PMT ver. no. change: monitors PMT:
   tsp -I dvb --uhf-channel 25 -P tables --flush --log -p 430 -O drop
   Only outputs new line when PMT changes. Programmatically detect the
   change, and issue a udp RC command to tsswitch to pass 2,000 packets
   from the buffer.
4) tsswitch: normally not receiving from buffer. Only switched to
   buffer by udp command from 3)
Thinks that's possible with a bash script (all I know!). However, I can't figure-out the file output bit (open / close .ts file, with a unique name, each time there is a PMT change). Is this the right way to go about the requirement?

BR, Jon

lelegard, 2019-10-20:
Hi @samuraibrian,

No problem, this area is routinely used as an exchange forum, not only to report bugs or "issues".

You precisely describe _how_ you want to do something. You should start by describing _what_ you want to do, what is your target purpose. We may suggest slightly different options to achieve the same goal.

samuraibrian, 2019-10-20:
Hi Thierry - 'target purpose' (I kept it brief!) was this bit "Requirement: to capture SPTS for a few seconds, either side of a PMT change." It's a troubleshooting application for 'timex' services. The most recent issue was a service that exhibited frequent PMT updates (typically, switching encryption on/off for ad breaks); we suspected there was a CAS issue, and that V/A encryption was not synchronised with the PMT changes. Another one was a service that issued a new PMT every time the source Video Aspect Ratio changed.
It's handy to be able to 'snapshot' the service either side of the PMT Version No. change, for offline inspection.
Ultimately, the 'trigger source' might be anything diagnostically useful (not just a PMT version no. change) - might make the selection of that sort of thing a bash script menu!

Thanks for your help, Jon

lelegard, 2019-10-20:
OK, that makes sense. In the past, I already had to investigate similar problems. It depends on whether you want to make it a permanent 24x7 monitoring tool, collecting all samples of PMT change, or if you want to investigate it during a limited period of time.

In the case of a permanent monitoring tool, we need to find some robust solution.

If you need to investigate a limited period of time (which was my need in similar problems), I did the following:

- Record a complete SPTS (removing all other services for obvious disk space constraint). Do `dvb` -> `zap` -> `tables` (to be informed of PMT change) -> `file`.
- Stop the recording after one or two PMT changes.
- Inspect the file using the `history` plugin. See the various options and examples in the user's guide. See how to sort event in chronological order. See how to add packet numbers.
- Use plugins `skip` and `until` using packet numbers and index to extract a segment of the SPTS around the PMT change event.

I already did that in the past with success. Of course, this is more a one-shot solution that a permanent one. It depends whether you need troubleshooting or monitoring.

samuraibrian, 2019-10-20:
That would work! Still not happy though! There have been cases where I've had to wait maybe 16 hours before an unpredicted event occurred. So, that's quite a few GB (not impossibly large, but 'unwieldy') to inspect.
I'll have a go at a bash script to do what you suggested - it'll help me learn more about tsduck, of course!
Then, maybe I could attempt my more 'automated' proposal... (where I end-up with a small 'library' of auto-generated .ts files, each one containing <configurable_number_of> packets either side of the trigger).

Thanks, Jon (feel free to close thread)

closed 2019-10-20 by lelegard