tsduck/tsduck #435: IDEA: NULL stuff for HLS input

lars18th, 2019-10-22:
Hi @lelegard ,

I'm triying to transform some HLS stream (in fact, I'm testing with `https://mcdn.daserste.de/daserste/de/master.m3u8`) to craete a suitable full MPEG-TS compatible with DVB-T. However, as the HLS input doesn't have NULL stuff, it's impossible to add missing tables.

So, please can you add some option in the HLS input plugin to add NULL stuff?

Thank you!

lars18th, 2019-10-22:
I missed to comment the option... I suggest to use the same as for the global tsp `-a nullpkt/inpkt
--add-input-stuffing nullpkt/inpkt`. 

So you can use:
`tsp -I HSL --add-input-stuffing 1/10 https://mcdn.daserste.de/daserste/de/master.m3u8 ... `
to add 1 NULL packet every 10 readed.

I hope you agree.
Regards.

lelegard, 2019-10-22:
This is exactly what the option `--add-input-stuffing` is already doing. Did you try it?

lars18th, 2019-10-22:
> This is exactly what the option `--add-input-stuffing` is already doing. Did you try it?

Oppss! I discovered now... I think it only works with other inputs.

Thank you!

closed 2019-10-22 by lars18th