tsduck/tsduck #436: Build with musl libc

rossixolit, 2019-10-22:
I am building tsduck on Alpine Linux, which has the musl libc. There are some places where the build fails, as it tries to use glibc specific functions.

Those are:
1) ioctl is called the glibc way with ioctl(int, unsigned long int, ...), musl/POSIX has ioctl(int, int, ...) [https://pubs.opengroup.org/onlinepubs/009604599/functions/ioctl.html]
2) HAVE_INT_STRERROR_R is defined for TS_MAC, but musl would need it POSIX-compliant too. 
3) In tsThread, only TS_MAC is using sched_yield, otherwise pthread_yield is used. Linux seems to call sched_yield for pthread_yield anyways, and that way it works with musl too. [http://man7.org/linux/man-pages/man3/pthread_yield.3.html]

I have quick-fixed those issues for my use-case (as in "not good for a pull request, I'm not a c++ dev"), but it would be great if someone could look into those issues and to have support for musl libc upstream (or even an Alpine package =) ).

lelegard, 2019-10-22:
Hi @rossixolit,

I never used Alpine Linux. So, I quickly installed it on a VirtualBox VM (vbox display not working perfectly though). I see the issues. This is really an inconsistent distro. In the system headers, it defines ioctl request codes which are incompatible with its own declaration of ioctl(). If defines a POSIX-compliant version of strerror_r while it says that it is a GNU non-POSIX compliant context. Completely insane...

I fixed a couple of issues in a clean way. I still do not know how to decide - in a portable way - that strerror_r is a POSIX version while it says it is not.

I will let you know when the fix are complete.

But don't count on me to develop an Alpine apk package. This is not the kind of distro I feel like investing on :(

lelegard, 2019-10-22:
And also, after installing Alpine man pages, `man pthread_yield` is documented but the function is not present. Really inconsistent distro....

rossixolit, 2019-10-23:
Wow, thanks for the quick reaction!

About strerror_r:
I read a suggestion to overload a function for that [https://stackoverflow.com/questions/41953104/strerror-r-is-incorrectly-declared-on-alpine-linux] and also, that strerror_r has a complicated history and should be replaced by strerror_l [http://austingroupbugs.net/view.php?id=655]

Btw.: We are using Alpine Linux to build a docker image with a go server, ffmpeg and tsstuff which is just 72MB. So it's true that it's sub-optimal for building, but it's good for small, fast, resource-efficient containers.

lelegard, 2019-10-23:
I fixed the strerror and pthread_yield issues in a portable way.

Then, I discovered that supporting Dektec devices will not be possible on Alpine because the DTAPI is delivered in binary format by Dektec and was compiled with glibc. It contains references to low-level glibc symbols that are not available with musl libc (namely `__isnan`). So, I disabled DTAPI on Alpine (same as ARM Linux platforms such as Raspberry for instance, for which DTAPI is not available).

Now, some TSDuck unitary tests fail (while they all pass on macOS, Windows and all other Linux distros including ARM platforms). I need to investigate.

Apart from that, I could not make Alpine Linux work perfectly in VirtualBox. The guest additions from vbox do not compile and using the guest additions from Alpine repositories does not allow virtual screen resizing, making it barely useable for development (fine for compiling on a small terminal, useless with emacs editor). So, I use the Alpine VM through ssh sessions from another Fedora VM. The worst Linux distro I have ever seen. Really not a distro I feel like investing on....

lelegard, 2019-10-23:
Hi @rossixolit,

I pushed various fixes last night to compile on Alpine Linux. The [global test suite](https://github.com/tsduck/tsduck-test) passes. So, TSDuck - as a set of applications - seems to work on Alpine Linux 3.10.3.

However, the unitary tests fail sometimes because of an out-of-sequence memory corruption in some low-level threading operations. My investigations did not produce any evidence even though unitary tests are simple by definition. Googling a bit, I have seen a few reports on incompatibilities on threading between GNU libc and musl libc. I cannot guarantee that the problem is not in TSDuck. I can only say that TSDuck has been running for ten years in operating systems which are as different as Windows, Linux (with GNU libc) and macOS (BSD, not GNU). The basic unitary tests have always passed everywhere.

So, I cannot guarantee that TSDuck works on Alpine Linux. If you encounter weird problems, including segmentation faults, it may be due to this kind of problems.

rossixolit, 2019-10-24:
Hi @lelegard 
Thank you very much for supporting Alpine, even though you don't seem to love it very much =)

closed 2019-11-04 by lelegard

lelegard, 2019-11-11:
Hi @rossixolit,

I come back to this issue a few weeks after closing it. I was not comfortable with the fact that there was a dirty crash in the unitary tests on Alpine. There is always a small risk that some nasty hidden bug remains in the code and wakes up only in a given environment. So I spent a few hours investigating the problem on an Alpine VM.

I found the root cause. The conclusion is that there is no _proven_ bug in musl libc but a clear lack of robustness compared to glibc or macOS.

The root cause of the crash is an interaction between `pthread_mutex_unlock()` and `pthread_mutex_destroy()`. The behaviour of `pthread_mutex_destroy()` is not clearly defined when the calling thread still holds the mutex when the same thread destroys it. The man pages do not specify this. But I have seen notes mentioning that the behaviour is _undefined_ in that case.

In the TSDuck application code, all mutexes are released exactly the same number of times they are acquired thanks to a `Guard` class. So, a mutex is never locked when destroyed and the problem never appears.

But the unitary test program is here to test individual classes and tests almost all possible cases, including borderline cases. In one test case, a mutex goes out of scope (ie. is destroyed) while being held by the calling thread. All other threading implementations (Windows, macOS, glibc) are robust to that case. However, the musl libc is not. The musl libc behaves the worst possible way: it _silently corrupts the memory_. The mutex destruction completes silently and successfully but the application memory is corrupted. Later, much later in the code, some other threading operation, on completely differents treads and mutexes corrupts the stack and the application code crashes on return.

I have modified the unitary test suite to remove this borderline situation so that it no longer crashes on Alpine Linux. And since this situation never appears in the application code because mutexes are managed through a `Guard` class which prevents this situation from happening, we can now feel safe on Alpine Linux as far as this problem is concerned.

You mentioned that I "don't seem to love Alpine very much". I have nothing specific against Alpine as a distro. However, I now do hate musl libc :)

I perfectly understand what an "undefined behaviour" is, even though this is not clearly specified in that case. But theory is one thing, practice and reality are different. The art of software includes robustness. Software shall be resistent to all kinds of flaws. TSDuck is designed to resist to most programming errors or ambiguities. If something weird appears, the software detects it and either correctly reports it or recovers. But not detecting an inconsistency and, worse, silently corrupting the memory is the worst sin of software.

I will remember the lesson: keep away from musl libc !