tsduck/tsduck #437: Request: Interleave merge

lars18th, 2019-10-24:
Hi @lelegard ,

At time I'm very satisfied with the `interleave` parameter of the `FILE` input. With this approach is quite easy to mux different SPTS. Using the `tsstuff` tool it's possible to convert _any_ SPTS to a fixed CBR and then mux any number of thse SPTS with the same bitrate using the `interleave` approach. Finally using the `reduce` plugin it's possible to remove all NULL stuff and target a final CBR for the MPTS. To complete the processing at the end it's mandatory to use the `pcradjust` plugin to smooth the PCRs in the MPTS.

So, now I want to try to reproduce this in real-time. But the `merge` plugin lacks the `interleave` functionality. The idea is very simple: read 1 packet from the PIPE and 1 packet from the CHAIN. And if one of the queues is empty then inject a NULL packet.

What you think?

lelegard, 2019-10-24:
It does not work like that.

The `file` plugin is an input plugin. It is the source of the chain. It has complete freedom on which packet to inject and when. Everything is possible (in theory).

The `merge` plugin is a packet processor plugin. It never inserts new packets. It can only transform (or delete) existing packets in the chain. As any other packet processor plugin which needs to insert something, it can only replace existing null packets. In other words, if you use the `merge` plugin on a stream where there is no null packet, nothing will be merged. The `merge` plugin has no freedom to decide when to insert a packet from the merge stream. It is dependent on the content of the main stream from the chain.

lars18th, 2019-10-24:
Hi @lelegard ,

Yes, you're right!

What then of a `pipe` input with interleave functionality? The idea will be a new input module that creates _N_ child process and reads from all of them. Could that work?

lelegard, 2019-10-24:
> What then of a pipe input with interleave functionality? 

I would prefer an `--interleave` option to `tsswitch`. This would be more generic. When I have time...

lars18th, 2019-10-24:
> > What then of a pipe input with interleave functionality?
> 
> I would prefer an `--interleave` option to `tsswitch`. This would be more generic. When I have time...

Good idea! In this case (when you have time, no problem), the `tsswitch` will read from _all_ inputs at time (or from a subset of them)?

lelegard, 2019-10-24:
>  `tsswitch` will read from all inputs at time (or from a subset of them)?

All of them, this is the purpose of `--interleave`.