tsduck/tsduck #438: PIPE timeout

lars18th, 2019-10-24:
Hi @lelegard ,

I'm trying to use the `tsp` binary inside a child subprocess from another process (nodejs script). And I need one functionality that perhaps it's easy to implement (I don't know it).

The request is about to "auto quit when the PIPE is stalled". So, if the parent process isn't reading from the pipe for _X_ seconds, then the `tsp` process ends.

What you think? It has sense and it's easy to implement?
Thank you for your support.

lelegard, 2019-10-24:
It could be done at `tsp` level. Thus, it would work for any input plugin. If no input is receive within XX ms, then abort (ie. emulate an end of file). `tsswitch` does this already for failover purposes.

Note that there is no `pipe` plugin. This is just the `file` plugin reading on the standard input (the default when no input plugin is specified).

lars18th, 2019-11-06:
Hi @lelegard ,

I feel this commit https://github.com/tsduck/tsduck/commit/95275aa85c057e5f8f2355671287a24cd1dbf01a implements this, right?

lelegard, 2019-11-06:
> I feel this commit 95275aa implements this, right?

Indeed, but I didn't have time to test it yet.

lelegard, 2019-11-08:
Hi @lars18th,
Could you confirm now if the latest repo works for you?

closed 2019-12-02 by lelegard