tsduck/tsduck #439: ATSC_EIT

thrower77, 2019-10-24:
First time on github so bare with me...

Trying to use tspacketize to convert an atsc_eit xml into ts. Her is the first snip from my xml:
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
<atsc_eit version="0" source_id="1" protocol_version="0">
<event event_id="1" start_time="2019-10-24 00:00:00" ETM_location="1" length_in_seconds="3600">
<title_text>
<string language="eng" text="Program Guide Test Program: 1000"/>
</title_text>
<_any in="_descriptors"/>
</event>

but tspacketize is throwing "tspacketize: unexpected node <atsc_eit> in <tsduck>, line 3" error.

This is the windows 32bit 3.19-1410 build. 

I'm guessing I need to switch over to a linux build of some kind? im just testing right now -- if atsc_eit isnt working anymore, there's no point for me to load up a demo box.

Thanks

thrower77, 2019-10-24:
it chopped off my snip.:
<?xml version="1.0" encoding="UTF-8"?>
	<tsduck>
		<atsc_eit version="0" source_id="1" protocol_version="0">
			<event event_id="1" start_time="2019-10-24 00:00:00" ETM_location="1" length_in_seconds="3600">
			<title_text>
				<string language="eng" text="Program Guide Test Program: 1000"/>
			</title_text>
			<_any in="_descriptors"/>
		</event>

lelegard, 2019-10-24:
Sorry, there is a typo in the user's guide. The expression `<_any in="_descriptors"/>` is used in XML model files only. It means that is must be replaced by a list of zero or more descriptors. I will fix the user's guide.

lelegard, 2019-10-24:
Your message means that `<ATSC_EIT>` is not recognised as a valid table. The complete ATSC signalisation was added piece by piece in the development of version 3.19.  In commit 1410, the ATSC EIT was not yet implemented. You need to build a more recent one from the repo. [The instructions are here](https://tsduck.io/doxy/building.html), it uses only free software which can be installed on any Windows machine.

The last official version is 3.18. Version 3.19 is still under development. I may build a more recent pre-release of 3.19 on Windows, but not before this week-end.

thrower77, 2019-10-24:
I kept poking around a bit and noticed that was the case. I just finished spinning up an Ubuntu VM and im compiling the latest source now. Thanks for your work on this!

closed 2019-11-04 by lelegard