tsduck/tsduck #440: logical_channel_descriptor 0x83 not recognized

enricovittorini, 2019-10-25:
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.18-1361

tspsi.exe test.ts
The descriptor tag 0x83 is not decoded

```
* NIT Actual, TID 64 (0x40), PID 16 (0x0010)
  Version: 2, sections: 1, total size: 88 bytes
  - Section 0:
    Network Id: 318 (0x013E)
    Network information:
    - Descriptor 0: Network Name (0x40, 64), 3 bytes
      Name: "Test"
    Transport Stream Id: 12400 (0x3070), Original Network Id: 318 (0x013E)
    - Descriptor 0: Satellite Delivery System (0x43, 67), 11 bytes
      Orbital position: 13.0 degree, east
      Frequency: 10.99200 GHz
      Symbol rate: 27.5000 Msymbol/s
      Polarization: linear - vertical
      Modulation: DVB-S, QPSK
      Inner FEC: 2/3
    - Descriptor 1: Service List (0x41, 65), 24 bytes
      Service id: 272 (0x0110), Type: 0x0C (Data broadcast service)
      Service id: 273 (0x0111), Type: 0x0C (Data broadcast service)
      Service id: 274 (0x0112), Type: 0x0C (Data broadcast service)
      Service id: 8511 (0x213F), Type: 0x19 (Advanced codec HD digital television service)
      Service id: 8512 (0x2140), Type: 0x01 (Digital television service)
      Service id: 8513 (0x2141), Type: 0x01 (Digital television service)
      Service id: 8516 (0x2144), Type: 0x01 (Digital television service)
      Service id: 8557 (0x216D), Type: 0x02 (Digital radio sound service)
    - Descriptor 2: unknown (0x83, 131), 20 bytes
      0000:  21 6D FC 01 21 40 FC 02 21 3F FC 03 21 41  !m..!@..!?..!A
      000E:  FC 04 21 44 FC 0B
```

lelegard, 2019-10-25:
Your NIT is invalid.

In a DVB network, descriptor ids 0x80 and higher are private descriptors. To be correctly interpreted, they must be preceded by a _private_data_specifier_descriptor_ which establishes the context of the subsequent private descriptors, up to the next _private_data_specifier_descriptor_.

In your case, this is probably an EACEM/EICTA-defined _logical_channel_number_descriptor_. But 0x83 is also used in Eutelsat networks as a _channel_number_descriptor_. The two descriptors have a similar semantics (assigning numbers to TV channels) but their syntax is different.

If you built this NIT, fix it. If you collected this NIT from an actual network, then it is incorrect. This situation is unfortunately not unusual. This is the reason why commands such as `tstables`, `tspsi` and the like have an option named `--default-pds` with which you can force a default private data specifier when an actual one is missing.

In your case, it the 0x83 is the EACEM-defined private descriptor, then use `--default-pds 0x00000028`. If this is the Eutelsat one, then use `--default-pds 0x0000055F`.

enricovittorini, 2019-10-27:
Understood.
i tried to use 
`tstables.exe -p 16 ive.ts --xml-output test1.xml --default-pds EICTA `

tag **0x83** despite the --default-pds is not decoded. Maybe this is normal in XML otuput
```
  <generic_descriptor tag="0x83">
        00 03 FC 04 00 02 FC 05 00 01 FC 06 00 47 FC 07
        00 04 FC 0B 00 07 FC 0C 00 0A FC 10 00 09 FC 11
        00 08 FC 19 00 48 FC 1D 00 0C FC 27 00 06 FC 28
        00 0D FC 29 00 69 FE 67 00 68 FE 68 00 65 FE 69
        00 66 FE 6A 00 67 FE 6B 03 25 FF 25 03 83 FF 83
        00 0E FC 3C 00 0F FC 14
      </generic_descriptor>
```

lelegard, 2019-10-28:
See an example using a hand-written XML file.
The _private_data_specifier_descriptor_ is intentionally missing.
~~~
$ cat test.xml
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <NIT network_id="0">
    <service_list_descriptor>
      <service service_id="10" service_type="1"/>
      <service service_id="20" service_type="1"/>
    </service_list_descriptor>
    <logical_channel_number_descriptor>
      <service service_id="10" logical_channel_number="1"/>
      <service service_id="20" logical_channel_number="2"/>
    </logical_channel_number_descriptor>
  </NIT>
</tsduck>
$ tspacketize test.xml -o test.ts -p 16
$ tstables test.ts 

* NIT Actual, TID 64 (0x40), PID 16 (0x0010)
  Version: 0, sections: 1, total size: 34 bytes
  - Section 0:
    Network Id: 0 (0x0000)
    Network information:
    - Descriptor 0: Service List (0x41, 65), 6 bytes
      Service id: 10 (0x000A), Type: 0x01 (Digital television service)
      Service id: 20 (0x0014), Type: 0x01 (Digital television service)
    - Descriptor 1: unknown (0x83, 131), 8 bytes
      0000:  00 0A FC 01 00 14 FC 02                    ........

$ tstables test.ts --default-pds eacem

* NIT Actual, TID 64 (0x40), PID 16 (0x0010)
  Version: 0, sections: 1, total size: 34 bytes
  - Section 0:
    Network Id: 0 (0x0000)
    Network information:
    - Descriptor 0: Service List (0x41, 65), 6 bytes
      Service id: 10 (0x000A), Type: 0x01 (Digital television service)
      Service id: 20 (0x0014), Type: 0x01 (Digital television service)
    - Descriptor 1: Logical Channel Number (0x83, 131), 8 bytes
      Service Id:    10 (0x000A), Visible: 1, Channel number:   1
      Service Id:    20 (0x0014), Visible: 1, Channel number:   2

$ tstables test.ts --xml /dev/stdout
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <NIT version="0" current="true" network_id="0x0000" actual="true">
    <!-- PID 0x0010 (16) -->
    <service_list_descriptor>
      <service service_id="0x000A" service_type="0x01"/>
      <service service_id="0x0014" service_type="0x01"/>
    </service_list_descriptor>
    <generic_descriptor tag="0x83">
      00 0A FC 01 00 14 FC 02
    </generic_descriptor>
  </NIT>
</tsduck>
$ tstables test.ts --xml /dev/stdout --default-pds eacem
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <NIT version="0" current="true" network_id="0x0000" actual="true">
    <!-- PID 0x0010 (16) -->
    <service_list_descriptor>
      <service service_id="0x000A" service_type="0x01"/>
      <service service_id="0x0014" service_type="0x01"/>
    </service_list_descriptor>
    <logical_channel_number_descriptor>
      <service service_id="0x000A" logical_channel_number="1" visible_service="true"/>
      <service service_id="0x0014" logical_channel_number="2" visible_service="true"/>
    </logical_channel_number_descriptor>
  </NIT>
</tsduck>
$
~~~

enricovittorini, 2019-10-29:
Yes clear thanks a lot.
Do you confirm Nordig 0x83 is not supported?

lelegard, 2019-10-29:
> Do you confirm Nordig 0x83 is not supported?

It is not. I just had a quick look at the NorDig "Rules of Operations" (v2.5). Their logical channel descriptor has exactly the same layout as the EACEM one. But the PDS shall be 0x29 (the NorDig one) instead of the 0x28 (the EACEM one). So, a _private_data_specifier_descriptor_ is required.

The NorDig spec about logical channel descriptor is a bit weird. The syntax is unclear. The visible flag in v1 does not appear in the diagram. The v2 has no diagram. Without more information, it is not possible to implement the NorDig version of this descriptor. If you have a clear non-ambiguous spec, it can be added in TSDuck.

lelegard, 2019-10-29:
I have found the document named "NorDig Unified Requirements ver. 3.1.1". In section 12.2.9, the  syntax of the two versions is clearly explained. It will be supported by TSDuck when I have some time.

lelegard, 2019-10-30:
Done. (I am currently traveling in a train with nothing else to do...)

closed 2019-11-04 by lelegard