tsduck/tsduck #441: Ubuntu 18 build error

MarcusLindberg123, 2019-10-28:
Step by step on Ubuntu 18 machine:
1. Download TSDuck source
2. Follow instruction on https://tsduck.io/doxy/building.html:
   apt install g++ dpkg-dev doxygen dos2unix graphviz curl pcscd libpcsclite-dev libcurl4 libcurl4-openssl-dev
3. make deb

Result:
$ sudo make deb
make -C build deb
make NOTEST=true -C /home/CORPUSERS/23055076/repos/tsduck -j12
Makefile:82: *** Invalid file operation: <LinuxSDK.version.  Stop.
Makefile:58: recipe for target 'default' failed
make[2]: *** [default] Error 2
Makefile:129: recipe for target 'deb-main' failed
make[1]: *** [deb-main] Error 2
Makefile:93: recipe for target 'deb' failed
make: *** [deb] Error 2

Seems to be something wrong in dectec/Makefile

There is no problem building on Ubuntu 16, so I guess I will use it. Is there any side effects installling a debian built on Ubuntu 16 on a Ubuntu 18? There was no error when I tried.

lelegard, 2019-10-28:
Hi @MarcusLindberg123,

I made a recent modification to simplify the dektec configuration. Unfortunately, this simplification relied on [a feature which was introduced in GNU Make 4.2](https://fossies.org/diffs/make/4.1_vs_4.2/NEWS-diff.html). I assume that Ubuntu 18 has an older version.

I just modified the Makefile to make sure it no longer uses that feature of GNU Make 4.2. The build should work now. Sorry for the inconvenience.

MarcusLindberg123, 2019-10-29:
I tested and the fix worked. Thank you for the quick fix!

We need a local repository with debians. Is there any difference between the debians depending in Ubuntu versions?

lelegard, 2019-10-29:
> Is there any difference between the debians depending in Ubuntu versions?

Generally speaking, a package which was build on version X of a distro will work on higher versions (with some exceptions when a distro version breaks backward compatibility) while it is quite possible that it fails on lower versions because the package depends on "future" ABI's in some external libraries.

As a rule of thumb, when you have different versions of a distro, build on the oldest one and deploy everywhere.

MarcusLindberg123, 2019-10-29:
Great advice. Thank you!

closed 2019-10-29 by MarcusLindberg123