tsduck/tsduck #442: Accuracy of the pcradjust plugin

matlo, 2019-10-29:
Hello,

I observe PCR/PTS accuracy issues (mainly drift) while using the pcradust plugin.

I had a look at the source code and I identified the following points as potential inaccuracy causes:
* truncating the computed PCR in the ts::NextPCR method; the error accumulates over time, creating drift
* wrapping PCR in ts::NextPCR by subtracting MAX_PCR; I think the subtracted value should be (MAX_PCR + 1); the ts::DiffPCR and ts::DiffPTS which are not used by the pcradjust plugin also have similar wrapping issues (MAX_PCR instead of MAX_PCR + 1 and MAX_PTS_DTS instead of MAX_PTS_DTS + 1)
* using an integer for the bitrate; this causes drift when the real bitrate is not an integer; the drift is more impacting for very low bitrates, when the truncated decimals are close to 1

matlo, 2019-10-31:
4. placing PTS relative to PCR; -I file --infinite produces jitter (+/- 1 @ 90KHz) on PTS period after first loop; this is due to the PCR/PTS loss of sync detection activating a different code path

lelegard, 2019-11-03:
Hi @matlo,

You are right on most points. I should review the handling of PCR's and bitrates in depth to increase the precision. But I do not want to do it in the `pcradjust` plugin alone. It should be applied everywhere because PCR and bitrates are deeply intricated in many places. So, I should create some general-purpose class which can be used everywhere, for all kinds of interactions between PCR's and bitrates. I will think about it but I need some quite time to do this properly.

Instead of using fractions, it would be more practical to use "fixed point" values. In C/C++ where there is no such fixed point types, we use a fixed factor. For instance, PCR and bitrate values could be manipulated with a fixed factor of 1,000,000. Thus, the base inaccuracy would be one micro-PCR only, which seems neglictible to me, instead of one PCR unit.

I propose a factor of 1,000,000 because a factored 42-bit PCR value would fit on 62 bits and could still be computed on 64-bit unsigned integer types.

Concerning item 2, `PCR_SCALE` should have been used instead of `MAX_PCR` (same for PTS). I just fixed it. For my own defence, this code came from an external contribution but I plead guilty for not having correctly reviewed that code. That being said, this error introduced a bias of only 0.000000037 second every 26 hours.

About item 4, I think I have fixed it in method `updatedPDTS()`. Please check it addresses your point on a real stream.

matlo, 2019-11-04:
Hi @lelegard,

Thanks for having a look at this, and for the fixes!

A fixed factor of one million looks good enough.

I am unsure your changes will fix point 4. Let me explain my use case. I am using tsduck to stream a ts file that contains a series of PAT+PMT+PES packets (3*188 bytes) being sent at 29.97 per second. Each PES packet contains a PCR and a PTS, and a timestamp in the payload. PCR has a bit of jitter in the ts file. This jitter is fixed by the pcradjust plugin. When sync_pdts becomes true the PTS values are changed relative to the PCR. This translates the PCR jitter to a PTS jitter.

Instead of placing the PTS relative to the PCR, maybe it could be placed relative to the previous value, taking the bitrate adjustment into account?

lelegard, 2019-11-04:
> I am unsure your changes will fix point 4.

I think it addresses the nominal case of reaching the end of PCR/PTS range, when the PCR just wraps down to zero while the PTS is still near the end of range (or the other way round). This case is expected and was not correctly addressed previously.

> a series of PAT+PMT+PES packets (3*188 bytes) being sent at 29.97 per second.

Well, this is hardly a typical case. This is not only a matter of bitrate but having as many PES packets as PAT or PMT packets... If this can be addressed as part as a general case, that is fine. But we cannot afford addressing all pathological cases individually.

>  When sync_pdts becomes true the PTS values are changed relative to the PCR. This translates the PCR jitter to a PTS jitter.

I assume you mean "when sync_pdts becomes false".

When sync_pdts becomes false, this is the symptom of some kind of "time warp" in the PCR. There are many reasons for such a "time warp". Each case would probably need a specific treatment for PTS/DTS. But it looks impossible for the `pcradjust` plugin to determine the exact cause. So, whatever processing we use, it will be correct in some cases and incorrect in others.

Looping a file is such a cause. Probably the most probable one. Switching to a new TS is another one. At the point of the loop or switch, we cannot know how to exactly process the PTS and DTS. So, there is one glitch here. But there won't be any other glitch until the next loop or switch (not really sure, I have to check and fix if needed).

If your file has a small series of PAT+PMT+PES with a PCR in every PES, then you are clearly maximising the probability of a glitch.

lelegard, 2019-11-04:
> Instead of placing the PTS relative to the PCR, maybe it could be placed relative to the previous value (with a PCR precision), taking the bitrate adjustment into account?

I do not think that this is correct. The PCR is the clock of the TS and it can be computed from a previous PCR and a bitrate. But PTS and DTS are going back and forth, depending on the decoding process (I/P/B frames). In the general case, PTS are even not in a sequential order.

matlo, 2019-11-06:
Hi @lelegard,

Thanks for the answers.

Yes, I meant "when sync_pdts becomes false". I understand that having a perfect result for all situations is nearly impossible, but still, I think plugin parameters can help. The plugin can be told how to behave on special events.

About PTS/DTS placement, I did not clearly explain my thought. A non-constant PTS interval is something that can cause compliance issues. If PCR rate is adjusted by some factor (adjusted bitrate / original bitrate), then applying the same factor on the intervals between successive PTS/DTS values in ts stream could help keeping PTS/DTS in sync with the PCR. Otherwise they will just drift until sync_pdts becomes false, after what the PTS interval will start varying with PCR jitter.

I'm using tsduck for testing purposes, so I'll probably make a pcradjust2 for my own needs, and experiment. I'll let you know the outcome.