tsduck/tsduck #444: Issue with HEVC video descriptor

dehaanw, 2019-11-05:
Hi, 
First of all I'd like to thank you for creating this great tool with superb documentation! I'm using it to update service information in a TS with an HEVC video stream by reading the original SI with tstables, modifying the xml file and then updating SI using the inject plugin. That works great!

I believe I found a bug in the implementation of the HEVC_video_descriptor (C.6.12). The two 3-bit parameters temporal_id_min and temporal_id_max should both be put (and read) ahead of 5 reserved bits (1), but TSDuck puts them after the reserved bits. See ISO/IEC 13818-1 Table 2-111.

dehaanw, 2019-11-05:
As a workaround I converted the xml file to a bin file (using tstabcomp), manually corrected the data with a hex editor and tried to inject the bin file into the TS. Unfortunately that results in an error message (invalid section).

lelegard, 2019-11-05:
Hi @dehaanw,

Thank you for reporting this. Are you sure about the layout of the HEVC video descriptor?

I have this:
[image]

I think that the TSDuck implementation is conformant with the diagram above. However, I have already seen several discrepancies between ISO 13818-1 amendments over time. Do you have another version?

dehaanw, 2019-11-06:
The order of the fields has changed at the end of the structure. The latest version from ISO/IEC (not for free) is from 2019, but the previous (2018) version from ITU-T can be downloaded here:
https://www.itu.int/rec/T-REC-H.222.0-201703-S/en
Modified part:

[image]

lelegard, 2019-11-06:
All right, thank you for the update. This is unfortunately not the first time I see incompatibilities between amendments and later integration of these amendment in ISO 13818-1.

I see that the updated version also contains new fields that were previously reserved bits.

I will update the TSDuck implementation and XML definition to include the new fields as well. Should be easy.

lelegard, 2019-11-06:
> As a workaround I converted the xml file to a bin file (using tstabcomp), manually corrected the data with a hex editor and tried to inject the bin file into the TS. Unfortunately that results in an error message (invalid section).

If you manually hack a long section, you must also recompute the CRC32. This is the reason for the invalid section.

dehaanw, 2019-11-06:
> I will update the TSDuck implementation and XML definition to include the new fields as well. Should be easy.

Thanks!

> If you manually hack a long section, you must also recompute the CRC32. This is the reason for the invalid section.

Oops, yes, good point. Thanks again!

lelegard, 2019-11-06:
Hi @dehaanw 

The HEVC descriptors have been fixed according to the latest version of the standard.

You can pull the repo and rebuild using [instructions here](https://tsduck.io/doxy/building.html).

dehaanw, 2019-11-07:
I rebuilt and tested TSDuck and confirm that this issue is solved. Thanks!

closed 2019-11-07 by dehaanw