tsduck/tsduck #445: TSAnalyzer memory leak

yaneti, 2019-11-05:
Hello,

Somewhere between 3.14 and 3.18 the analyzer seems to have developed a memory leak.
Here I am on Fedora 30, building tsduck master and I am running something like:
```
tsp -m -I ip -r --default-interface 224.15.15.15:1500 -P analyze -i8 --normalized -O drop
```
And the process is slowly but steadily growing.

Here is what `valgrind --tool=memcheck --leak-check=full`  has to say about it after 30 seconds:
```
==13507== 
==13507== HEAP SUMMARY:
==13507==     in use at exit: 83,626 bytes in 629 blocks
==13507==   total heap usage: 729,497 allocs, 728,868 frees, 45,178,261 bytes allocated
==13507== 
==13507== 11,452 (336 direct, 11,116 indirect) bytes in 14 blocks are definitely lost in loss record 25 of 29
==13507==    at 0x4838E86: operator new(unsigned long) (vg_replace_malloc.c:344)
==13507==    by 0x4C265EC: ts::TSAnalyzer::getPID(unsigned short, ts::UString const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4C28E52: ts::TSAnalyzer::analyzePAT(ts::PAT const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4C2DCAA: ts::TSAnalyzer::handleTable(ts::SectionDemux&, ts::BinaryTable const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4BD39B0: ts::SectionDemux::ETIDContext::notify(ts::SectionDemux&, bool, bool) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4BD464B: ts::SectionDemux::processPacket(ts::TSPacket const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4BD496D: ts::SectionDemux::feedPacket(ts::TSPacket const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4C2DE2B: ts::TSAnalyzer::feedPacket(ts::TSPacket const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x5776EB7: ts::AnalyzePlugin::processPacket(ts::TSPacket&, ts::TSPacketMetadata&) (in /home/builder/t/usr/bin/tsplugin_analyze.so)
==13507==    by 0x4072A6: ts::tsp::ProcessorExecutor::main() (in /home/builder/t/usr/bin/tsp)
==13507==    by 0x4C7898C: ts::Thread::mainWrapper() (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4C789DC: ts::Thread::ThreadProc(void*) (in /home/builder/t/usr/bin/tsduck.so)
==13507== 
==13507== 25,152 (768 direct, 24,384 indirect) bytes in 32 blocks are definitely lost in loss record 27 of 29
==13507==    at 0x4838E86: operator new(unsigned long) (vg_replace_malloc.c:344)
==13507==    by 0x4C265EC: ts::TSAnalyzer::getPID(unsigned short, ts::UString const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4C2BAE7: ts::TSAnalyzer::analyzePMT(unsigned short, ts::PMT const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4C2DC7D: ts::TSAnalyzer::handleTable(ts::SectionDemux&, ts::BinaryTable const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4BD39B0: ts::SectionDemux::ETIDContext::notify(ts::SectionDemux&, bool, bool) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4BD464B: ts::SectionDemux::processPacket(ts::TSPacket const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4BD496D: ts::SectionDemux::feedPacket(ts::TSPacket const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4C2DE2B: ts::TSAnalyzer::feedPacket(ts::TSPacket const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x5776EB7: ts::AnalyzePlugin::processPacket(ts::TSPacket&, ts::TSPacketMetadata&) (in /home/builder/t/usr/bin/tsplugin_analyze.so)
==13507==    by 0x4072A6: ts::tsp::ProcessorExecutor::main() (in /home/builder/t/usr/bin/tsp)
==13507==    by 0x4C7898C: ts::Thread::mainWrapper() (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4C789DC: ts::Thread::ThreadProc(void*) (in /home/builder/t/usr/bin/tsduck.so)
==13507== 
==13507== 42,346 (1,296 direct, 41,050 indirect) bytes in 54 blocks are definitely lost in loss record 29 of 29
==13507==    at 0x4838E86: operator new(unsigned long) (vg_replace_malloc.c:344)
==13507==    by 0x4C265EC: ts::TSAnalyzer::getPID(unsigned short, ts::UString const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4C2DE6B: ts::TSAnalyzer::feedPacket(ts::TSPacket const&) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x5776EB7: ts::AnalyzePlugin::processPacket(ts::TSPacket&, ts::TSPacketMetadata&) (in /home/builder/t/usr/bin/tsplugin_analyze.so)
==13507==    by 0x4072A6: ts::tsp::ProcessorExecutor::main() (in /home/builder/t/usr/bin/tsp)
==13507==    by 0x4C7898C: ts::Thread::mainWrapper() (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x4C789DC: ts::Thread::ThreadProc(void*) (in /home/builder/t/usr/bin/tsduck.so)
==13507==    by 0x501E4BF: start_thread (in /usr/lib64/libpthread-2.29.so)
==13507==    by 0x529F552: clone (in /usr/lib64/libc-2.29.so)
==13507== 
==13507== LEAK SUMMARY:
==13507==    definitely lost: 2,400 bytes in 100 blocks
==13507==    indirectly lost: 76,550 bytes in 515 blocks
==13507==      possibly lost: 0 bytes in 0 blocks
==13507==    still reachable: 4,676 bytes in 14 blocks
==13507==         suppressed: 0 bytes in 0 blocks
==13507== Reachable blocks (those to which a pointer was found) are not shown.
==13507== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==13507== 
==13507== For lists of detected and suppressed errors, rerun with: -s
==13507== ERROR SUMMARY: 3 errors from 3 contexts (suppressed: 0 from 0)

```

lelegard, 2019-11-06:
Hi @yaneti,

Thank you for reporting this. Well done!

I fixed the problem.

For those who are interested in the details, the problem was a memory leak in the _move assignment operator_ of the class `ts::SafePtr`. This template class is designed to implement generic thread-safe, auto-deallocating pointers. It is used everywhere in TSDuck. Using that _safe pointer_ class, you can write C++ code as you would write Java code, as far as memory management is concerned.

Being a fundamental class, I developed it a long time ago, before the advent of the C++11 standard. At the time, this class has been thoroughly tested and many application-level scenarios were run under valgrind control to make sure that the memory control was effective. And it was. Not a single byte was ever lost.

Nowadays, TSDuck relies on many new features of the C++11 standard which make the language much more robust. Amongst these new features, the _move semantics_ is very convenient to improve the performances through _move constructors_ and _move assignment operators_. Recently (and this is why you see a difference), I added a move constructor and a move assignment operator to the safe pointer class. __But__, I made a mistake in the move assignment operator which introduced a subtle memory leak. The code is functional, the performances are much better, especially for thread-safe smart pointers, but, but, but...

It is fixed now.

yaneti, 2019-11-06:
Thanks for the quick fix and the curious details @lelegard .
Applied it to a machine with a lot of tsp-s and memory usage is now steady.

closed 2019-11-06 by yaneti