tsduck/tsduck #447: EIT-PF: Suggestion plugin

lars18th, 2019-11-06:
Hi @lelegard ,

As the DVB specifications dictate to include EIT p/f in every MPTS, in order to create home-made DVB-T conformant MPTS broadcasts from user sources (cameras and recordings) I need to inject some very simple EIT p/f information. And the current `eit` plugin only extracts data.

So a new _simple_ plugin called `eit-pf` will be very useful. And I'm only suggesting for Present/Following information, not for the complex _Schedule_ data, it's clear that any schedule data is out-of-scope.

What you think?
Regards.

lars18th, 2019-11-12:
Hi @lelegard ,

If you don't want to implement this becaues the KISS paradigm... then please, let me know how to insert a table from XML with EIT p/f information. Any idea?

Thank you for this powerful project!

lelegard, 2019-11-12:
If I understand correctly, your need is:

- PID 18 (EIT) is originally empty.
- Some external tool generates files (XML or binary) containing EIT p/f.
- These files need to be cycled on PID 18.
- From time to time (typically at an event boundary), some or all of these files are regenerated with new EITp/f content. The new new EIT p/f shall replace the previous ones in PID 18.

Is that correct?

If this is correct, `-P inject --pid 18 --poll-files` should do the job. You could typically generate one EIT p/f file per service and use a wildcard as file name or use one single large EIT p/f for the MPTS.

lars18th, 2019-11-12:
Thank you @lelegard for your support!

> Is that correct?

Yes and no. I don't have any external tool that generates any XML EIT p/f table. I want to create one XML _manually_ with some basic info (for example: "Camera 1" as the title, for present event and similar for the following event). And then I want to insert it with TSDuck.

> If this is correct, `-P inject --pid 18 --poll-files` should do the job. You could typically generate one EIT p/f file per service and use a wildcard as file name or use one single large EIT p/f for the MPTS.

Great! So, then the `inject` plugin is the key to do it. However, I request if you can provide some _simple_ (really simple, nothing complex only the most basic version) plugin/tool that can generate such EIT p/f tables from the _basic_ XML.

If you don't agree, don't worry. Your work is impressive. However, I think a tool like this would be very useful.
Regards.

lelegard, 2019-11-12:
If you do not need to replace the files, you do not need the `--poll-files` option.

And the simplest tool to generate a simple EIT p/f XML file is a ... text editor 😉 with the unvaluable assistance of the copy/paste feature from the user's guide. The XML file can be directly injected by the plugin.

lars18th, 2019-11-12:
> And the simplest tool to generate a simple EIT p/f XML file is a ... text editor  with the unvaluable assistance of the copy/paste feature from the user's guide. The XML file can be directly injected by the plugin.

OK. But, where to find the example? In the TSDuck User's Guide I don't see any of such XML files (141 ocurrences of `eit` word, and only "C.3.4 Event Information Table (EIT)" has some info, but... I can use this for the XML?). Futhermore the `eit` plugin only _analyzes_ (aka extracts metadata) but not the content of the tables (XML files).

Please, with all respect: Can you provide some example or guide?
Thank you!

lelegard, 2019-11-12:
First, copy / paste the structure of an EIT from C.3.4 and replace the attribute values by real values. Then, decide how many events you want and which descriptors should be present according to DVB or national usage rules. Then copy / paste the structure of these descriptors from their corresponding section in appendix C.

For instance, the simplest form of EIT p/f, with one single event and one single short event descriptor is shown below :
~~~
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>

  <EIT type="pf" service_id="0x1234" transport_stream_id="0x5678" original_network_id="0x9ABC">
    <event event_id="0xABCD" start_time="2019-11-22 12:25:44" duration="01:15:34">
      <short_event_descriptor language_code="eng">
        <event_name>Event Foo</event_name>
      </short_event_descriptor>
    </event>
  </EIT>

</tsduck>
~~~

closed 2019-12-02 by lelegard