tsduck/tsduck #448: Suggestion: Portable mode for UNIX

lars18th, 2019-11-07:
Hi @lelegard ,

First let me to explain my use case: I'm calling to `tsp` from a nodeJS script. In my environment is quite complex to set the two environment vars required to run the TSDuck: `LD_LIBRARY_PATH` and `TSPLUGINS_PATH`.

Before opening this issue I have readed the old issue #13 about the impossibility of creating a full static binary of the `tsp` tool. So, I request for some solution for UNIX like the Portable mode for Windows.

When using the Portable version in Windows we only need to put inside the same directory the executables (\*.exe binaries), shared objects (\*.dll files) and extra files (\*.xml and similar). But in any UNIX machine we need to call to the `setenv.sh` script to set the two environment vars. Futhermore, in fact the `LD_LIBRARY_PATH` it's only used to found the `tsduck.so` library (the core). And the `TSPLUGINS_PATH` for searching the plugin shared objects (the other \*.so files).

So I suggest to add support for a new parameter (for UNIX) to use a hardcoded path. For example with `tsp --portable ...` we can overcome the environment vars and search _only_ in the same path where is running the tsp binary.

Please, can you consider this improvement?
I hope you will agree to implement some of this request.
Thank you!

lelegard, 2019-11-07:
There is no need for a portable mode on UNIX. It's already there, just install all files in the same place. This can be done manually or using a `make install` command as [explained here](https://tsduck.io/doxy/building.html#nonstdinst).

The `setenv.sh` is just a convenience to run the TSDuck tools directly from the build area, where the files are spread over the various directories. For a developer who cycles through build / test / modify / build / test... it is more convenient to skip the installation phase. But if you deploy a stable set of binaries, just use `make install` (or build a `.rpm` or `.deb` package if your distro supports them).

lars18th, 2019-11-07:
Hi @lelegard ,

Thank you for the tip! :wink: 
I never use `make install` as I prefer to _manually_ install _testing_ tools (that's I run from the source directory in my home or from the target directory).

After reading the documentation (thank you for it) I copied all required files to the same directory, and then the call to `tsp` (and other binaries) works without troubles.

Thank you a lot! :+1:

closed 2019-11-07 by lars18th