tsduck/tsduck #449: HLS file input support

lars18th, 2019-11-07:
Hi @lelegard ,

I'm testing the HLS output, and I lack the functionality of the HLS _file_ input. I want to use something like `tsp -I hls file:///test.m3u8 -P analyze -O drop` with the output (created with `tsp ... -O hls -d 1 -l 2 ...`)  but it doesn't work.

Any idea?
Thank you!

lelegard, 2019-11-07:
The HLS input plugin is based on the Web request class (libcurl or WinInet depending on the OS). Adding a layer to analyze the URL is some work with low value. It does not make sense in operation. I agree that it can be useful during a testing phase. But in that case, it is easy to setup a local Apache server which serves this files. This is the way I tested the HLS input plugin.

lars18th, 2019-11-07:
> But in that case, it is easy to setup a local Apache server which serves the files which are generated.

I'm using the httpd from the busybox... more easy to use than setup an Apache server. You just need to execute `busybox httpd -p <IP:PORT> -h <DIR_PATH>`.

lars18th, 2019-12-13:
Hi @lelegard ,

I reopen this issue as the use case using files has sense: I'm trying to do some processing based on _chunks_. So I use the HLS output plugin to generate the TS chunk files. And as you know this plugin writes _to the filesystem_. However, in order to recompose the stream _after_ another tool has processed a chunk, I can't read from the filesystem using `tsp -I hls file:///...`.

I feel the restriction to support HTTP sources _only_ it's quite hard. And as I know the cURL library supports `file:///` sources. So, why not consider it please?

I hope you want to reconsider it.
Thank you!

lars18th, 2019-12-13:
The trick to continue using the cURL library and the `ts::WebRequest` class is to use the `CURLPROTO_FILE` protocol with the `curl_easy_setopt()` function.

See:
https://curl.haxx.se/libcurl/c/CURLINFO_PROTOCOL.html
https://curl.haxx.se/libcurl/c/CURLOPT_DEFAULT_PROTOCOL.html

I feel this simplifies a lot the required changes.
It's true @lelegard or I'm missing something?

lelegard, 2019-12-13:
Seriously, did you try it?

There is nothing special to do. On macOS, Linux and Windows, it works out of the box. With some weirdness on Windows, as usual.

Sample session on my iMac:

~~~
$ tsp -I http https://tsduck.io/streams/france-astra-19.2E/astra192E-ts1080-2018-05-11.ts -O hls -p foo.m3u foo.ts
$
$ ls -l foo*
-rw-r--r--  1 thierry  staff       414 Dec 13 22:32 foo.m3u
-rw-r--r--  1 thierry  staff  72046676 Dec 13 22:32 foo000000.ts
-rw-r--r--  1 thierry  staff  72334504 Dec 13 22:32 foo000001.ts
-rw-r--r--  1 thierry  staff  74201532 Dec 13 22:32 foo000002.ts
-rw-r--r--  1 thierry  staff  72258552 Dec 13 22:32 foo000003.ts
-rw-r--r--  1 thierry  staff  71973544 Dec 13 22:32 foo000004.ts
-rw-r--r--  1 thierry  staff  73392380 Dec 13 22:32 foo000005.ts
$
$ tsp -v -I hls file:///Users/thierry/tmp/foo.m3u -P analyze -O drop
* hls: downloaded foo.m3u, media playlist, static, 6 segments, 10 seconds/segment
* hls: downloading from file:///Users/thierry/tmp/foo000000.ts
* hls: MIME type: unknown, expected size: unknown
* hls: unknown initial input bitrate
* hls: downloading from file:///Users/thierry/tmp/foo000001.ts
* hls: MIME type: unknown, expected size: unknown
* hls: downloading from file:///Users/thierry/tmp/foo000002.ts
* hls: MIME type: unknown, expected size: unknown
* hls: downloading from file:///Users/thierry/tmp/foo000003.ts
* hls: MIME type: unknown, expected size: unknown
* hls: downloading from file:///Users/thierry/tmp/foo000004.ts
* hls: MIME type: unknown, expected size: unknown
* hls: downloading from file:///Users/thierry/tmp/foo000005.ts
* hls: MIME type: unknown, expected size: unknown
* hls: HLS playlist completed

===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ....... 1080 (0x0438)  |  Services: ............. 11  |
|  Bytes: ....................... 436,207,188  |  PID's: Total: ........ 138  |
|  TS packets: .................... 2,320,251  |         Clear: ........ 109  |
|     With invalid sync: .................. 0  |         Scrambled: ..... 29  |
|     With transport error: ............... 4  |         With PCR's: .... 10  |
|     Suspect and ignored: ................ 1  |         Unreferenced: ... 1  |
|-----------------------------------------------------------------------------|
... remaining output skipped ...
~~~

The "unknown" messages you see in verbose mode are due to the lack of MIME headers (because there is no HTTP server to send them). But the transport stream is there as you can see from the output.

Setting some standard HTTP server to serve the files locally is better because you test the "real" behaviour, with HTTP responses and headers. But the `file:` URL works.

On Windows, WinInet is used instead of libcurl (macOS and Linux). The handling of the `file:` URL is Windowsish (ie. strange). If you use the full `file:` URL (the one which works with Edge for instance), then WinInet reports an error and `tsp` immediately terminates:

~~~
PS C:\> tsp -v -I hls file:///C:/Users/thierry/Documents/tmp/foo.m3u -P analyze -O drop
* Error: hls: Web error: error opening URL (The filename, directory name, or volume label syntax is incorrect.)
PS C:\>
~~~

However, when you use two slashes instead of the expected three after `file:`, it works:
~~~
PS C:\> tsp -v -I hls file://C:/Users/thierry/Documents/tmp/foo.m3u -P analyze -O drop
* Error: hls: Web error: error getting HTTP response headers (The supplied handle is the wrong type for the requested operation)
* hls: downloaded foo.m3u, media playlist, static, 6 segments, 10 seconds/segment
* Error: hls: Web error: error getting HTTP response headers (The supplied handle is the wrong type for the requested operation)
* hls: downloading from file://C:/Users/thierry/Documents/tmp/foo000000.ts
* hls: MIME type: unknown, expected size: unknown
* hls: unknown initial input bitrate
* Error: hls: Web error: error getting HTTP response headers (The supplied handle is the wrong type for the requested operation)
[... 2 lines cut ...]
* Error: hls: Web error: error getting HTTP response headers (The supplied handle is the wrong type for the requested operation)
* hls: downloading from file://C:/Users/thierry/Documents/tmp/foo000002.ts
* hls: MIME type: unknown, expected size: unknown
* Error: hls: Web error: error getting HTTP response headers (The supplied handle is the wrong type for the requested operation)
* hls: downloading from file://C:/Users/thierry/Documents/tmp/foo000003.ts
* hls: MIME type: unknown, expected size: unknown
* Error: hls: Web error: error getting HTTP response headers (The supplied handle is the wrong type for the requested operation)
* hls: downloading from file://C:/Users/thierry/Documents/tmp/foo000004.ts
* hls: MIME type: unknown, expected size: unknown
* Error: hls: Web error: error getting HTTP response headers (The supplied handle is the wrong type for the requested operation)
* hls: downloading from file://C:/Users/thierry/Documents/tmp/foo000005.ts
* hls: MIME type: unknown, expected size: unknown

===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
[... cut ...] ...... 11  |
|  Bytes: ....................... 436,207,188  |  PID's: Total: ........ 138  |
|  TS packets: .................... 2,320,251  |         Clear: ........ 109  |
|     With invalid sync: .................. 0  |         Scrambled: ..... 29  |
|     With transport error: ............... 4  |         With PCR's: .... 10  |
|     Suspect and ignored: ................ 1  |         Unreferenced: ... 1  |
|-----------------------------------------------------------------------------|
...
~~~

lars18th, 2019-12-18:
> Seriously, did you try it?

In fact, only in Windows with syntax `tsp -I hls file:///C:/...` so for this reason I opened the issue.

In any case, if it works I suggest a _simple_ enhancement: support it with an agnostic mode. The syntax can be `tsp -I hls filename.ts`. This it's quite simple to implement: if the URL doesn't starts with `http://` (or similar syntax with `protocol://`) then add `file:///` if the system is Unix or `file://` if it's Windows. That's will work in all systems. And you can document it and overcome the error warning messages.

What you think?

lelegard, 2019-12-19:
>  This it's quite simple to implement: if the URL doesn't starts with http:// (or similar syntax with protocol://) then add file:///

Done

lars18th, 2019-12-19:
> > This it's quite simple to implement: if the URL doesn't starts with http:// (or similar syntax with protocol://) then add file:///
> 
> Done

As usual you're the best!
I hope you want to update the documentation too.

Regards.

lelegard, 2019-12-19:
Doc updated (as it would have been without flattery anyway).

closed 2019-12-20 by lars18th