tsduck/tsduck #451: TS Buffer plugin: New plugin suggestion

samuraibrian, 2019-11-07:
And yet another request! Not sure how many folks would find this useful (but might be simple to implement...). It's pretty much for live streams only - a tsp plugin that delays it's input by a configurable time delay. I.e. a 'timeshift' function.

BR, Jon

lars18th, 2019-11-07:
Hi @samuraibrian ,

Already requested #346.

samuraibrian, 2019-11-07:
Kind of @lars18th ! However, I'm not wanting to modify (add stuffing to) the source. Merely delay it. See #434 for a use.

lars18th, 2019-11-07:
> Kind of @lars18th ! However, I'm not wanting to modify (add stuffing to) the source. Merely delay it. See #434 for a use.

I agree with you. However, if you can _delay_ it, then you can change it in the future. Me too interested in such `buffer` plugin.

And just a "simple" suggestion: please, support not only memory buffers, but FIFO files.

Regards.

lelegard, 2019-11-12:
The following feature could be easily implemented:

- Specify a fixed-size buffer either using a number of packets or a duration (which is immediately converted into a number of packets using the initial bitrate of the stream).
- Initially, while the buffer is filling, either output nothing or null packets.
- Once the buffer is full, each input packet is stored in the buffer and replaced by the oldest packet from the buffer.
- At the end of input, the content of the buffer _is lost_.

Would that be ok? Including the loss of the buffer content at end of input?

lars18th, 2019-11-12:
Hi @lelegard ,

> The following feature could be easily implemented:

Great!

> * Specify a fixed-size buffer either using a number of packets or a duration (which is immediately converted into a number of packets using the initial bitrate of the stream).

OK. I recommend to put a _warning_: "Only for CBR streams. With VBR stream the result will be unknown."

> * Initially, while the buffer is filling, either output nothing or null packets.

I suggest to output null packets when filling. That's the only way to guarantee the CBR output.

> * Once the buffer is full, each input packet is stored in the buffer and replaced by the oldest packet from the buffer.

:+1: 

> * At the end of input, the content of the buffer _is lost_.

You mean _at the end of the process_? In this case then OK. When the process is killed the buffer is not flushed but discarded.
Futhermore, I suggest to fill the buffer with nulls if the input is empty/stalled. Why this? Because the input can be from non-constant source (HTTP, tswitch, etc.). Please, catch eventual interruptions in the input.

> Would that be ok? Including the loss of the buffer content at end of input?

In my opinion: YES! :wink:

lelegard, 2019-11-12:
The following is based on the fundamental principle of tsp: processing packets one by one on a stream flowing through tsp, without adding new packets.

> You mean at the end of the process? In this case then OK. When the process is killed the buffer is not flushed but discarded.

I mean also on a regular end of input. After the last input packet, you still have N packets in the time-shift buffer. Then, what? What do you do with them? You cannot insert new packets in the stream, you can only replace existing packets. And after the end of input, there are no more packet to replace, so your time-shift buffer, containing the last N packets of the input stream, is lost. To cope with this, the user can specify the global option `tsp --add-stop-stuffing N`. But this cannot be automated from the time-shift plugin.

> Futhermore, I suggest to fill the buffer with nulls if the input is empty/stalled.

Again, tsp only replaces packets. A plugin is invoked each time there is a packet to process. When the input is stalled, no packet comes in and no plugin is invoked. So, when the input is stalled, the output is stalled as well as soon as the global buffer is empty (the internal tsp buffer, where all plugins process packets, not the time-shift buffer which is hidden in one plugin).

lars18th, 2019-11-12:
Hi @lelegard ,

> The following is based on the fundamental principle of tsp: processing packets one by one on a stream flowing through tsp, without adding new packets.

OK. OK. I agree. This new `buffer` plugin will _only_ delay packets replacing every new packet with the most ancient packet. OK. It's simple and effective. :+1: 

However, regarding the second point (filling the buffer with nulls when stalled input), and respecting the KISS model, perhaps it's preferable to create (or modify some current input) to garantee the streaming. And for sure only for CBR streams. You agree with this idea?

lelegard, 2019-11-12:
> perhaps it's preferable to create (or modify some current input) to garantee the streaming. And for sure only for CBR streams. You agree with this idea?

Redesigning the tsp input system to guarantee a CBR input is a distinct topic which is more difficult and not planned.

Additionally, a CBR stream and a CBR input are different topics. You are clearly focused on real-time production where both concepts are identical. But for tsp, the stream bitrate and the real time are distinct. This is why the plugin `regulate` was developed to synchronize the two. This is also why the tsp documentation insists on the difference between rendering speed and transmission speed.

To summarize, the time-shift plugin can be implemented but tsp will remain a stream processor, not a real-time generator. If the input stalls, the output stalls.

lars18th, 2019-11-12:
Hi @lelegard ,

I agree with the sentence that the guarantee of a CBR _input_ it's a different topic.

I hope we can discuss about this when this simple buffer plugin is done. But speaking of the need for a CBR input, please think only on the behaviour of the `buffer` plugin in this scenerario: Using a tuner, you unse the `zap` plugin to get a service. Then with `buffer` you add a delay. However, if the input fails (for a second, or more... because a glitch in the anntena or a problem with the wire), then the output from the `buffer` has double interference. So for this reason a request for an input with a _robust_ streaming.

samuraibrian, 2019-11-12:
+1 to the proposal, @lelegard 

BR, Jon

lelegard, 2019-11-14:
The plugin `timeshift` has been added.
Please test it.

samuraibrian, 2019-11-14:
Soon as I can - my dvb stick ("August") has stopped working after a system update :-(
Thanks for the rapid release!

samuraibrian, 2019-11-14:
From a quick test - delaying a "-P time <start> <stop>" recording of a 24Mbps DVB-T off-air mux by 160,000 packets (~10 secs.), and inspecting the TDT against the timed recording parameters - it worked very well. Thanks @lelegard 

Jon

closed 2019-12-02 by lelegard