tsduck/tsduck #452: tsduck DVBSky s960

lucdig, 2019-11-08:
Hi, I use tsduck for testing DVBSky s960 device, and it works perfectly. Now, I want to go to production, anyway, the tool (dvblast or tvheadend) that I use works with DVBSky s960 only when the device is switched on for the first time. The second. third,... time that I use the tool with the device, I can't get the dvb-s stream. I have to unplug the power supply of the device and plug it again.
With tsduck, I can use the device not only when it's switched on for the first time. Every time that I use a tsp command, the device works, without unplug/plug.
I wonder if you could tell me what tsduck does with the device. Maybe, I could run a tsduck command line command to "reset" the device, achieving that the device works the second, third,.. time with dvblast or tvheadend.

Thanks, regards

lelegard, 2019-11-08:
Hi,

I have an S390 too and it works fine as well. TSDuck does not do anything special, it just configures a tuner, whatever it is. Specifically, it does not do anything for this model of tuner.

You did not mention if you were on Linux or Windows. The two OS have completely different infrastructures for tuner reception. This has been discussed several times here, especially for Windows which uses an awfully complex framework.

In practice, it running `tsp` once unblocks your other applications, the "reset" command could be something like `tsp -I dvb ... -P until --packet 1 -O drop`

lucdig, 2019-11-08:
It's a Linux: uname -a:
`Linux raspberrypi 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux`
I tried with the `tsp -I dvb ... -P until --packet 1 -O drop` command, and then with a `dvblast`, anyway, the outcome is the same. dvblast, after a while, shows:
`warning: no DVR output, resetting
reset cause: dvr`

lucdig, 2019-11-08:
I found this related to Raspberry and DVBSky s960 [https://forum.libreelec.tv/thread/12452-problems-with-dvbsky-technotrend-usb-dvb-s2-tuners-since-le8/?pageNo=2](url), but I can't find `mutex.c` in my system...
Anybody could help me on recompiling this? I'm not an expert in recompiling...
Regards

lelegard, 2019-11-08:
The thread is not clear. The file `mutex.c` is part of the kernel. So, this means patching and rebuilding a new kernel for Raspberry Pi. This is much more than recompiling one file. Rebuilding a native kernel on a Raspberry Pi probably takes ages. Rebuilding a cross-compiled kernel on a server requires a specific setup.

Moreover, the thread mentions that the kernel patch may not be appropriate and a patch in the driver is suggested. So, it is unclear what is the right way.

If there is a problem in the driver, if it works with TSDuck, this is probably a matter of luck only. `tsp` and `dvblast` are probably both right. The problem is elsewhere. Could you check on a PC with a 5.x kernel? Latest versions of Ubuntu and Fedora use 5.x, so you may choose your favorite. If it works correctly on 5.x, then you need to get an updated kernel for Raspberry. TSDuck cannot help you.

lucdig, 2019-11-08:
Hi, thanks for your replies. I haven't the chance to test it with a 5.x kernel. Anyway, I agree with you that the matter is elsewhere. I don't think that it's only luck, because tsp always works, and dvblast works the first time.

I tried in the past dvblast and tvheadend with the PCTV 461e and I had no problems at all, only DVBSky s960 gives me troubles.

Thanks again, regards

lucdig, 2019-11-08:
closed the issue because it is not a tsduck issue

closed 2019-11-08 by lucdig