tsduck/tsduck #453: Switch power mode of Dektec DTU-315 device

mdegura, 2019-11-08:
Hi,

I'm using a Dektec DTU-315 device with tsduck and it works just fine:

> $ tsdektec -av
> 
> DTAPI: 5.35.0.124, Dtu: 4.13.5.82
> 
> * Device 0: DTU-315 (Universal modulator)
>   Physical ports: 1
>   Channels: input: 0, output: 1
>   Output 0: DTU-315 port 1, Modulator (L-band 950-2150MHz, VHF-band 47-470MHz,
>       UHF-band 400-862MHz, Uni-directional output, MPEG-2 transport stream,
>       Modulator output, High-quality modulation, Low-power mode, ATSC 8-VSB
>       modulation, DVB-S modulation, DVB-S2 modulation, DVB-T modulation, QAM-A
>       modulation, QAM-B modulation, QAM-C modulation, DVB-T2 single PLP
>       modulation, Adjustable output level, Continuous wave, DVB carrier ID ,
>       Mute RF output signal, Adjustable roll-off factor, DVB-S2 16-APSK/32-APSK,
>       ATSC 8-VSB reception, Lock TS rate to input port)
>   Subsystem id: 0x0000 (DTU-315)
>   Subsystem vendor id: 0x0000
>   Device id: 0x030F
>   Vendor id: 0x1297
>   Serial number: 0000000012C6897B
>   Firmware version: 1 (0x00000001)
>   **Firmware variant: 0 (0x00000000)**
>   USB address: 0
>   Customer id: 301819
>   Engineering change level: Rev 4
>   Manufacture id: 03
>   Production date: 2018.12
>   Part number: DTU-315
>   Serial number: 315001211

However, whenever I try to use a frequency over 2GHz or a symbol rate over 43.75MBd, I have driver errors:

> $ tsp --verbose --input file --infinite ~/streams/20190604_HOR4_15mn.ts --output dektec --modulation DVB-S --satellite-frequency 12610000000 --lnb 9750,10600,11700 --convolutional-rate 5/6 --symbol-rate 30000000
> * file: initial input bitrate is 46,077,826 b/s
> * dektec: using Dektec device 0, output channel 0 (DTU-315 port 1)
> * dektec: using DVB-S FEC 5/6
> * dektec: setting output TS bitrate to 46,078,431 b/s
> * dektec: setting output carrier frequency to 2,010,000,000 Hz
> * Error: dektec: set modulator frequency error: **DTAPI_E_INVALID_RATE** (DTAPI status 4109)

> $ tsp --verbose --input file --infinite ~/streams/20190604_HOR9_2mn.ts --output dektec --modulation DVB-S2 --satellite-frequency 11171000000 --lnb 9750,10600,11700 --convolutional-rate 3/4 --symbol-rate 45001000
> * file: initial input bitrate is 67,000,056 b/s
> * dektec: using Dektec device 0, output channel 0 (DTU-315 port 1)
> * dektec: setting output TS bitrate to 66,937,776 b/s
> * dektec: setting output carrier frequency to 1,421,000,000 Hz
> * dektec: output fifo size: 16,777,216 bytes, max: 16,777,216 bytes, typical: 16,777,216 bytes
> * Error: dektec: output device start send error: **DTAPI_E_INVALID_SYMRATE** (DTAPI status 4213)

I found out that it is because the device can operate on 2 different modes according to [the device datasheet](https://www.dektec.com/products/USB/DTU-315/downloads/DTU-315%20Datasheet.pdf) (see 2.2 Low-Power versus High-Quality Mode).
And since I could not find a way to switch this power mode, I made a quick and dirty hack in  `tsdektec`: it just switches the power mode whenever I execute it (see attached patch [power_mode.patch.txt](https://github.com/tsduck/tsduck/files/3823644/power_mode.patch.txt)).
I can check that the change was taken into account with the `tsdektec -av` command, which displays `Firmware variant: 0 (0x00000000)` when in low power mode and `Firmware variant: 2 (0x00000002)` when in high quality mode.

Is there already a way I am not aware of to switch power mode?
If not, is it possible to make it available?

Thanks!

lelegard, 2019-11-08:
This feature was not addressed so far. It will be added as an option to `tsdektec` and to plugin `dektec` so that it could be set just before starting the modulation.

I can usually have access to a DTU-315 but some of my colleagues brought it away for a demo and I do not know when I can use it again. So, you will have to test it yourself when implemented.

lelegard, 2019-11-08:
The following option has been added to `tsdektec` and to output plugin `dektec`:
~~~
--power-mode value
    On DTU-315 USB modulators, set the power mode to the specified value.
    Must be one of "high-quality", "low-power".
~~~
As I mentioned, I do not have access to a DTU-315 right now. Could you please test it ?

mdegura, 2019-11-12:
Thanks. I ran some tests with your latest commits with the DTU-315 device.

**Setting power mode on `tsdektec`**

Plug DTU-315 device in:
> $ tsdektec -av
> 
> DTAPI: 5.35.0.124, Dtu: 4.13.5.82
> 
> * Device 0: DTU-315 (Universal modulator)
>   Physical ports: 1
>   Channels: input: 0, output: 1
>   Output 0: DTU-315 port 1, Modulator (L-band 950-2150MHz, VHF-band 47-470MHz,
>       UHF-band 400-862MHz, Uni-directional output, MPEG-2 transport stream,
>       Modulator output, High-quality modulation, Low-power mode, ATSC 8-VSB
>       modulation, DVB-S modulation, DVB-S2 modulation, DVB-T modulation, QAM-A
>       modulation, QAM-B modulation, QAM-C modulation, DVB-T2 single PLP
>       modulation, Adjustable output level, Continuous wave, DVB carrier ID ,
>       Mute RF output signal, Adjustable roll-off factor, DVB-S2 16-APSK/32-APSK,
>       Internal RF clock reference, Internal TS rate clock reference)
>   Subsystem id: 0x0000 (DTU-315)
>   Subsystem vendor id: 0x0000
>   Device id: 0x030F
>   Vendor id: 0x1297
>   Serial number: 0000000012C6897B
>   Firmware version: 1 (0x00000001)
>   **Firmware variant: 0 (0x00000000)**
>   USB address: 0
>   Customer id: 301819
>   Engineering change level: Rev 4
>   Manufacture id: 03
>   Production date: 2018.12
>   Part number: DTU-315
>   Serial number: 315001211

--> power mode is _low-power_ (default).

Try to use frequency over 2GHz in _low-power_ power mode:
> tsp --verbose --input file --infinite ~/streams/20190604_HOR4_15mn.ts --output dektec --modulation DVB-S --satellite-frequency 12610000000 --lnb 9750,10600,11700 --convolutional-rate 5/6 [...]

--> error (as expected).

Set _high-quality_ power mode:
> $ tsdektec --power-mode high-quality
> $ tsdektec -av
> 
> DTAPI: 5.35.0.124, Dtu: 4.13.5.82
> 
> * Device 0: DTU-315 (Universal modulator)
>   Physical ports: 1
>   Channels: input: 0, output: 1
>   Output 0: DTU-315 port 1, Modulator (L-band 950-2150MHz, VHF-band 47-470MHz,
>       UHF-band 400-862MHz, Uni-directional output, MPEG-2 transport stream,
>       Modulator output, High-quality modulation, Low-power mode, ATSC 8-VSB
>       modulation, DVB-S modulation, DVB-S2 modulation, DVB-T modulation, QAM-A
>       modulation, QAM-B modulation, QAM-C modulation, DVB-T2 single PLP
>       modulation, Adjustable output level, Continuous wave, DVB carrier ID ,
>       Mute RF output signal, Adjustable roll-off factor, DVB-S2 16-APSK/32-APSK,
>       Internal RF clock reference, Internal TS rate clock reference)
>   Subsystem id: 0x0000 (DTU-315)
>   Subsystem vendor id: 0x0000
>   Device id: 0x030F
>   Vendor id: 0x1297
>   Serial number: 0000000012C6897B
>   Firmware version: 1 (0x00000001)
>   **Firmware variant: 2 (0x00000002)**
>   USB address: 0
>   Customer id: 301819
>   Engineering change level: Rev 4
>   Manufacture id: 03
[... 201 lines cut ...]
>   Manufacture id: 03
>   Production date: 2018.12
>   Part number: DTU-315
>   Serial number: 315001211

--> now the device is actually set in _low-power_ power mode.

Let's try one more time:

> $ tsp --verbose --input file --infinite ~/streams/20190604_HOR4_15mn.ts --output dektec --power-mode low-power --modulation DVB-S --satellite-frequency 12610000000 --lnb 9750,10600,11700 --convolutional-rate 5/6 --symbol-rate 30000000
> * file: initial input bitrate is 46,077,826 b/s
> * dektec: using Dektec device 0, output channel 0 (DTU-315 port 1)
> * dektec: using DVB-S FEC 5/6
> * dektec: setting output TS bitrate to 46,078,431 b/s
> * dektec: setting output carrier frequency to 2,010,000,000 Hz
> * Error: dektec: set modulator frequency error: DTAPI_E_INVALID_RATE (DTAPI status 4109)

--> error (as expected)

**Conclusions**
- `tsdektec` seems to work fine
- when device is in _low-power_ power mode, `dektec` ouput plugin must be run twice to actually enable _high-quality_ power mode
- when device is in _high-quality_ power mode, `dektec` ouput plugin must be run three times to actually enable _low-power_ power mode

Hope this helps :)

lelegard, 2019-11-12:
Hi @mdegura,

Thank you for this very detailed report. It is appreciated.

So, `tsdektec` is fine. In the plugin, I assume that there is an ordering issue between the power mode setting and the initialization of the channel. It works when the power setting is done in `tsdektec` and then used in the plugin where the channel is opened and reset.

I have made a modification in the plugin. The order of operation is now:
- Open the device
- Set the power mode on the port
- Attach the channel on the port
- Reset the port

Could you please test this update?

mdegura, 2019-11-12:
Hi,

Thanks for the quick update. The output plugin seems to work fine now.

Initial state is _low-power_ power mode:
> $ tsdektec -av
>
> DTAPI: 5.35.0.124, Dtu: 4.13.5.82
>
> * Device 0: DTU-315 (Universal modulator)
>   Physical ports: 1
>   Channels: input: 0, output: 1
>   Output 0: DTU-315 port 1, Modulator (L-band [...]

Trying to use a frequency over 2GHz in _high-quality_ power mode works:
> $ tsp --verbose --input file --infinite ~/streams/20190604_HOR4_15mn.ts --output dektec --power-mode high-quality --modulation DVB-S --satellite-frequency 12610000000 --lnb 9750,10600,11700 [...]

The device is now in _high-quality_ power mode:
> $ tsdektec -av
>
> DTAPI: 5.35.0.124, Dtu: 4.13.5.82
>
> * Device 0: DTU-315 (Universal modulator)
>   Physical ports: 1
>   Channels: input: 0, output: 1
>   Output 0: DTU-315 port 1, Modulator (L-band [...]

Now forcing _low-power_ power mode with a frequency over 2GHz gives the expected error right away.
> $ tsp --verbose --input file --infinite ~/streams/20190604_HOR4_15mn.ts --output dektec --power-mode low-power --modulation DVB-S --satellite-frequency 12610000000 --lnb 9750,10600,11700 [...]

Thank you very much for your work!

lelegard, 2019-11-12:
Cool, that's how cooperation and joint development / testing should always work !
Closing this issue...

closed 2019-11-12 by lelegard