tsduck/tsduck #454: zap plugin discards global pids

lucdig, 2019-11-09:
Hi, I need to do some tests with a service and all the global pids.

I see that the zap plugin doesn't keep all the global pids (see below). How can I achieve to zap one channel but **keeping** all global pids?

`===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 1,171, PID's: 3 (clear: 3, scrambled: 0)                       |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C      376,988 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C      376,988 b/s  |
|  0x0000  PAT .......................................... C       96,259 b/s  |
|  0x0011  SDT/BAT ...................................... C      261,413 b/s  |
|  0x0014  TDT/TOT ...................................... C       19,316 b/s  |`

If I don't zap to one service, the list of global pids is:

`===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 34,037, PID's: 11 (clear: 11, scrambled: 0)                    |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C    2,143,983 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C      178,702 b/s  |
|  0x0000  PAT .......................................... C       15,054 b/s  |
|  0x0001  CAT .......................................... C       14,991 b/s  |
|  0x0010  DVB-NIT ...................................... C        5,795 b/s  |
|  0x0011  SDT/BAT ...................................... C       41,384 b/s  |
|  0x0012  EIT .......................................... C       98,453 b/s  |
|  0x0014  TDT/TOT ...................................... C        3,023 b/s  |
|  0x00B8  Nagravision EMM .............................. C        2,582 b/s  |
|  0x00BD  MediaGuard Group EMM for OPI 0 (0x0000) ...... C            0 b/s  |
|  0x00BE  Nagravision EMM .............................. C        6,487 b/s  |
|  0x00C2  Nagravision EMM .............................. C       15,117 b/s  |
|  0x0137  Nagravision EMM .............................. C      206,795 b/s  |
|  0x1FFF  Stuffing ..................................... C    1,734,298 b/s  |`

lelegard, 2019-11-09:
Did you try options `--cas` and `--eit`?

lucdig, 2019-11-09:
They are parameters of sifilter plugin, I try to added -P sifilter --cas and --eit in the zap command, but I can't see the global pids yet.

lelegard, 2019-11-09:
> They are parameters of sifilter plugin, 

They are also options of the zap plugin. Many plugins share consistent options.

Please remember that RTFM has always been the most important rule in software engineering...

lucdig, 2019-11-10:
Hi @lelegard , thanks, the 2 parameters work with zap, anyway, I still have some problems.

When I have an encrypted channel and try to decrypt with tsdecrypt (using a regular nagra card, no hacking at all !), I see this error:

`!!! Wrong PMT table CRC! It should be 0 but it is 0x6df9a660 (CRC in data is 0x1da917d2)`.

If I use `dvblast` instead of `tsp`, tsdecrypt doesn't show this error. In fact, with dvblast I use the option

`-z --any-type: pass through all ESs from the PMT, of any type`

and I can't find any similar parameter for the `zap` plugin.

I would like to do complete tests with tsduck because I think that it is a fantastic tool, and no other tool gives me the chance of debugging and troubleshooting.

Thanks again, regards

lelegard, 2019-11-10:
> When I have an encrypted channel and try to decrypt with tsdecrypt (using a regular nagra card, no hacking at all !),

Let's be very clear here and let's stop playing with words. A conditional access system is designed to reserved the usage of protected content to specific operator-approved devices, typically the association of a smart-card and an STB or CAM. A CAS smart-card is not made for free-to-use general purpose. Nowadays, most CAS, including Nagra, encrypt the CW between the smart-card and the hardware key ladder of the chipset to make sure that the CW is not used by some third-party software. The fact that some legacy systems or operators still use clear CW out of the smart-card is an artefact of history. Using a valid smart-card and a third-party software to extract the CW is the basis of "card sharing" or "control word sharing" and _this is hacking by definition_.

As I mentioned several times here, this forum will never be used to discuss hacking and offenders will be banned. TSDuck is a toolbox for digital TV engineers or enthusiasts, not a hacking tool.

> I see this error:
> !!! Wrong PMT table CRC! It should be 0 but it is 0x6df9a660 (CRC in data is 0x1da917d2).

This message is insane. A PMT CRC shall be computed according to ISO 13818-1, this is the rule of the standard. Expecting the CRC of a PMT to be zero is insane, so is the corresponding software.

> If I use dvblast instead of tsp, tsdecrypt doesn't show this error. In fact, with dvblast I use the option
> -z --any-type: pass through all ESs from the PMT, of any type
> and I can't find any similar parameter for the zap plugin.

This is the default behaviour of the `zap` plugin. All components of the selected service are passed  by default and the PMT is left unmodified. Specific options such as `--audio`, `--subtitles`, `--no-subtitles`, `--pes-only`, `--no-ecm` are used to remove selected components.

lucdig, 2019-11-10:
Ok, thanks @lelegard for explaining the behavior about computing the CRC

closed 2019-11-10 by lucdig