tsduck/tsduck #455: Support for nVidia AI Jetson platforms (Ubuntu 18.04 aarch64)

rodolfo-vargas, 2019-11-11:
Hi there, I'm trying to compile tsDuck on the Jetson Nano. The new AI platform from nVidia.  unfortunately it looks like this platform is not supported.  I wish I can help, but my programming skills are not good.

cc1plus: error: unknown value 'aarch64' for -march
../../Makefile.common:903: recipe for target 'release-aarch64/tsAACDescriptor.o' failed

lelegard, 2019-11-12:
Hi @rodolfo-vargas 

The makefile of TSDuck tried to be clever about the target architecture. That was fine with Intel platforms but more and more different platforms make it more difficult. I will simplify this.

In the meantime, you may try:
~~~
make TARGET_ARCH="-march=native"
~~~
If you have enough CPU and memory, I suggest to use parallel compilation. Otherwise, building TSDuck takes ages. I routinely use the make option `-j10` .

lars18th, 2019-11-12:
> If you have enough CPU and memory, I suggest to use parallel compilation. Otherwise, building TSDuck takes ages. I routinely use the make option `-j10` .

Hi @lelegard ,

I see this for a long time! :wink:

lelegard, 2019-11-12:
I made the modification in the makefile. You can now try a regular `make -j10`. Please let us know if this works. It is always interesting to get feedback about new platforms.

For more tests, use `make test`. It runs the unitary test suites.

For complete non-regression tests, first clone the repository tsduck-test side by side with the repository tsduck (same URL, just use tsduck-test instead of tsduck). Then, from the tsduck directory, run `make test-suite`.

rodolfo-vargas, 2019-11-15:
Downloaded the latest master, and compiles now!

Enabling jetjson_clocks (A utility that powers on the 4 ARM Cpus at full speed) the compilation of :

$ make NOPCSC=1 NOCURL=1 NODTAPI=1 -j10
toke 15min to compile.

After this I decided to do a make distclean and compile:
$ make -j10 
and The time to compile was 15:41 min

The result for $make test is as follow:
$ make test
source release-aarch64/setenv.sh && release-aarch64/utest 
OK (399 tests, 9559 assertions)

source release-aarch64/setenv.sh && release-aarch64/utest_static 
OK (396 tests, 9547 assertions)

rodolfo-vargas, 2019-11-15:
Success! Ran the $ make test-suite -j10, (17min)

----  ALL 846 tests passed

rodolfo-vargas, 2019-11-15:
Last question, (I know I'm pushing my luck too far...) How can I build a static version?

lelegard, 2019-11-15:
> Last question, (I know I'm pushing my luck too far...) How can I build a static version?

Just add `STATIC=1`:
~~~
make STATIC=1 NOPCSC=1 NOCURL=1 NODTAPI=1 NOTEST=1 -j10
~~~

However, there are several limitations:
- No static library for curl and pcsc on most distros, use `NOPCSC=1 NOCURL=1`
- Must install additional static library packages on some distros.
- The unitary tests do no build, use `NOTEST=1`.
- All plugins are statically linked in `tsp`, making it quite big (11 MB on Intel) and without possibility to load additional plugins.
- The static glibc disables some features. Last time I looked into it, hostname resolution (`gethostbyname`) used dynamically loaded libraries, using a plugin-scheme for host name resolution. So, AFAIK, static binaries on Linux cannot use host names, you must use plain IP addresses.

Note that `NODTAPI=1` is always implicit on ARM systems because Dektec provides DTAPI in binary form only and on Intel CPU's only.

> toke 15min to compile.

I seems quite a lot. On an old 2-core i7 laptop using Ubuntu, `make -j10` takes 6 minutes. It is even much faster on my 4-core i7 iMac, even though clang is notoriously slower to compile than gcc.

I do not know the nVidia AI Jetson. From the Web, it says "supercomputer-in-a-module" and the CPU is a "6-core Carmel Arm 64-bit CPU, 6MB L2 + 4MB L3". A 6-core Arm 64-bit being 2.5 times slower than a 2-core Intel 64-bit is strange. I know that CPU's do not compare equally, cache size, system bus, clock speed, all of these system characteristics matter. And I assume that "supercomputer-in-a-module" is just the usual marketing bullshit and mostly refer to the AI capabilities (although I see no reference to an NPU) rather than to the ARM CPU.

rodolfo-vargas, 2019-11-15:
Thanks a lot!

About the Jetson Nano: It features 4 ARMv8 Processors (fp asimd evtstrm aes pmull sha1 sha2 crc32)

However, it has 256 core GPUs, capable to process 472 GFLOPS using around 10W power power!!, It can decode eight 1080i/p HD channels in H262/4/5 simultaneously. It allows you to run tensaflow, OpenCV and other great stuff... 

The multimedia API capabilities are great, however it is laking stable support for professional broadcasting, And lacks the support of FFMPEG and a stable GSTREAMER (which has some issues on MPEGTS). TSDuck is an amazing thing to have, as I believe it will open the opportunities to enable new AI broadcasting applications. such as QOS, Video Fidelity and quality measurements based on the HVS, etc...

So thank you guys!

closed 2019-12-02 by lelegard