tsduck/tsduck #456: Web Radio

fabsconnect, 2019-11-12:
Hi all,

is it possible to receive the Webradio and then play it on a multicast?
I triend in Windows the following commands without success:

PS C:\Users\administrator> tsp --debug=5 -O ip --local-address 169.254.212.65 239.140.140.140:1234 -I hls http://mp3.radio.ch/radiozuerisee128k
tsp: Debug[5]: debug level set to 5
* tsp options:
  --add-input-stuffing: 0/0
  --bitrate: 0 b/s
  --bitrate-adjust-interval: 5,000 milliseconds
  --buffer-size-mb: 16,777,216 bytes
  --debug: 5
  --list-processors: 0
  --max-flushed-packets: 0
  --max-input-packets: 0
  --realtime: unknown
  --monitor: 0
  --verbose: 1
  Number of packet processors: 0
  Input plugin 1:
    Name: hls
    Type: input
    Arg[0]: "http://mp3.radio.ch/radiozuerisee128k"
  Output plugin 1:
    Name: ip
    Type: output
    Arg[0]: "--local-address"
    Arg[1]: "169.254.212.65"
    Arg[2]: "239.140.140.140:1234"
* Debug[5]: debug level set to 5
tsp: Debug: trying to load "C:\test\TSDuck\bin\tsplugin_hls.dll"
tsp: Debug: symbol tspNewProcessor not found in C:\test\TSDuck\bin\tsplugin_hls.dll
tsp: Debug: trying to load "C:\test\TSDuck\bin\tsplugin_ip.dll"
tsp: Debug: symbol tspNewProcessor not found in C:\test\TSDuck\bin\tsplugin_ip.dll
tsp: Debug: realtime: yes, using --max-input-packets 1,000 --max-flushed-packets 1,000
* Debug[5]: hls: debug level set to 5
* Debug[5]: ip: debug level set to 5
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: hls: downloading http://mp3.radio.ch/radiozuerisee128k
* Debug: hls: HTTP header: HTTP/1.0 200 OK
* Debug: hls: HTTP header: X_RadioName: Radio Netstream
* Debug: hls: HTTP header: icy-br: 128
* Debug: hls: HTTP header: icy-pub: 1
* Debug: hls: HTTP header: icy-description: Radio Z├╝risee
* Debug: hls: HTTP header: icy-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
* Debug: hls: HTTP header: icy-url: http://www.radio.ch
* Debug: hls: HTTP header: Instance-id: 0440d0eceff8aaed497863d31236eaf9
* Debug: hls: HTTP header: Cache-Control: no-cache
* Debug: hls: HTTP header: Server: Radio Netstream Streaming Server 8.0.16
* Debug: hls: HTTP header: icy-genre: Rock
* Debug: hls: HTTP header: Expires: Mon, 26 Jul 1997 05:00:00 GMT
* Debug: hls: HTTP header: icy-metaint: 0
* Debug: hls: HTTP header: Pragma: no-cache
* Debug: hls: HTTP header: icy-name: Radio Z├╝risee
* Debug: hls: HTTP header: Connection: close
* Debug: hls: HTTP header: Content-Type: audio/mpeg

Thank in advance for you help.

Regards

lelegard, 2019-11-12:
Hi @fabsconnect,

TSDuck processes MPEG-2 transport streams. In your case, the MIME content type audio/mpeg, not transport stream. So, TSDuck cannot work on it.

TSDuck does not convert or process audio and video. It does not convert between container formats either. I think that the tool you need is ffmpeg, not TSDuck. 

Note: Additionally, you used the `hls` input plugin. HLS is a protocol which points to a play list (a text file) which in turns points to transport stream segments. So, even if your radio was sent as transport stream, it would not have worked. The input plugin which directly reads a transport stream over a web connection is named `http`.

fabsconnect, 2019-11-12:
Hi @lelegard,

many thanks for your answer. Ok, I had the doubt, I also tried without success.

Have a nice day.

Regards

lars18th, 2019-11-12:
Hi @fabsconnect ,

When using not MPEG-2 TS streams you need to use other tools. For exemple with the good `ffmpeg` you can do the same: read the stream (in this case an elementary mp3 stream) for one HLS source and restream it to a multicast address. You can do it using the same elementary mp3 stream container, or you can encapsulate it inside one MPEG-2 TS stream. So, after that you can use TSDuck to process it (however, I feel you don't want to do it).

I hope it helps.

closed 2019-12-02 by lelegard