tsduck/tsduck #460: BUG? NIT after ZAP doesn't work

lars18th, 2019-11-19:
Hi @lelegard ,

I'm trying to craete a conformant DVB stream from ffmpeg using TSDuck. However, the idea is to get _any_ TS and create/modify it to be conformant.

Look at this:
```
tsp --debug -a 1/10 -I ip 239.0.0.1:1234 -P zap 1 -P nit --create-after 1 -v 1 -O ip 239.0.0.2:1234
```

The input stream is from ffmpeg. And the problem here is that: the NIT table is **NEVER** created. Futhermore, if I change the `-P nit` to `-P bat` the BAT table is created. And if I remove the `-P zap 1` part then the NIT table is created.

Why this problem?

Please, note that my objective is to "capture any (or the first) service in the TS input stream, then remove any other data, and create all tables from scratch".
Any idea?

Regards.

lelegard, 2019-11-19:
As usual, this is probably a problem of absence of null packets for table insertion.

I assume that there is no input NIT PID. Since the `zap` plugin removes everything but the selected service, all null packets are removed. The next plugin has no null packet to replace and cannot insert anything.

I also assume that there is an input SDT. Since the BAT is on the same PID as the SDT, then inserting a new BAT works because the SDT/BAT PID is replaced with the content of the existing SDT and the created BAT.

If there is more than one input service, I suggest to use `-P zap --stuffing`. The removed services are replaced by stuffing, creating plenty of room for data insertion. If there is only one input service and no input null packets, then use `tsp --add-input-stuffing`. In all cases, use `-P filter -n -p 0x1FFF` as last filter to remove remaining null packets.

lars18th, 2019-11-19:
Hi @lelegard ,

> As usual, this is probably a problem of absence of null packets for table insertion.

Sorry, I feel no in this case. I check it before writing the message. See the first part of the command: `tsp --debug -a 1/10 ...`. Futhermore, the original TS doesn't have any NULL stuff, but the output has it. So, the NULL stuff is created and usable for any transformation.

Please, can you do some check, please?

lelegard, 2019-11-19:
> See the first part of the command: tsp --debug -a 1/10 ...

Yes, but this is immediately followed by `-P zap 1` which removes all null packets, including those which were inserted at input.

> but the output has it. 

I do not believe so if the command is exactly the one that you copied in the initial post.

lars18th, 2019-11-19:
> > See the first part of the command: tsp --debug -a 1/10 ...
> 
> Yes, but this is immediately followed by `-P zap 1` which removes all null packets, including those which were inserted at input.

Ah! So, the problem is this... "`-P zap` removes ALL null packets". So, this is the cause.
So, it's possible to add some option to pass _original_ null packets and remove only the _deleted_ packets?

lelegard, 2019-11-19:
A null packet is a null packet.

> So, it's possible to add some option to pass original null packets and remove only the deleted packets?

No need to. Just do what I proposed in my first answer:
~~~
-P zap ... --stuffing -P plugin-which-inserts-something ... -P filter -n -p 0x1FFFF -O ...
~~~

lars18th, 2019-11-19:
Great!

If I insert the `--stuffing` option all works. Thank you!
```
tsp --debug -a 1/10 -I ip 239.0.0.1:1234 -P zap 1 -s -P nit --create-after 1 -v 1 -O ip 239.0.0.2:1234
```

Regards.

closed 2019-11-19 by lars18th