tsduck/tsduck #461: Suggestion: Add all services from SDT to NIT

lars18th, 2019-11-19:
Hi @lelegard ,

After the creation of a new empty NIT table, I feel it will be very useful to have one option to "insert all services listed in the SDT inside the NIT table". More or less `-P nit -c --copy-from-sdt`

You think this have sense?
Regards.

lelegard, 2019-11-19:
>  "insert all services listed in the SDT inside the NIT table".

What do you mean by that? How would you specify it precisely?

Is this only the insertion of a service_list_descriptor with service ids and types ?

What do we do if some services of the actual TS are missing in the SDT? Or if there is no service_descriptor for some of the services in the SDT?

Since the NIT contains the description of several TS, shall we use only the SDT Actual or also use the SDT Others?

What do we do if we discover a new SDT Other after the new NIT has already started to be sent? Shall we generate a new NIT and increment its version?

lars18th, 2019-11-19:
Hi @lelegard ,

> What do you mean by that? How would you specify it precisely?

OK. I describe the use case:

- You have a list of different sources. Some are SPTS and some are MPTS. And you want to filter _all_ of them with a single command. That's quite easy using the current implementation of the TSDuck. So after this initial filtering you have N SPTS streams, with unique Service ID (you enforce this). However, the rest of the PSI/SI data is missing (the SPTS only have PAT and PMT). And you want to create conformant DVB streams. So you need to add SDT and NIT tables. And the NIT table can describe _only_ this Stream. So, you don't need to care about other TS and with other services not running inside the same TS.

Then, with a command line `-P nit -c --copy-from-sdt` the services currently listed in the SDT table are inserted in the empty NIT table in the service_list_descriptor. That's all. I don't need more. Keep it simple!

You agree?

lelegard, 2019-11-21:
Option `--build-service-list-descriptors` has been added.

lars18th, 2019-11-21:
> Option `--build-service-list-descriptors` has been added.

Nice! I'm looking forward to trying it.
Thank you!

closed 2019-12-02 by lelegard