tsduck/tsduck #462: Preserving source port on MPE with --local-address forwarding

igorauad, 2019-11-20:
So, I'm trying to decapsulate MPE from a TS stream piped to `tsp` as follows:
```
... | tsp --realtime -P mpe -l --pid 32-33 --udp-forward --local-address 127.0.0.1  -O drop
```

The UDP datagrams are successfully extracted and forwarded to the loopback interface. However, I've noticed that the source port of the original UDP datagrams changes along this process. It comes originally say as `4434`, but as it goes through the loopback interface, it changes randomly. Is there any easy way to preserve the source port with the `mpe` plugin?

FYI, I'm following this approach because I have an application that is listening to multicast packets received on a specific interface. So I'm running it to listen on the loopback interface. However, this application also expectes that the source port matches with the target port.

lelegard, 2019-11-21:
> So I'm running it to listen on the loopback interface. However, this application also expectes that the source port matches with the target port.

This is unusual. Multicast applications normally filter on _destination_ address and port. Sometimes, the filtering is done on the source _address_ (SSM or for security reason).

You could probably easily implement a UDP gateway using netcat (command `nc`) with which you can specify a source port using option `-p`.

lelegard, 2019-11-21:
Better solution: I just added option `--local-port` to plugins `ip` (output) and `mpe`.

igorauad, 2019-11-22:
@lelegard that's great! Thanks so much for pushing this feature promptly.

Best regards

closed 2019-11-22 by igorauad