tsduck/tsduck #464: Suggestion: trigger plugin catching actions "START" and "STOP"

lars18th, 2019-11-21:
Hi @lelegard ,

I want to receive an UDP message when the `tsp` process is started, and another one when the process ends. I suspect that the plugin `trigger` can be a good candidate.

Any idea to do it?

Futhermore, I want also to ask about some _watchdog_ functionality. Or inside the same `trigger` plugin, of with another plugin (or internal functionality) that generates a signal (set flag) at regular intervals. The objective is quite simple: send an UDP message every _X_ seconds (or every _N_ packets) while the process chain is running.

What you think?

Thank you for this impresive and amazin project!

lelegard, 2019-11-21:
> I want to receive an UDP message when the tsp process is started, and another one when the process ends. I suspect that the plugin trigger can be a good candidate.

I can add `--start` and  `--stop` options to trigger the action on start and/or stop.

Otherwise, you may simply add commands in a script, just before and after the `tsp` command with the same result.

> Futhermore, I want also to ask about some watchdog functionality. Or inside the same trigger plugin, of with another plugin (or internal functionality) that generates a signal (set flag) at regular intervals.

With a regular stream, when packets are flowing at a regular rate, a time interval is equivalent to a packet interval. In that case, `-P filter --every ... --set-label ...` would do the job.

lars18th, 2019-11-21:
> > I want to receive an UDP message when the tsp process is started, and another one when the process ends. I suspect that the plugin trigger can be a good candidate.
> 
> I can add `--start` and `--stop` options to trigger the action on start and/or stop.

Great!
And please, if you can, catch also "kill" calls for sending the "stop" message (I want to receive the signal, if it's possible, with a regular quit). And if you want, as a bonus the "error" option will be useful too (so `--start`, `--stop` and `--error`).

> Otherwise, you may simply add commands in a script, just before and after the `tsp` command with the same result.

I request this, because this isn't possible. My idea is to use the tsp process as a childprocess of another process. In this case, the execution of the script can generate other side errors. So, I prefer to just run one process: the tsp one.

> With a regular stream, when packets are flowing at a regular rate, a time interval is equivalent to a packet interval. In that case, `-P filter --every ... --set-label ...` would do the job.

Yes. However I don't know how to do it with VBR streams. Any option to have some `-P filter --every-seconds XX` ?

Regards.

lelegard, 2019-11-21:
> And please, if you can, catch also "kill" calls for sending the "stop" message (I want to receive the signal, if it's possible, with a regular quit). And if you want, as a bonus the "error" option will be useful too (so --start, --stop and --error).

A plugin has three main methods: `start()` (called once), `processPacket()` (called for every packet) and `stop()` (called once). The `stop()` method is called when the process terminates or is interrupted (Ctrl-C is caught by `tsp` which then properly terminates all plugins one by one).

So, the "stop" trigger will be sent on Ctrl-C also but it will not be possible to discriminate between regular termination and interruption.

Note that if the interruption is a violent crash (assertion failure, segmentation fault, etc.), you will not be notified. If you need strict robustness, then you should go through a script approach. And if the script is a problem for your launching application, then you should modify your launching application.

> Any option to have some -P filter --every-seconds XX ?

In that case, this is no longer a packet processing because it does not read input packet and does not modify them. It simply ignores packets. However, as a packet processing plugin, it will be called on TS packets only.

And remember that, even with CBR streams, the plugins are not invoked at regular time interval. To maximize the throughput, all packets are in a large common buffer and each plugin executes in its own thread, notifying the next thread every N packets only. Because the processing time can dramatically vary between plugins and even between packets in one plugin, a CBR stream is not a guarantee for a strict time interval. Your external monitoring application should have some tolerance in the timeout. For instance, if you send a trigger every 5 seconds, the monitoring application should use a 8 or 10 seconds timeout. In this approach, CBR vs. VBR does not make much difference.

lars18th, 2019-11-21:
Hi @lelegard ,

> So, the "stop" trigger will be sent on Ctrl-C also but it will not be possible to discriminate between regular termination and interruption.
> Note that if the interruption is a violent crash [...]

I know, I know it! However two things:

1. If the process is "terminated" (end-of-input), then this is END (termination). In any other case it's STOP (interruption).
2. And I'm not speaking about "crash", but about "kill" signals. I feel you catch them, right? So, the objective is quite simple: when a tsp process is running as a childprocess and it receives a kill signal, then the intrruption message can be sended, right?

>> Any option to have some -P filter --every-seconds XX ?

>In that case, this is no longer a packet processing because it does not read input packet and does not modify them. It simply ignores packets. However, as a packet processing plugin, it will be called on TS packets only.

Well, what I'm thinking is a plugin that read packets and it does this:
- For every new packet check the time of the packet, and if the this time is greater than the interval, then set the label, without touching packets (it pass all packets).
- And the time is calculated based on bitrate and PCR marks.
- So, if the stream is paused, then no packets flow and no messages are passed. And if packets are passing, then based on PCR every _N_ seconds a flag is set.

We can use this as a simple _external_ stream whatchdog, because if the stream flows, then we receive "pings" at some intervals.
What you think?

lelegard, 2019-11-21:
Options `--start` and `--stop` have been added. The `--stop` option also includes user interrupt (Ctrl-C) and `kill` command on Unix systems (Linux, macOS). There won't be any `--error` option since a plugin does not know why it is terminated. Of course, `tsp` can make the difference between a normal termination and an interrupt but the information is not passed to the plugin. Passing this information would require to break the plugin API and modify the source code of all plugins. It does not worth it.

Concerning the time interval, the feature was already present, at least indirectly. I just did not think about it. First, without specific label or start/stop, the trigger occurs by default on _all_ packets. Second, to avoid "avalanche effects", the option `--min-inter-time` limits the time between two triggers. So, the command `-P trigger --min-inter-time 5000` does what you want (here trigger the action every 5 seconds).

lars18th, 2019-11-22:
> Options `--start` and `--stop` have been added. The `--stop` option also includes user interrupt (Ctrl-C) and `kill` command on Unix systems (Linux, macOS). There won't be any `--error` option since [...]

Great! That's sufficient to me. :+1: 

> Concerning the time interval, the feature was already present, at least indirectly. I just did not think about it. First, without specific label or start/stop, the trigger occurs by default on _all_ [...]

Nice! I'll try it. :wink:

closed 2019-12-02 by lelegard